openhab/org.openhab.binding.zwave #1258: Missing semicolon for channel color_temperature results in "No command converter set for command" error

Noobud, 2019-11-25:
* OH2 Version: 2.5M5
* Binding Version: 2.5M5

* Device name: AEON Labs ZW121, other
* Device version: 1.3, other
* Link to database: https://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/639

Hi all,
I found an issue with the device database. All devices with colorMode=DIFF_WHITE are broken. In the resulting OpenHab2 XML File is a semicolon missing.

Problem is the channel configuration.
Channel: color_temperature
colorMode=DIFF_WHITE

Results in: `<property name="binding:*:PercentType">COMMAND_CLASS_SWITCH_COLORcolorMode=DIFF_WHITE</property>`

Expected is: `<property name="binding:*:PercentType">COMMAND_CLASS_SWITCH_COLOR;colorMode=DIFF_WHITE</property>`

I documented the issue in the openHAB community:
https://community.openhab.org/t/zw121-aeotec-led-strips-no-command-converter/85286

The definition of the devices seems to be ok, only the export to openHAB struggels.

Hope its clear,
Many thanks,
Daniel

cdjackson, 2019-11-25:
The exporter should now be fixed.

Thanks.

Chris

closed 2019-11-25 by cdjackson

cdjackson, 2019-11-25:
Oh - and I also re-exported the ZW121 so this should be updated tonight.