openhab/org.openhab.binding.zwave #1261: Database Export needed for 18 devices, missing semicolon in DIFF_WHITE channel

Noobud, 2019-11-30:
Hi Chris,
you fixed my issue in ticket #1258 , many thanks.
There are still 18 devices in the Database that needs a new export to solve the missing semicolon for DIFF_WHITE Channel.

1. https://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/91
2. https://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/527
3. https://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/1134
4. https://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/942
5. https://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/949
6. https://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/1127
7. https://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/1095
8. https://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/598
9. https://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/730
10. https://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/836
11. https://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/843
12. https://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/1126
13. https://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/567
14. https://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/1067
15. https://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/1012
16. https://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/793
17. https://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/619
18. https://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/237

Daniel

cdjackson, 2019-11-30:
I don't really want to do this before 2.5 is released. This will be updated at the next database update, but there's not really a lot to do so Im not sure we need to leave this open?

Noobud, 2019-11-30:
I do not care about it, I don't own any of this devices.
I wanted only to document this so it will not be forgotten.

cdjackson, 2019-11-30:
It can't be forgotten - it will automatically be completed at the next full database update since the database issue was resolved as per the previous issue you raised. It "just" requires a full update, or for someone to go through and manually request an update for all these devices. A full update of the 1000+ devices is to risky to do right before the 2.5 release and will be done again after (I did the 2.5 export a few weeks back).

closed 2019-11-30 by cdjackson