openhab/org.openhab.binding.zwave #1263: Eurotronic Spirit - Unsupported mode type 31

FPille, 2019-12-01:
* OH2 Version: 2.5.0~M5-1
* Binding Version: 2.5.0.M5
* Device name: Eurotronic Spirit Z-Wave Plus
* Device version: refer to database link
* Link to [database](http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list): https://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/710
* If there is an XML being generted for the device (in the /userdata/zwave folder), please provide it. [network_ee3a339d__node_3.xml.log](https://github.com/openhab/org.openhab.binding.zwave/files/3908663/network_ee3a339d__node_3.xml.log)

This issue refers to: to: [Eurotronic Spirit - Unsupported mode type 31](https://community.openhab.org/t/eurotronic-spirit-unsupported-mode-type-31/57265)
In order to directly control the Spirit's valve a channel linked to the COMMAND_CLASS_BASIC is required. 

Currently selecting _MANUAL_ (Manufacturer Specific mode) of COMMAND_CLASS_THERMOSTAT_MODE leads to the following error:
`2018-11-19 19:13:08.108 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 13: Command received zwave:device:512:node13:thermostat_mode --> 31 [DecimalType]`
`2018-11-19 19:13:08.116 [DEBUG] [lass.ZWaveThermostatModeCommandClass] - NODE 13: setValueMessage 31, modeType empty false`
`2018-11-19 19:13:08.121 [ERROR] [lass.ZWaveThermostatModeCommandClass] - NODE 13: Unsupported mode type 31`
`2018-11-19 19:13:08.125 [WARN ] [nverter.ZWaveThermostatModeConverter] - NODE 13: Generating message failed for command class = COMMAND_CLASS_THERMOSTAT_MODE, endpoint = 0`
`2018-11-19 19:13:08.128 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 13: No messages returned from converter`

[Jeremys explanation:](https://community.openhab.org/t/eurotronic-spirit-unsupported-mode-type-31/57265/72?u=frapi)
> Yes, the Manual mode is already available on the device as a Thermostat Mode. However it needs to be sent the value ‘0xFE’ (‘Manufacturer Specific’) via the Basic command class first before it will recognise it. Once that has been sent, the device will respond to the ‘0x1F’ (‘Manufacturer Specific’) command via the existing ‘thermostat_mode’ Channel. At that point, direct control of the valve position will be available via the existing ‘switch_dimmer’ Channel.

Here is the solution of [Mihai_Badea](https://community.openhab.org/t/eurotronic-spirit-unsupported-mode-type-31/57265/59?u=frapi): 
[spirit_0_0.xml.log](https://github.com/openhab/org.openhab.binding.zwave/files/3908668/spirit_0_0.xml.log)

(xml files have been renamed to log to get them uploaded)

cdjackson, 2020-02-01:
Added channel ```basic_number```.

closed 2020-02-01 by cdjackson

gmagnotta, 2020-03-23:
Hi,

I'm using openhab 2.5.3 and I'm still having the "Unsupported mode type 31" error although I try to send the basic 254 command before switching to manual mode.

Please, can you check the attached log?

Thanks

[openhab.log](https://github.com/openhab/org.openhab.binding.zwave/files/4368543/openhab.log)

radix1989, 2020-11-15:
Hi,

I have some Eurotronic Spirit Z-Wave Thermostats and some identically constructed from Aeotec.
So could you please add the basic_number channel also to that device?
[(https://opensmarthouse.org/zwavedatabase/1150)](url)

Thanks in advance!

domik82, 2021-01-23:
@cdjackson I'm using Openhab 3 and I'm still getting errors while trying to switch using rule.
Us there a “proper” way of switching it to manual control of the valve? 

`2021-01-23 18:17:52.027 [INFO ] [openhab.event.ItemCommandEvent      ] - Item 'GabinetGrzejnikNode10_Thermostatmode' received command 31

2021-01-23 18:17:52.055 [INFO ] [penhab.event.ItemStatePredictedEvent] - Item 'GabinetGrzejnikNode10_Thermostatmode' predicted to become 31

2021-01-23 18:17:52.075 [INFO ] [openhab.event.ItemStateChangedEvent ] - Item 'GabinetGrzejnikNode10_Thermostatmode' changed from 1 to 31

==> /var/log/openhab/openhab.log <==

2021-01-23 18:17:52.080 [ERROR] [lass.ZWaveThermostatModeCommandClass] - NODE 10: Unsupported mode type 31

2021-01-23 18:17:52.084 [WARN ] [nverter.ZWaveThermostatModeConverter] - NODE 10: Generating message failed for command class = COMMAND_CLASS_THERMOSTAT_MODE, endpoint = 0`