openhab/org.openhab.binding.zwave #1264: Interview is repeatedly canceled

GottaMakeANameFast, 2019-12-04:
* OH2 Version: **2.5.0 Milestone 5**
* Binding Version: **binding-zwave - 2.5.0.M5**
* Device name: 11-01-011 Strips-MaZW
* Link to [database]: https://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/251 
* XML: https://products.z-wavealliance.org/Products/2146/XML

When adding the above listed device, sometimes it is included without interview being completed. Device is put to sleep before commands such as lifeline association or wake up interval are set. 

If i manually wake up the device, controller will seemingly send one command in the interview queue and then put the device back to sleep. So for me to complete the interview i effectively need to wake up the device several times in order to complete the entire interview.

In the link below i have attached some debug and z-wave sniffer logs testing this with the above mentioned Strips-MaZw device. There are 5 inclusions in total. The first two were fine, with interview going through and association set being sent during inclusion. The other three however, did not get association set and thus had to be woken up and fixed in various ways to get working.
https://www.dropbox.com/s/loip5gphadtddor/20191125%20Testing%20Openhab.7z?dl=0

The .zlf files can be opened with the SiLabs zniffer tool if you have it.

Please read discussions here for more information:
https://community.openhab.org/t/z-wave-inclusion-finished-without-lifeline-association-set/86820

Timtam, 2020-03-28:
Furthermore, adding the named device will end up in receiving a message and registering the device as OFFLINE: not communicating with the controller, even though it just sent a wake-up message and cannot even be offline, due to being battery-driven. I'll attach the log. Anyone else got them to work somehow? I tried for several hours now, but its always the same. Inclusion is ok, but it seems to not being properly registered (no xml file within the userdata/zwave folder either). 
Any way to fix this for later?
Relevant node is node 8.
[zwave-log.txt](https://github.com/openhab/org.openhab.binding.zwave/files/4397841/zwave-log.txt)

stefan-hoehn, 2020-11-16:
@cdjackson Will there be a fix eventually? Any way I can support?

bwosborne2, 2020-11-16:
Assuming this is a battery operated device I think the issue is with the device firmware. 

Unfortunately most battery device need to be awakened repeatedly before fully responding to the NIF request as required by the Z-wave standard.

chris is free to correct me.