openhab/org.openhab.binding.zwave #1266: [Doc] Provide link for adding information about missing things

austvik, 2019-12-10:
When you add a thing of a unknown type, you get this information:

> The device is not in the database. If the device attributes show that this device has a valid manufacturer ID, device ID and type, then this is likely the case (eg. you see a label like "Z-Wave node 1 (0082:6015:020D::2.0)"). Even if the device appears to be in the database, some manufacturers use multiple sets of references for different regions or versions, and your device references may not be in the database. In either case, the database must be updated and you should raise an issue to get this addressed.

"you should rause an issue to get this addressed" was hard to figure out how to do for a newcomer.

This is what I tried, and as you see, I did it wrong: https://github.com/openhab/openhab-core/issues/1266

My suggestion is to add a link to the prefered way. E.g. https://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-database-guide

cdjackson, 2019-12-11:
> This is what I tried, and as you see, I did it wrong: openhab/openhab-core#1266 <https://github.com/openhab/openhab-core/issues/1266>

The link is already in the readme isn’t it? You raised an issue in totally the wrong repository - I guess if you had looked in the zwave repository you would have seen this?

openhab-5iver, 2019-12-11:
There is a link in the documentation... way down at the bottom of[ the Supported Things](https://www.openhab.org/addons/bindings/zwave/doc/things.html) section. I think it would be good to also have a link [right into the main doc](https://www.openhab.org/addons/bindings/zwave/#supported-things), or just state that the link is at the bottom of the Things list. Could a link also be placed in https://github.com/openhab/org.openhab.binding.zwave/blob/master/src/main/resources/ESH-INF/thing/device.xml#L16?

cdjackson, 2019-12-11:
Feel free to propose a PR...

cdjackson, 2020-01-04:
Closing this. To me, the bottom of the documentation is still the main doc and there's a clear section there titled "ZWave Device Database".

If someone thinks this is insufficient, then please propose improvements via a PR.

closed 2020-01-04 by cdjackson