openhab/org.openhab.binding.zwave #1274: ZWave 2.5 binding Java Exception

kholthaus, 2019-12-27:
Since upgrading to OpenHAB2 v2.5 I can no longer use the REST API to update the configuration of a ZWave Thing. Specifically, I am trying to update the key codes on a BE469 SmartLock.

I get a Java exception:
`2019-12-27 14:44:08.366 [ERROR] [st.core.internal.thing.ThingResource] - Exception during HTTP PUT request for update config at 'things/zwave:device:16bc3381d30:node14/config'
java.lang.ClassCastException: org.openhab.binding.zwave.handler.ZWaveSerialHandler cannot be cast to org.openhab.binding.zwave.handler.ZWaveControllerHandler
        at org.openhab.binding.zwave.internal.ZWaveConfigProvider.getConfigDescription(ZWaveConfigProvider.java:180) ~[?:?]
        at org.eclipse.smarthome.config.core.ConfigDescriptionRegistry.fillFromProviders(ConfigDescriptionRegistry.java:214) ~[bundleFile:?]
        at org.eclipse.smarthome.config.core.ConfigDescriptionRegistry.getConfigDescription(ConfigDescriptionRegistry.java:183) ~[bundleFile:?]
        at org.eclipse.smarthome.config.core.ConfigDescriptionRegistry.getConfigDescription(ConfigDescriptionRegistry.java:236) ~[bundleFile:?]
        at org.eclipse.smarthome.io.rest.core.internal.thing.ThingResource.normalizeConfiguration(ThingResource.java:840) ~[bundleFile:?]
        at org.eclipse.smarthome.io.rest.core.internal.thing.ThingResource.updateConfiguration(ThingResource.java:440) [bundleFile:?]
        at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method) ~[?:1.8.0_232]
        at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62) ~[?:1.8.0_232]
        at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43) ~[?:1.8.0_232]
        at java.lang.reflect.Method.invoke(Method.java:498) ~[?:1.8.0_232]
        at org.glassfish.jersey.server.model.internal.ResourceMethodInvocationHandlerFactory$1.invoke(ResourceMethodInvocationHandlerFactory.java:81) [bundleFile:?]
        at org.glassfish.jersey.server.model.internal.AbstractJavaResourceMethodDispatcher$1.run(AbstractJavaResourceMethodDispatcher.java:144) [bundleFile:?]
        at org.glassfish.jersey.server.model.internal.AbstractJavaResourceMethodDispatcher.invoke(AbstractJavaResourceMethodDispatcher.java:161) [bundleFile:?]
        at org.glassfish.jersey.server.model.internal.JavaResourceMethodDispatcherProvider$ResponseOutInvoker.doDispatch(JavaResourceMethodDispatcherProvider.java:160) [bundleFile:?]
        at org.glassfish.jersey.server.model.internal.AbstractJavaResourceMethodDispatcher.dispatch(AbstractJavaResourceMethodDispatcher.java:99) [bundleFile:?]
        at org.glassfish.jersey.server.model.ResourceMethodInvoker.invoke(ResourceMethodInvoker.java:389) [bundleFile:?]
        at org.glassfish.jersey.server.model.ResourceMethodInvoker.apply(ResourceMethodInvoker.java:347) [bundleFile:?]
        at org.glassfish.jersey.server.model.ResourceMethodInvoker.apply(ResourceMethodInvoker.java:102) [bundleFile:?]
        at org.glassfish.jersey.server.ServerRuntime$2.run(ServerRuntime.java:326) [bundleFile:?]
        at org.glassfish.jersey.internal.Errors$1.call(Errors.java:271) [bundleFile:?]
        at org.glassfish.jersey.internal.Errors$1.call(Errors.java:267) [bundleFile:?]
        at org.glassfish.jersey.internal.Errors.process(Errors.java:315) [bundleFile:?]
        at org.glassfish.jersey.internal.Errors.process(Errors.java:297) [bundleFile:?]
        at org.glassfish.jersey.internal.Errors.process(Errors.java:267) [bundleFile:?]
        at org.glassfish.jersey.process.internal.RequestScope.runInScope(RequestScope.java:317) [bundleFile:?]
        at org.glassfish.jersey.server.ServerRuntime.process(ServerRuntime.java:305) [bundleFile:?]
        at org.glassfish.jersey.server.ApplicationHandler.handle(ApplicationHandler.java:1154) [bundleFile:?]
        at org.glassfish.jersey.servlet.WebComponent.serviceImpl(WebComponent.java:473) [bundleFile:?]
        at org.glassfish.jersey.servlet.WebComponent.service(WebComponent.java:427) [bundleFile:?]
        at org.glassfish.jersey.servlet.ServletContainer.service(ServletContainer.java:388) [bundleFile:?]
        at org.glassfish.jersey.servlet.ServletContainer.service(ServletContainer.java:341) [bundleFile:?]
        at org.glassfish.jersey.servlet.ServletContainer.service(ServletContainer.java:228) [bundleFile:?]
        at com.eclipsesource.jaxrs.publisher.internal.ServletContainerBridge.service(ServletContainerBridge.java:76) [bundleFile:?]
        at org.eclipse.jetty.servlet.ServletHolder.handle(ServletHolder.java:852) [bundleFile:9.4.20.v20190813]
        at org.eclipse.jetty.servlet.ServletHandler.doHandle(ServletHandler.java:544) [bundleFile:9.4.20.v20190813]
        at org.ops4j.pax.web.service.jetty.internal.HttpServiceServletHandler.doHandle(HttpServiceServletHandler.java:71) [bundleFile:?]
        at org.eclipse.jetty.server.handler.ScopedHandler.handle(ScopedHandler.java:143) [bundleFile:9.4.20.v20190813]
        at org.eclipse.jetty.security.SecurityHandler.handle(SecurityHandler.java:536) [bundleFile:9.4.20.v20190813]
        at org.eclipse.jetty.server.handler.HandlerWrapper.handle(HandlerWrapper.java:127) [bundleFile:9.4.20.v20190813]
        at org.eclipse.jetty.server.handler.ScopedHandler.nextHandle(ScopedHandler.java:235) [bundleFile:9.4.20.v20190813]
        at org.eclipse.jetty.server.session.SessionHandler.doHandle(SessionHandler.java:1581) [bundleFile:9.4.20.v20190813]
        at org.eclipse.jetty.server.handler.ScopedHandler.nextHandle(ScopedHandler.java:233) [bundleFile:9.4.20.v20190813]
        at org.eclipse.jetty.server.handler.ContextHandler.doHandle(ContextHandler.java:1307) [bundleFile:9.4.20.v20190813]
        at org.ops4j.pax.web.service.jetty.internal.HttpServiceContext.doHandle(HttpServiceContext.java:293) [bundleFile:?]
        at org.eclipse.jetty.server.handler.ScopedHandler.nextScope(ScopedHandler.java:188) [bundleFile:9.4.20.v20190813]
        at org.eclipse.jetty.servlet.ServletHandler.doScope(ServletHandler.java:482) [bundleFile:9.4.20.v20190813]
        at org.eclipse.jetty.server.session.SessionHandler.doScope(SessionHandler.java:1549) [bundleFile:9.4.20.v20190813]
        at org.eclipse.jetty.server.handler.ScopedHandler.nextScope(ScopedHandler.java:186) [bundleFile:9.4.20.v20190813]
        at org.eclipse.jetty.server.handler.ContextHandler.doScope(ContextHandler.java:1204) [bundleFile:9.4.20.v20190813]
        at org.eclipse.jetty.server.handler.ScopedHandler.handle(ScopedHandler.java:141) [bundleFile:9.4.20.v20190813]
        at org.ops4j.pax.web.service.jetty.internal.JettyServerHandlerCollection.handle(JettyServerHandlerCollection.java:80) [bundleFile:?]
        at org.eclipse.jetty.server.handler.HandlerWrapper.handle(HandlerWrapper.java:127) [bundleFile:9.4.20.v20190813]
        at org.eclipse.jetty.server.Server.handle(Server.java:494) [bundleFile:9.4.20.v20190813]
        at org.eclipse.jetty.server.HttpChannel.handle(HttpChannel.java:374) [bundleFile:9.4.20.v20190813]
        at org.eclipse.jetty.server.HttpConnection.onFillable(HttpConnection.java:268) [bundleFile:9.4.20.v20190813]
        at org.eclipse.jetty.io.AbstractConnection$ReadCallback.succeeded(AbstractConnection.java:311) [bundleFile:9.4.20.v20190813]
        at org.eclipse.jetty.io.FillInterest.fillable(FillInterest.java:103) [bundleFile:9.4.20.v20190813]
        at org.eclipse.jetty.io.ssl.SslConnection$DecryptedEndPoint.onFillable(SslConnection.java:426) [bundleFile:9.4.20.v20190813]
        at org.eclipse.jetty.io.ssl.SslConnection.onFillable(SslConnection.java:320) [bundleFile:9.4.20.v20190813]
        at org.eclipse.jetty.io.ssl.SslConnection$2.succeeded(SslConnection.java:158) [bundleFile:9.4.20.v20190813]
        at org.eclipse.jetty.io.FillInterest.fillable(FillInterest.java:103) [bundleFile:9.4.20.v20190813]
        at org.eclipse.jetty.io.ChannelEndPoint$2.run(ChannelEndPoint.java:117) [bundleFile:9.4.20.v20190813]
        at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.runTask(EatWhatYouKill.java:336) [bundleFile:9.4.20.v20190813]
        at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.doProduce(EatWhatYouKill.java:313) [bundleFile:9.4.20.v20190813]
        at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.tryProduce(EatWhatYouKill.java:171) [bundleFile:9.4.20.v20190813]
        at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.run(EatWhatYouKill.java:129) [bundleFile:9.4.20.v20190813]
[... cut ...]
* OH2 Version: 2.5
* Binding Version 2.5

If the problem is associated with a device, please provide the following -:

* Device name: Allegion BE469 Touchscreen Deadbolt
* Device version:113.22
* Link to [database](http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list): https://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/303

Querying devices using the REST API still appears to work, as does updating the state of the associated items.

openhab-5iver, 2019-12-28:
Put triple ticks before and after your log, so that it is readable! What exactly did you send through the REST API, so that it can be reproduced?

kholthaus, 2019-12-28:
Apologies about the ticks - used the "insert code" button in the editor which only produced single ones.

Regarding what to send through the API:

The URL would be (using PUT): ```/things/<thingID>/config```
The request body would be something like this:
```
{
  usercode_code_1: "12345",
  usercode_label_1: "User 1",
  usercode_code_2: "12345",
  usercode_label_2: "User 2",
  usercode_code_3: "",
  usercode_label_3: ""
}
```

(in my case, I would send code and label for all 30 key slots, but chose not to copy/paste the remaining 27 empty ones).

kholthaus, 2020-01-10:
Tried to do some digging myself... Java isn't really my language, but this error message doesn't seem to make any sense to me. ```ZWaveSerialHandler extends ZWaveControllerHandler``` is how ```ZWaveSerialHandler``` is implemented.
Wouldn't that mean that I any ```ZWaveSerialHandler``` is a ```ZWaveControllerHandler``` automatically? I mean, how can that cast fail? Do I have some weird mix of code on my system? Any advice would really be appreciated.

kholthaus, 2020-01-18:
This appears to be fixed after upgrading to 2.5.1.

closed 2020-01-18 by kholthaus