openhab/org.openhab.binding.zwave #1278: Unknown device 0063:5044:3037:5.21

ryanillman, 2019-12-30:
I have a Jasco ZW3104 / 28166 that is not being recognized. It's a Jasco-branded plug vs a GE-branded one, and appears to have an off-by-one device ID compared to the GE 28167. 

OpenHab 2.5.0-Release

Device Name: Jasco ZW3104 / 28166-2
Device Version: ?
ThingType: 0063:5044:3037:5.21
Database Link: Not in database
Info Sheet: https://products.z-wavealliance.org/products/126

[network_d07a6f8f__node_73.zip](https://github.com/openhab/org.openhab.binding.zwave/files/4010980/network_d07a6f8f__node_73.zip)

Debug log:
```
Receive Message = 01 09 00 04 00 49 03 26 03 44 D9 
Transaction SendNextMessage 1 out at start. Holdoff false.
processReceiveMessage input 0<>128 : Message: class=ApplicationCommandHandler[4], type=Request[0], dest=73, callback=0, payload=00 49 03 26 03 44 
Received msg (0): Message: class=ApplicationCommandHandler[4], type=Request[0], dest=73, callback=0, payload=00 49 03 26 03 44 
 lastTransaction null
NODE 73: Application Command Request (ALIVE:DYNAMIC_VALUES)
NODE 73: Incoming command class COMMAND_CLASS_SWITCH_MULTILEVEL, endpoint 0
NODE 73: SECURITY not supported
NODE 73: Received COMMAND_CLASS_SWITCH_MULTILEVEL V2 SWITCH_MULTILEVEL_REPORT
NODE 73: Switch Multi Level report, value = 68
NODE 73: Got an event from Z-Wave network: ZWaveCommandClassValueEvent
NODE 73: Got a value event from Z-Wave network, endpoint=0, command class=COMMAND_CLASS_SWITCH_MULTILEVEL, value=68
NODE 73: Commands processed 1.
NODE 73: Checking command org.openhab.binding.zwave.internal.protocol.ZWaveCommandClassPayload@88583.
NODE 73: Command verified org.openhab.binding.zwave.internal.protocol.ZWaveCommandClassPayload@88583.
NODE 73: notifyTransactionResponse TID:3005 DONE
NODE 73: Got an event from Z-Wave network: ZWaveTransactionCompletedEvent
TID 3005: Transaction event listener: DONE: DONE -> 
Transaction completed - outstandingTransactions 1
NODE 73: Node Init response (0) org.openhab.binding.zwave.internal.protocol.ZWaveTransactionResponse@4997e0
Transaction completed - outstandingTransactions 0 
NODE 73: Node Init transaction completed with response COMPLETE
ZWaveReceiveThread queue empty
NODE 73: Node advancer: Initialisation complete!
Transaction SendNextMessage 0 out at start. Holdoff false.
NODE 73: Node advancer - advancing to DYNAMIC_END
NODE 73: Got an event from Z-Wave network: ZWaveInitializationStateEvent
NODE 73: Serializing to file /var/lib/openhab2/zwave/network_d07a6f8f__node_73.xml
NODE 73: Node advancer - advancing to DONE
NODE 73: Got an event from Z-Wave network: ZWaveInitializationStateEvent
NODE 73: Serializing to file /var/lib/openhab2/zwave/network_d07a6f8f__node_73.xml
NODE 73: Device discovery could not resolve to a thingType! 0063:5044:3037::5.21
```

cdjackson, 2019-12-30:
I've updated the database so it will be updated in the snapshot binding in a week or so.  If you unsure how to upgrade, please discuss on the forum.

closed 2019-12-30 by cdjackson

ryanillman, 2019-12-30:
Wow that was amazingly fast. Thanks!