openhab/org.openhab.binding.zwave #1288: August Lock - 'alarm_entry' COMMAND_CLASS_ALARM uses incorrect item type

JackBeQuick87, 2020-01-09:
The binding incorrectly assigns the "Alarm (access)" channel to an OpenHab Item type of 'Switch'.  The value reported from the lock is an integer status code.  Left unfixed, an item assigned to that channel is useless.

Environment:
- OH2 Version: 2.4.0
- Binding Version 2.4.0

Associated device:
- Device name: ASL-03 August Smart Lock Pro 3rd Gen (august_asl03_00_000)
- Device firmware version: 1.13.2-1.59.0
- Link to database: [ASL-03 | August Smart Lock Pro 3rd Gen](https://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/743)

Generated device node XML file:
[network_[REDACTED]__node_3.xml.txt](https://github.com/openhab/org.openhab.binding.zwave/files/4037603/network_.REDACTED.__node_3.xml.txt)

### Suggested fix:
Change  org.openhab.binding.zwave/src/main/resources/ESH-INF/thing/august/asl03_0_0.xml as follows:

``` XML
  <channel id="alarm_entry" typeId="alarm_entry">
    <label>Alarm (access)</label>
    <properties>
      <property name="binding:*:OnOffType">COMMAND_CLASS_ALARM;type=ACCESS_CONTROL</property>
    </properties>
  </channel>
```
to
``` XML
  <channel id="notification_access_control" typeId="notification_access_control">
    <label>Notification (Access Control)</label>
    <properties>
      <property name="binding:*:DecimalType">COMMAND_CLASS_ALARM;type=ACCESS_CONTROL</property>
    </properties>
  </channel>
```

## Other information:
### Finding out the status codes:
Configuring the bindings log:set to DEBUG exposes the values sent by the lock:
``` Shell
>grep ZWaveAlarmCommandClass zwave.log
2020-01-08 16:28:17.791 [DEBUG] [ernal.protocol.commandclass.ZWaveAlarmCommandClass] - NODE 3: NOTIFICATION report - 0 = 0, event=5, status=255, plen=0
2020-01-08 16:28:17.792 [DEBUG] [ernal.protocol.commandclass.ZWaveAlarmCommandClass] - NODE 3: Alarm Type = ACCESS_CONTROL (0)
2020-01-08 16:32:42.458 [DEBUG] [ernal.protocol.commandclass.ZWaveAlarmCommandClass] - NODE 3: NOTIFICATION report - 0 = 0, event=2, status=255, plen=0
2020-01-08 16:32:42.459 [DEBUG] [ernal.protocol.commandclass.ZWaveAlarmCommandClass] - NODE 3: Alarm Type = ACCESS_CONTROL (0)
2020-01-08 16:35:44.373 [DEBUG] [ernal.protocol.commandclass.ZWaveAlarmCommandClass] - NODE 3: NOTIFICATION report - 0 = 0, event=3, status=255, plen=0
```

I have verified the 'event' codes to mean the following:

| Event value | Association
| ----------: | -----------
| 01 | Manually Locked
| 02 | Manually Unlocked
| 03 | Remotely Locked (via app or Z-Wave)
| 04 | Remotely Unlocked (via app or Z-Wave)
| 05 | Locked via Keypad
| 06 | Unlocked via Keypad
| 09 | Automatically Locked
| 11 | Lock Jammed

Credit to this post on a home-assistant forum [here](https://community.home-assistant.io/t/august-smart-lock-pro-zwave/28654/10) for hinting me on this status's existence.
It's possible one of the unseen values correlates to the lock's 'Auto Unlock' feature.  Further testing needed.

## For further consideration:
Evidently the Z-Wave Alliance has changed renamed the COMMAND_CLASS_ALARM to COMMAND_CLASS_NOTIFICATION.  (Source not directly linked.  Use: https://www.google.com/search?q=list+of+Z-Wave+command+classes+site%3Asilabs.com+type%3Axlsx)
I'm not sure what other implications the renaming will have for the binding, but perhaps 'type_id' should be changed to a custom mapping, either alarm_number or , specific to this August lock/binding.  

Thank you!
    -- John

cdjackson, 2020-01-09:
I suspect what you need to do ```alarm_raw``` channel which will provide the information you need. Please feel free to update the database -:

https://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-database-guide

Adding the above channel (not removing or renaming) should be sufficient I think, but if you have any questions, it would be best to ask on the openhab forum.

The change from ALARM to NOTIFICATION is quite an old one - the binding will not be changed as there is no need.

JackBeQuick87, 2020-01-09:
There is already a channel type in channels.xml that seems to properly enumerate these status codes.

``` XML
 <!-- Channel definition for zwave notification type=0x06/Access Control -->
    <channel-type id="notification_access_control" advanced="true">
        <item-type>Number</item-type>
        <label>Access Control</label>
        <description>Access Control</description>
        <state readOnly="true">
            <options>
                <option value="0">Previous Events cleared</option>
                <option value="1">Manual Lock Operation</option>
                <option value="2">Manual Unlock Operation</option>
                <option value="3">RF Lock Operation</option>
                <option value="4">RF Unlock Operation</option>
                <option value="5">Keypad Lock Operation</option>
                <option value="6">Keypad Unlock Operation</option>
                <option value="7">Manual Not Fully Locked Operation</option>
                <option value="8">RF Not Fully Locked Operation</option>
                <option value="9">Auto Lock Locked Operation</option>
                <option value="10">Auto Lock Not Fully Operation</option>
                <option value="11">LockJammed</option>
                <option value="12">All user codes deleted</option>
                <option value="13">Single user code deleted</option>
                <option value="14">New user code added</option>
                <option value="15">New user code not added due to duplicate code</option>
                <option value="16">Keypad temporary disabled</option>
                <option value="17">Keypad busy</option>
                <option value="18">New Program code Entered- Unique code for lock configuration</option>
                <option value="19">Manually Enter user Access code exceeds code limit</option>
                <option value="20">Unlock by RF with invalid user code</option>
                <option value="21">Locked by RF with invalid user code</option>
                <option value="22">Window/Door is open</option>
                <option value="23">Window/Door is closed</option>
                <option value="64">Barrier performing initialization process</option>
                <option value="65">Barrier operation (Open / Close) force has been exceeded</option>
                <option value="66">Barrier motor has exceeded manufacturer's operational time limit</option>
                <option value="67">Barrier operation has exceeded physical mechanical limits</option>
                <option value="68">Barrier unable to perform requested operation due to UL requirements</option>
                <option value="69">Barrier Unattended operation has been disabled per UL requirements</option>
                <option value="70">Barrier failed to perform Requested operation, device malfunction</option>
                <option value="71">Barrier Vacation Mode</option>
                <option value="72">Barrier Safety Beam Obstacle</option>
                <option value="73">Barrier Sensor Not Detected / Supervisory Error</option>
                <option value="74">Barrier Sensor Low Battery Warning</option>
                <option value="75">Barrier detected short in WallStation wires</option>
                <option value="76">Barrier associated with non-Z-wave remote control</option>
            </options>
        </state>
    </channel-type>
```

Wouldn't it be better to use this type instead of alarm_raw?

mhilbush, 2020-01-09:
I agree it would be helpful for the binding to expose this data for the August lock. I would prefer to add a new channel rather than changing the existing one, which might break rules for people who use the `alarm_entry` channel.

@cdjackson  Instead of using channel type `alarm_raw` for the new channel, wouldn't it be more straightforward to use the `notification_access_control` channel type?

cdjackson, 2020-01-09:
Sure - I don't mind using that channel instead - that's fine. Either way, it requires the update of the database as I posted earlier.

alarm_raw will not be broken by doing this - that channel provides all data provided from the lock in json format. However, I don't mind - you guys have the lock so if you want to use the notification_access_control, then please feel free to use it.

mhilbush, 2020-01-09:
> alarm_raw will not be broken by doing this

Right. I was referring to breaking `alarm_entry`, which was the original request.

cdjackson, 2020-01-09:
Ok, so I think we agree - shall we close this then? I'll leave it to @JackBeQuick87 to update the database.

mhilbush, 2020-01-09:
Ok by me.

closed 2020-01-09 by cdjackson