openhab/org.openhab.binding.zwave #1289: Abus Multisensor SHBW10000 motion sensor has no channel

Peter-DD, 2020-01-10:
I think The Abus Multisensor SHBW1000 has a wrong channel "alarm_burglar" in the database.
I see the same logs as in  #1112 

I already post a question [here](https://community.openhab.org/t/z-wave-abus-multisensor-shbw10000-motion-sensor-seems-to-doesnt-work/89663l) 

There is only an alarm_buglar offered. The alarm_buglar will only updated to off, after 10 seconds (10 seconds is set as parameter "13: Waiting time between two PIR triggers") of detected motion. No other channels for motions are offered and triggered on motion. Temperature-, huminidity-, luminance sensor items works pretty good.

Here is the log:
`2020-01-09 17:52:13.821 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 11 00 04 00 04 09 71 05 00 00 00 FF 07 08 00 7E 00 1D 
2020-01-09 17:52:13.830 [DEBUG] [nal.protocol.ZWaveTransactionManager] - processReceiveMessage input 0<>128 : Message: class=ApplicationCommandHandler[4], type=Request[0], dest=4, callback=0, payload=00 04 09 71 05 00 00 00 FF 07 08 00 7E 00 
2020-01-09 17:52:13.837 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Received msg (0): Message: class=ApplicationCommandHandler[4], type=Request[0], dest=4, callback=0, payload=00 04 09 71 05 00 00 00 FF 07 08 00 7E 00 
2020-01-09 17:52:13.842 [DEBUG] [nal.protocol.ZWaveTransactionManager] - lastTransaction null
2020-01-09 17:52:13.845 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 4: Application Command Request (ALIVE:DONE)
2020-01-09 17:52:13.851 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 4: resetResendCount initComplete=true isDead=false
2020-01-09 17:52:13.855 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 4: Incoming command class COMMAND_CLASS_ALARM, endpoint 0
2020-01-09 17:52:13.860 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 4: SECURITY NOT required on COMMAND_CLASS_ALARM
2020-01-09 17:52:13.863 [DEBUG] [tocol.commandclass.ZWaveCommandClass] - NODE 4: Received COMMAND_CLASS_ALARM V8 NOTIFICATION_REPORT
2020-01-09 17:52:13.869 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 4: NOTIFICATION report - 0 = 0, event=8, status=255, plen=0
2020-01-09 17:52:13.873 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 4: Alarm Type = BURGLAR (0)
2020-01-09 17:52:13.878 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 4: Got an event from Z-Wave network: ZWaveAlarmValueEvent
2020-01-09 17:52:13.882 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 4: Got a value event from Z-Wave network, endpoint=0, command class=COMMAND_CLASS_ALARM, value=255
2020-01-09 17:52:13.887 [DEBUG] [ternal.converter.ZWaveAlarmConverter] - NODE 4: Alarm converter processing NOTIFICATION
2020-01-09 17:52:13.891 [DEBUG] [ternal.converter.ZWaveAlarmConverter] - NODE 4: Alarm converter NOTIFICATION event is 8, type OnOffType
2020-01-09 17:52:13.894 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 4: Commands processed 1.
2020-01-09 17:52:13.898 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 4: Checking command org.openhab.binding.zwave.internal.protocol.ZWaveCommandClassPayload@2fd628.
2020-01-09 17:52:13.901 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Transaction completed - outstandingTransactions 0
2020-01-09 17:52:13.905 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Transaction completed - outstandingTransactions 0
2020-01-09 17:52:13.908 [DEBUG] [nal.protocol.ZWaveTransactionManager] - ZWaveReceiveThread queue empty
2020-01-09 17:52:13.914 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Transaction SendNextMessage 0 out at start. Holdoff false.
2020-01-09 17:52:23.939 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 12 00 04 00 04 0A 71 05 00 00 00 FF 07 00 01 08 7E 00 1C 
2020-01-09 17:52:23.951 [DEBUG] [nal.protocol.ZWaveTransactionManager] - processReceiveMessage input 0<>128 : Message: class=ApplicationCommandHandler[4], type=Request[0], dest=4, callback=0, payload=00 04 0A 71 05 00 00 00 FF 07 00 01 08 7E 00 
2020-01-09 17:52:23.956 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Received msg (0): Message: class=ApplicationCommandHandler[4], type=Request[0], dest=4, callback=0, payload=00 04 0A 71 05 00 00 00 FF 07 00 01 08 7E 00 
2020-01-09 17:52:23.960 [DEBUG] [nal.protocol.ZWaveTransactionManager] - lastTransaction null
2020-01-09 17:52:23.962 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 4: Application Command Request (ALIVE:DONE)
2020-01-09 17:52:23.967 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 4: resetResendCount initComplete=true isDead=false
2020-01-09 17:52:23.970 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 4: Incoming command class COMMAND_CLASS_ALARM, endpoint 0
2020-01-09 17:52:23.975 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 4: SECURITY NOT required on COMMAND_CLASS_ALARM
2020-01-09 17:52:23.978 [DEBUG] [tocol.commandclass.ZWaveCommandClass] - NODE 4: Received COMMAND_CLASS_ALARM V8 NOTIFICATION_REPORT
2020-01-09 17:52:23.982 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 4: NOTIFICATION report - 0 = 0, event=0, status=255, plen=1
2020-01-09 17:52:23.986 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 4: Alarm Type = BURGLAR (0)
2020-01-09 17:52:23.991 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 4: Got an event from Z-Wave network: ZWaveAlarmValueEvent
2020-01-09 17:52:23.994 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 4: Got a value event from Z-Wave network, endpoint=0, command class=COMMAND_CLASS_ALARM, value=255
2020-01-09 17:52:23.999 [DEBUG] [ternal.converter.ZWaveAlarmConverter] - NODE 4: Alarm converter processing NOTIFICATION
2020-01-09 17:52:24.002 [DEBUG] [ternal.converter.ZWaveAlarmConverter] - NODE 4: Alarm converter NOTIFICATION event is 0, type OnOffType
2020-01-09 17:52:24.006 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 4: Updating channel state zwave:device:16f8176ab25:node4:alarm_burglar to OFF [OnOffType]
2020-01-09 17:52:24.011 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 4: Commands processed 1.
2020-01-09 17:52:24.015 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 4: Checking command org.openhab.binding.zwave.internal.protocol.ZWaveCommandClassPayload@1bb3c73.
2020-01-09 17:52:24.018 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Transaction completed - outstandingTransactions 0
2020-01-09 17:52:24.020 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Transaction completed - outstandingTransactions 0
2020-01-09 17:52:24.025 [DEBUG] [nal.protocol.ZWaveTransactionManager] - ZWaveReceiveThread queue empty
2020-01-09 17:52:24.028 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Transaction SendNextMessage 0 out at start. Holdoff false.`

Peter-DD, 2020-01-10:
[log.txt](https://github.com/openhab/org.openhab.binding.zwave/files/4047865/log.txt)

cdjackson, 2020-01-10:
I've updated the database.

closed 2020-01-10 by cdjackson

Peter-DD, 2020-01-11:
Hi Chris,
Thank you very much for the update.
I've reinstalled the z-wave binding and excluded/included the Abus SHBW10000 but I don't see any alarm_motion channel. The behavior is the same as before. 
Should I wait a time and the try it again? Thank you, Peter

cdjackson, 2020-01-11:
Sorry - I should have said - I’ve updated the database,  but not the binding. I will do the binding update tonight and then the device will need to be rediscovered. To do this, update to the latest snapshot (tomorrow) and then delete the thing, hit the inbox/search button and add the device back in. That should make it pick up the newer definition.

Peter-DD, 2020-01-12:
After update to the latest snapshot and adding/removing thing the motion sensor is perfectly working now! Many thanks!