openhab/org.openhab.binding.zwave #1298: socat-based serial not recognized in 2.5.1

ratafakplachta, 2020-01-25:
Using Aeotec Z-Stick Gen5 remotely via socat:
socat -d pty,link=/dev/zwave,raw,user=openhab,group=dialout,mode=777 tcp:192.168.1.62:3333

Works fine in 2.5.0
Not recognized in 2.5.1 - Paper UI error:  Port /dev/zwave does not exist

ls -l /dev/zwave
lrwxrwxrwx 1 root root 10 Jan 25 13:19 /dev/zwave -> /dev/pts/0

Docker images tested:
openhab/openhab:2.5.0
openhab/openhab:2.5.1

cdjackson, 2020-01-25:
This is not related to teh binding. The binding uses core functions for accessing ports. I'd suggest to discuss this on the forum.

closed 2020-01-25 by cdjackson

ratafakplachta, 2020-01-25:
OK, thanks.
As 2.5.1 introduced serial port re-connection logic (ZWaveSerialHandler.java and others), thought it could have affected handling of virtual serial ports.

cdjackson, 2020-01-25:
The reconnection logic was just that though - I’m pretty sure that the change to use the OH serial port proxy was introduced before 2.4. I’d really suggest to discuss on the forum where there are a lot more people who can help - if there’s an issue identified, then we can open an issue here.