openhab/org.openhab.binding.zwave #1301: Cannot edit things in Openhab

mcmichels, 2020-01-26:
* OH2 Version: 2.5.1-2
* Binding Version (if SNAPSHOT, provide compile time as found on the filename in the console): 2.5.1

When i try editing a thing in PaperUI i get an Eroor 500: Internal Server Error. I set the debug level for PaperUI and can see the following error:
`{"error":{"message":"org.openhab.binding.zwave.handler.ZWaveSerialHandler cannot be cast to org.openhab.binding.zwave.handler.ZWaveControllerHandler","http-code":500,"exception":{"class":"java.lang.ClassCastException","message":"org.openhab.binding.zwave.handler.ZWaveSerialHandler cannot be cast to org.openhab.binding.zwave.handler.ZWaveControllerHandler","localized-message":"org.openhab.binding.zwave.handler.ZWaveSerialHandler cannot be cast to org.openhab.binding.zwave.handler.ZWaveControllerHandler"}}}`

scanlinevfx, 2020-05-02:
I'm having the same error here. Can't edit any z-wave things through habmin or paperui.

Openhab 2.5.4
Z-Wave plugin snapshot 2.5.5.202004271723 (same behaviour with 2.5.4)

What info would help to debug? I tried various trace level logs but couldn't find the right option to get a full stack trace of the error

curling the URLs that fail with 500 from the web logs shows

```
curl https://openhab.XXX/rest/config-descriptions/thing:zwave:device:XXXX:node2
{"error":{"message":"org.openhab.binding.zwave.handler.ZWaveSerialHandler cannot be cast to org.openhab.binding.zwave.handler.ZWaveControllerHandler","http-code":500,"exception":{"class":"java.lang.ClassCastException","message":"org.openhab.binding.zwave.handler.ZWaveSerialHandler cannot be cast to org.openhab.binding.zwave.handler.ZWaveControllerHandler","localized-message":"org.openhab.binding.zwave.handler.ZWaveSerialHandler cannot be cast to org.openhab.binding.zwave.handler.ZWaveControllerHandler"}}}%
```

jamesbraid, 2020-05-02:
I'm having the same error here. Can't edit any z-wave things through habmin or paperui.

Openhab 2.5.4
Z-Wave plugin snapshot 2.5.5.202004271723 (same behaviour with 2.5.4)

What info would help to debug? I tried various trace level logs but couldn't find the right option to get a full stack trace of the error

curling the URLs that fail with 500 from the web logs shows

```
curl https://openhab.XXX/rest/config-descriptions/thing:zwave:device:XXXX:node2
{"error":{"message":"org.openhab.binding.zwave.handler.ZWaveSerialHandler cannot be cast to org.openhab.binding.zwave.handler.ZWaveControllerHandler","http-code":500,"exception":{"class":"java.lang.ClassCastException","message":"org.openhab.binding.zwave.handler.ZWaveSerialHandler cannot be cast to org.openhab.binding.zwave.handler.ZWaveControllerHandler","localized-message":"org.openhab.binding.zwave.handler.ZWaveSerialHandler cannot be cast to org.openhab.binding.zwave.handler.ZWaveControllerHandler"}}}%
```

jamesbraid, 2020-05-04:
It seems like I had two instances of the org.openhab.binding.zwave module installed. Uninstalling the 2.5.4 version via "bundle:uninstall <id>" fixed it for me, and I no longer get the error when editing Zwave things. Sorry for the noise!