openhab/org.openhab.binding.zwave #1302: Aeotec WallMote (ZW130) No conversion in ZWaveMultiLevelSwitchConverter to StringType

vidia, 2020-01-28:
I have recently added an Aeotec Wallmote Quad to my network. After setting it up to detect swipes on the switch_startstop1 channels, I am getting the error `No conversion in ZWaveMultiLevelSwitchConverter to StringType` when interacting with the buttons. 

I did a bit of preliminary debugging. I am not familiar with this codebase, but I tried to track down the offending line and work backwards a bit. 
I have tracked down the [line of code](https://github.com/openhab/org.openhab.binding.zwave/blob/7cde8454169c45fa819497c6b203174b99b8060a/src/main/java/org/openhab/binding/zwave/internal/converter/ZWaveMultiLevelSwitchConverter.java#L131) that seems to be triggering the log and it looks like the Channel type is incorrect for this channel (it seems to be a String channel causing the issue). According to PaperUI, the only String type channels I have for this thing are the ones for `switch_startstop1` (etc), so I believe these channels can be the only culprits. It looks like it needs to be a Percent, OnOff, or IncreaseDecrease instead. If I am reading the XML correctly, it looks like this is set to be a String. This functionality was added in this [PR](https://github.com/openhab/org.openhab.binding.zwave/pull/1135/files).

I am not sure how to go about working in this codebase and testing the changes, or else I would have messed around a bit more, but hopefully what I have provided helps.

I have checked the forums for this issue and only one thread mentions it, but there has been no response on it so far (other than by me). 

https://community.openhab.org/t/aeotec-quad-wall-remote-no-conversion-in-zwavemultilevelswitchconverter-to-stringtype/76692

* OH2 Version: 2.5.1
* Binding Version: 2.5.1
* Device name: Aeotec WallMote Quad
* Device version: ZW 130-A
* Link to [database](http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list): https://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/535 
* Link to the manual for good measure: https://aeotec.freshdesk.com/support/solutions/articles/6000162392-wallmote-quad-user-guide- 
* I do not seem to have a /userdata/zwave folder. Happy to provide this detail if someone can point me in the right direction for it. 

I have loaded this output in the log viewer, but I don't see anything that would point to an issue. This log snippet represents the debug output for when I tap a button on the Wallmote. I can provide more data if it would be valuable, but I would hope this is the relevant portion.

```
2020-01-28 02:31:41.793 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 0F 00 04 00 07 09 60 0D 00 01 5B 03 12 00 04 D8 
2020-01-28 02:31:41.806 [DEBUG] [nal.protocol.ZWaveTransactionManager] - processReceiveMessage input 0<>128 : Message: class=ApplicationCommandHandler[4], type=Request[0], dest=7, callback=0, payload=00 07 09 60 0D 00 01 5B 03 12 00 04 
2020-01-28 02:31:41.811 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Received msg (0): Message: class=ApplicationCommandHandler[4], type=Request[0], dest=7, callback=0, payload=00 07 09 60 0D 00 01 5B 03 12 00 04 
2020-01-28 02:31:41.818 [DEBUG] [nal.protocol.ZWaveTransactionManager] - lastTransaction null
2020-01-28 02:31:41.818 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 0D 00 04 00 07 07 60 0D 04 01 20 01 00 BF 
2020-01-28 02:31:41.823 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 7: Application Command Request (ALIVE:ASSOCIATIONS)
2020-01-28 02:31:41.828 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 7: Decapsulating COMMAND_CLASS_MULTI_CHANNEL
2020-01-28 02:31:41.829 [DEBUG] [nal.protocol.ZWaveTransactionManager] - processReceiveMessage input 0<>128 : Message: class=ApplicationCommandHandler[4], type=Request[0], dest=7, callback=0, payload=00 07 07 60 0D 04 01 20 01 00 
2020-01-28 02:31:41.832 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 7: Incoming command class COMMAND_CLASS_CENTRAL_SCENE, endpoint 0
2020-01-28 02:31:41.836 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 7: SECURITY NOT required on COMMAND_CLASS_CENTRAL_SCENE
2020-01-28 02:31:41.841 [DEBUG] [tocol.commandclass.ZWaveCommandClass] - NODE 7: Received COMMAND_CLASS_CENTRAL_SCENE V2 CENTRAL_SCENE_NOTIFICATION
2020-01-28 02:31:41.842 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 0E 00 04 00 07 08 60 0D 04 01 26 01 00 FF 4A 
2020-01-28 02:31:41.845 [DEBUG] [dclass.ZWaveCentralSceneCommandClass] - NODE 7: Received scene 4 at key 0 [Single Press]
2020-01-28 02:31:41.850 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 7: Got an event from Z-Wave network: ZWaveCommandClassValueEvent
2020-01-28 02:31:41.851 [DEBUG] [nal.protocol.ZWaveTransactionManager] - processReceiveMessage input 1<>127 : Message: class=ApplicationCommandHandler[4], type=Request[0], dest=7, callback=0, payload=00 07 08 60 0D 04 01 26 01 00 FF 
2020-01-28 02:31:41.855 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 7: Got a value event from Z-Wave network, endpoint=0, command class=COMMAND_CLASS_CENTRAL_SCENE, value=4.0
2020-01-28 02:31:41.860 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 7: Updating channel state zwave:device:16866261:node7:scene_number to 4.0 [DecimalType]
2020-01-28 02:31:41.867 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 7: Commands processed 1.
2020-01-28 02:31:41.874 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 7: Checking command org.openhab.binding.zwave.internal.protocol.ZWaveCommandClassPayload@1040d6d.
2020-01-28 02:31:41.886 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Transaction completed - outstandingTransactions 0
2020-01-28 02:31:41.891 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Transaction completed - outstandingTransactions 0
2020-01-28 02:31:41.897 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Received msg (1): Message: class=ApplicationCommandHandler[4], type=Request[0], dest=7, callback=0, payload=00 07 07 60 0D 04 01 20 01 00 
2020-01-28 02:31:41.900 [DEBUG] [nal.protocol.ZWaveTransactionManager] - lastTransaction null
2020-01-28 02:31:41.903 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 7: Application Command Request (ALIVE:ASSOCIATIONS)
2020-01-28 02:31:41.908 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 7: Decapsulating COMMAND_CLASS_MULTI_CHANNEL
2020-01-28 02:31:41.910 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 7: Incoming command class COMMAND_CLASS_BASIC, endpoint 4
2020-01-28 02:31:41.912 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 7: SECURITY NOT required on COMMAND_CLASS_BASIC
2020-01-28 02:31:41.914 [DEBUG] [tocol.commandclass.ZWaveCommandClass] - NODE 7: Received COMMAND_CLASS_BASIC V0 BASIC_SET
2020-01-28 02:31:41.916 [DEBUG] [.commandclass.ZWaveBasicCommandClass] - NODE 7: Basic report, value = 0
2020-01-28 02:31:41.917 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 7: Got an event from Z-Wave network: ZWaveCommandClassValueEvent
2020-01-28 02:31:41.919 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 7: Got a value event from Z-Wave network, endpoint=4, command class=COMMAND_CLASS_BASIC, value=0
2020-01-28 02:31:41.921 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 7: Commands processed 1.
2020-01-28 02:31:41.922 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 7: Checking command org.openhab.binding.zwave.internal.protocol.ZWaveCommandClassPayload@a97ae3.
2020-01-28 02:31:41.924 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Transaction completed - outstandingTransactions 0
2020-01-28 02:31:41.926 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Transaction completed - outstandingTransactions 0
2020-01-28 02:31:41.928 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Received msg (0): Message: class=ApplicationCommandHandler[4], type=Request[0], dest=7, callback=0, payload=00 07 08 60 0D 04 01 26 01 00 FF 
2020-01-28 02:31:41.930 [DEBUG] [nal.protocol.ZWaveTransactionManager] - lastTransaction null
2020-01-28 02:31:41.931 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 7: Application Command Request (ALIVE:ASSOCIATIONS)
2020-01-28 02:31:41.933 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 7: Decapsulating COMMAND_CLASS_MULTI_CHANNEL
2020-01-28 02:31:41.934 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 7: Incoming command class COMMAND_CLASS_SWITCH_MULTILEVEL, endpoint 4
[... cut ...]
2020-01-28 02:31:41.945 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 7: Checking command org.openhab.binding.zwave.internal.protocol.ZWaveCommandClassPayload@113be0.
2020-01-28 02:31:41.946 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Transaction completed - outstandingTransactions 0
2020-01-28 02:31:41.948 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Transaction completed - outstandingTransactions 0
2020-01-28 02:31:41.949 [DEBUG] [nal.protocol.ZWaveTransactionManager] - ZWaveReceiveThread queue empty
2020-01-28 02:31:41.950 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Transaction SendNextMessage 0 out at start. Holdoff false.
```

frejos, 2020-04-18:
I recently added the WallMote and see this same issue, everything seems to be working I'm not sure what value is getting dropped, but looking at the code it is just returning null instead of the integer value for whichever channel it is trying to update. It would be great if the warning had more information like the item/channel name and value. 

The surface issue is that there is no case statement for StringType in the conversion, there doesn't seem to be any practical reason not to implement a conversion of an integer value to a StringType... which would presumably resolve this issue.

frejos, 2020-04-18:
I made the change I suggested above locally and indeed it does resolve the issue, but I'm not sure the "corrected" behavior is desired. I also added some extra logging so I could see exactly what was being updated.

2020-04-18 16:22:22.744 [INFO ] [.smarthome.model.script.frejos.house] - Wall mote [MainWallMoteScene] recieved update [1.0]
2020-04-18 16:22:22.835 [WARN ] [erter.ZWaveMultiLevelSwitchConverter] - Using added StringType conversion for value [0] target channel [zwave:device:088d62c3:node29:switch_startstop1]
2020-04-18 16:22:22.838 [INFO ] [.smarthome.model.script.frejos.house] - Wall mote [MainWallMoteSwipe1] recieved update [0]
2020-04-18 16:22:24.520 [INFO ] [.smarthome.model.script.frejos.house] - Wall mote [MainWallMoteScene] recieved update [1.0]
2020-04-18 16:22:24.617 [WARN ] [erter.ZWaveMultiLevelSwitchConverter] - Using added StringType conversion for value [100] target channel [zwave:device:088d62c3:node29:switch_startstop1]
2020-04-18 16:22:24.619 [INFO ] [.smarthome.model.script.frejos.house] - Wall mote [MainWallMoteSwipe1] recieved update [100]

This is the log from two sequential presses of the button 1 on the WallMote. Apparently, pressing one of the 4 buttons triggers both an update to the :scene_number channel and sets the corresponding :switch_startstopN channel to alternating 0 or 100. I think it was likely intended that the :switch_startstopN channel be a PercentType (Dimmer) instead of a String, but swiping up or down currently sets the :switch_startstopN channel to a JSON string indicating direction, instead of increasing/decreasing this value between 0 and 100. I think it is likely that this is just not "completely" implemented "correctly" for this device. The JSON string implementation seems to have been a hack to me.

dastrix80, 2021-09-30:
I too have this issue, but its quite odd because my wall mote has been working well for over 1 and now suddenly it starts!

dastrix80, 2021-09-30:
> I made the change I suggested above locally and indeed it does resolve the issue, but I'm not sure the "corrected" behavior is desired. I also added some extra logging so I could see exactly what [...]

Can you elborate on the fix?

cdjackson, 2021-10-02:
This warning can be ignored. It is caused by the StartStop channel incorrectly being defined as a String. I've now changed this to StartStopType, and the warning will then not be produced.

Closed by https://github.com/openhab/org.openhab.binding.zwave/pull/1664

closed 2021-10-02 by cdjackson

dastrix80, 2023-09-26:
Im on 3.4.5 OH and im still getting this error?