openhab/org.openhab.binding.zwave #1320: [bug] Could not register ThingType: zwave:aeotec_zw141_03_000

kubawolanin, 2020-04-19:
* OH2 Version: openHAB 2.5.4 Build #86
* Binding Version: 2.5.4.202004110848
---
Steps to reproduce:

1. Log into openHAB console
1. Find ZWave bundle ID and restart it
```
openhab> bundle:list | grep Wave
255 │ Active │  80 │ 2.5.4.202004110848      │ openHAB Add-ons :: Bundles :: ZWave Binding
openhab> bundle:restart 255
```
3. Observe openhab logs (in case you're using openHABian, it'll be `/var/log/openhab2/openhab.log`)

I don't own any device with ThingType `zwave:aeotec_zw141_03_000`. Quick googling tells me it's 
"Aeotec ZW141 Nano Shutter".
I get this error anyway:

```
2020-04-19 10:36:05.259 [ERROR] [ng.xml.internal.ThingTypeXmlProvider] - Could not register ThingType: zwave:aeotec_zw141_03_000

java.lang.IllegalArgumentException: ID segment ' switch_dimmer' contains invalid characters. Each segment of the ID must match the pattern [A-Za-z0-9_-]*.

	at org.eclipse.smarthome.core.common.AbstractUID.validateSegment(AbstractUID.java:97) ~[bundleFile:?]

	at org.eclipse.smarthome.core.common.AbstractUID.<init>(AbstractUID.java:75) ~[bundleFile:?]

	at org.eclipse.smarthome.core.common.AbstractUID.<init>(AbstractUID.java:49) ~[bundleFile:?]

	at org.eclipse.smarthome.core.thing.UID.<init>(UID.java:48) ~[bundleFile:?]

	at org.eclipse.smarthome.core.thing.type.ChannelTypeUID.<init>(ChannelTypeUID.java:40) ~[bundleFile:?]

	at org.eclipse.smarthome.core.thing.xml.internal.ChannelXmlResult.toChannelDefinition(ChannelXmlResult.java:135) ~[bundleFile:?]

	at org.eclipse.smarthome.core.thing.xml.internal.ThingTypeXmlResult.toChannelDefinitions(ThingTypeXmlResult.java:98) ~[bundleFile:?]

	at org.eclipse.smarthome.core.thing.xml.internal.ThingTypeXmlResult.getBuilder(ThingTypeXmlResult.java:148) ~[bundleFile:?]

	at org.eclipse.smarthome.core.thing.xml.internal.ThingTypeXmlResult.toThingType(ThingTypeXmlResult.java:156) ~[bundleFile:?]

	at org.eclipse.smarthome.core.thing.xml.internal.ThingTypeXmlProvider.addingFinished(ThingTypeXmlProvider.java:148) [bundleFile:?]

	at org.eclipse.smarthome.config.xml.osgi.XmlDocumentBundleTracker.addingFinished(XmlDocumentBundleTracker.java:265) [bundleFile:?]

	at org.eclipse.smarthome.config.xml.osgi.XmlDocumentBundleTracker.parseDocuments(XmlDocumentBundleTracker.java:424) [bundleFile:?]

	at org.eclipse.smarthome.config.xml.osgi.XmlDocumentBundleTracker.processBundle(XmlDocumentBundleTracker.java:398) [bundleFile:?]

	at org.eclipse.smarthome.config.xml.osgi.XmlDocumentBundleTracker.access$6(XmlDocumentBundleTracker.java:393) [bundleFile:?]

	at org.eclipse.smarthome.config.xml.osgi.XmlDocumentBundleTracker$2.run(XmlDocumentBundleTracker.java:363) [bundleFile:?]

	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511) [?:1.8.0_222]

	at java.util.concurrent.FutureTask.run(FutureTask.java:266) [?:1.8.0_222]

	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$201(ScheduledThreadPoolExecutor.java:180) [?:1.8.0_222]

	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:293) [?:1.8.0_222]

	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149) [?:1.8.0_222]

	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624) [?:1.8.0_222]

	at java.lang.Thread.run(Thread.java:748) [?:1.8.0_222]
```

kubawolanin, 2020-04-19:
I think I found the root cause:

https://github.com/openhab/org.openhab.binding.zwave/blob/986f96a18cc81496f461257c6dcdac97a3b484f8/src/main/resources/ESH-INF/thing/aeotec/zw141_3_0.xml#L17

channel IDs for `blind_control` and `switch_dimmer` contain spaces

cdjackson, 2020-04-19:
I think this is already fixed. I'm not sure how old build 86 is (the current ZWave build is 29) so I'd suggest to ensure you have the latest build.

If it's not already fixed, then it will be fixed in the next database update (tonight).

cdjackson, 2020-04-19:
I just checked - this was fixed 5 days ago, so I think it might not be in the latest build but will be included in tonights build.

kubawolanin, 2020-04-19:
Many thanks for the prompt reply. 
I'll close this issue then.

closed 2020-04-19 by kubawolanin

cdjackson, 2020-04-19:
Thanks Kuba.

The database exporter was also improved to trim whitespace so this hopefully won’t happen in future :)