openhab/org.openhab.binding.zwave #1324: NEO Flood sensor,cannot select multiple nodes to Basic_Set association

NikolayKash, 2020-04-24:
* OH2 Version: 2.5.4

If the problem is associated with a device, please provide the following -:

* Device [name:]  Shenzhen Neo Electronics Co., Ltd  Flood Sensor 
* Device version: 2.54
* Link to [database] (http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list): https://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-list/devicesummary/493
* If there is an XML being generated for the device (in the /userdata/zwave folder), please provide it.
https://drive.google.com/file/d/1n1hPBhxszvvwvxQoxP0efiBbGWxe4x_X/view?usp=sharing

There is no multiple node section, and I can select only 1 node to send basic set command. 

https://github.com/openhab/org.openhab.binding.zwave/blob/6e1c7d3742f13e4df8781cf906e39f72e17acfde/src/main/resources/ESH-INF/thing/shenzhen/floodsensor_0_0.xml#L131

Thank you.

cdjackson, 2020-04-24:
I'm afraid I don;t know what a "multiple node section" is.  You reference the associations - but the lifeline should (will) be configured and this should be all that is needed to notify OH of any commands.

Please can you better describe the issue, or better still, please discuss on the forum where others with the device may also assist you.

NikolayKash, 2020-04-24:
Hi @cdjackson.

Thanks for your reply.    I  posted a link to association 2 group and this is basic_set  group

[image]

I think on group_2 there should be <multipleLimit>5</multipleLimit>  node.

as I can select only one node for group_2 while it is used by a sensor to sent basic_set command which turn off  a valve.

cdjackson, 2020-04-24:
Thanks. I've now updated the database.

closed 2020-04-24 by cdjackson