openhab/org.openhab.binding.zwave #1331: Could not register ThingType: zwave:aeotec_zw141_03_000

lolodomo, 2020-05-10:
Here is an error in the logs when starting the Z-Wave binding (OH 2.5.4):
```
12:02:06.218 [ERROR] [ing.xml.internal.ThingTypeXmlProvider] - Could not register ThingType: zwave:aeotec_zw141_03_000
java.lang.IllegalArgumentException: ID segment ' switch_dimmer' contains invalid characters. Each segment of the ID must match the pattern [A-Za-z0-9_-]*.
        at org.eclipse.smarthome.core.common.AbstractUID.validateSegment(AbstractUID.java:97) ~[bundleFile:?]
        at org.eclipse.smarthome.core.common.AbstractUID.<init>(AbstractUID.java:75) ~[bundleFile:?]
        at org.eclipse.smarthome.core.common.AbstractUID.<init>(AbstractUID.java:49) ~[bundleFile:?]
        at org.eclipse.smarthome.core.thing.UID.<init>(UID.java:48) ~[bundleFile:?]
        at org.eclipse.smarthome.core.thing.type.ChannelTypeUID.<init>(ChannelTypeUID.java:40) ~[bundleFile:?]
        at org.eclipse.smarthome.core.thing.xml.internal.ChannelXmlResult.toChannelDefinition(ChannelXmlResult.java:135) ~[bundleFile:?]
        at org.eclipse.smarthome.core.thing.xml.internal.ThingTypeXmlResult.toChannelDefinitions(ThingTypeXmlResult.java:98) ~[bundleFile:?]
        at org.eclipse.smarthome.core.thing.xml.internal.ThingTypeXmlResult.getBuilder(ThingTypeXmlResult.java:148) ~[bundleFile:?]
        at org.eclipse.smarthome.core.thing.xml.internal.ThingTypeXmlResult.toThingType(ThingTypeXmlResult.java:156) ~[bundleFile:?]
        at org.eclipse.smarthome.core.thing.xml.internal.ThingTypeXmlProvider.addingFinished(ThingTypeXmlProvider.java:148) [bundleFile:?]
        at org.eclipse.smarthome.config.xml.osgi.XmlDocumentBundleTracker.addingFinished(XmlDocumentBundleTracker.java:265) [bundleFile:?]
        at org.eclipse.smarthome.config.xml.osgi.XmlDocumentBundleTracker.parseDocuments(XmlDocumentBundleTracker.java:424) [bundleFile:?]
        at org.eclipse.smarthome.config.xml.osgi.XmlDocumentBundleTracker.processBundle(XmlDocumentBundleTracker.java:398) [bundleFile:?]
        at org.eclipse.smarthome.config.xml.osgi.XmlDocumentBundleTracker.access$6(XmlDocumentBundleTracker.java:393) [bundleFile:?]
        at org.eclipse.smarthome.config.xml.osgi.XmlDocumentBundleTracker$2.run(XmlDocumentBundleTracker.java:363) [bundleFile:?]
        at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511) [?:1.8.0_152]
        at java.util.concurrent.FutureTask.run(FutureTask.java:266) [?:1.8.0_152]
        at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$201(ScheduledThreadPoolExecutor.java:180) [?:1.8.0_152]
        at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:293) [?:1.8.0_152]
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149) [?:1.8.0_152]
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624) [?:1.8.0_152]
        at java.lang.Thread.run(Thread.java:748) [?:1.8.0_152]
```
It looks like there is an unexpected space in the id ?

I have this error even if I don't have this kind of devices (aeotec_zw141_03_000).

cdjackson, 2020-05-10:
This issue has been closed already - please use a newer binding.

cdjackson, 2020-05-10:
Closed by #1320

closed 2020-05-10 by cdjackson

brianwarner, 2020-05-16:
I think this might be in the current stable binding, I installed OH2.5 stable last night and am getting the error.  Is it safe-ish to use the [3.0 snapshot binding currently on ci.openhab.org](https://ci.openhab.org/job/openHAB-ZWave/) with 2.5 stable?  And, this may be a dumb question, but what is the proper way to replace the stable binding with the snapshot?

cdjackson, 2020-05-16:
No - do not use the 3.0 binding. Use the 2.5.5 snapshot binding.

brianwarner, 2020-05-16:
Thanks @cdjackson!