openhab/org.openhab.binding.zwave #1343: Removal of dependency on 'org.apache.commons.*'

bodiroga, 2020-06-02:
As stated in https://github.com/openhab/openhab-addons/issues/7722, openHAB will no longer be shipped with org.apache.commons library, so the Z-Wave binding should remove all the library appearances for OH3.

I'm not sure if I could do it, but I'm creating the issue just as a reminder. In any case, what is going to be the branch strategy for OH3? The development and master branches are behind 2.5.x branch :confused: 

Thank you!

bodiroga, 2020-06-03:
As far as I know, the binding only uses two functions from the Apache Commons library:

- org.apache.commons.lang.ArrayUtils: [here](https://github.com/openhab/org.openhab.binding.zwave/blob/7cde8454169c45fa819497c6b203174b99b8060a/src/main/java/org/openhab/binding/zwave/internal/protocol/serialmessage/GetVersionMessageClass.java#L46) and [here](https://github.com/openhab/org.openhab.binding.zwave/blob/7cde8454169c45fa819497c6b203174b99b8060a/src/main/java/org/openhab/binding/zwave/internal/protocol/SerialMessage.java#L161)
- org.apache.commons.io.IOUtils: [here](https://github.com/openhab/org.openhab.binding.zwave/blob/59698f507b65ca67770850a3347f4c7ac3917f9d/src/main/java/org/openhab/binding/zwave/handler/ZWaveSerialHandler.java#L225)

Both can be easily replaced:

- For ArrayUtils, [here](https://www.techiedelight.com/get-subarray-array-specified-indexes-java/) we have different possible implementations.
- For IOUtils.closeQuietly(), @lolodomo has provided several PR in the openhab-addons repository replacing the function with a simple ([example](https://github.com/openhab/openhab-addons/pull/7739/files)):

```java
try {
    inputStream.close();
} catch (IOException e) {
    logger.debug("Error while closing the input stream: {}", e.getMessage());
}
```

@cdjackson, I can provide a PR for this, but first it should be decided if a new branch is going to be created for OH3 :wink:

cdjackson, 2020-06-03:
I don't want to create an OH3 branch just yet as I want to avoid managing two branches, but yes, at some point in the near future there will need to be one.

bodiroga, 2020-06-03:
Ok, I will wait then :+1:

cdjackson, 2020-07-05:
@bodiroga thinking about this further - there's no harm in removing any references now in the 2.5 branch if you like... This doesn't need to wait for OH3.

bodiroga, 2020-07-14:
I will try to have a look at it in the following weeks, thanks @cdjackson!

PR #1389 (2020-08-26): Remove dependency on 'org.apache.commons.*'
Hi @cdjackson!

As promised in #1343, I have removed all references to 'org.apache.commons.*' in line with https://github.com/openhab/openhab-addons/issues/7722.

Closes #1343
closed 2020-08-26 by cdjackson