openhab/org.openhab.binding.zwave #1350: Add support for COMMAND_CLASS_BATTERY to yale/yrd110_0_0.xml

adrianboimvaser, 2020-07-02:
According to this manual:

> The lock supports Battery CC per ZWave specifications V9. It is recommended that the
 network controller use this CC to retrieve the lock’s battery level at least once daily. The
 lock will send low battery alarms if low battery condition exists immediately after user access.

https://www.cd-jackson.com/zwave_device_uploads/302/Yale-Real-Living-300-ZWave-CC-Specs-10OCT12.pdf

I need it in order to be able to read battery level.

Thank you.

adrianboimvaser, 2020-07-02:
It's supported for `yale/yrd120_0_0.xml` and other yale locks. If it's possible to change the device type in openHAB (instructions please?) I may be able to get it to work while I wait for this to be addressed.

cdjackson, 2020-07-02:
This is already supported. All devices that support the battery command class have the battery state channel available. If you have an issue with this please discuss on the forum as there must be something else going on with your system.

closed 2020-07-02 by cdjackson

adrianboimvaser, 2020-07-02:
@cdjackson, I'm sorry, I'm a little lost, I can't find `COMMAND_CLASS_BATTERY` in https://github.com/openhab/org.openhab.binding.zwave/blob/2.5.x/src/main/resources/ESH-INF/thing/yale/yrd110_0_0.xml like in https://github.com/openhab/org.openhab.binding.zwave/blob/2.5.x/src/main/resources/ESH-INF/thing/yale/yrd120_0_0.xml
Am I looking at the wrong branch?

cdjackson, 2020-07-02:
As discussed on the forum, I’ve updated the database to add this already.