openhab/org.openhab.binding.zwave #1368: apparently the Zwave binding blocks the / dev / ttyUSB0 port in Combination with a CC2652RB Zigbee2mqtt Dongle

bifi2090, 2020-08-04:
Hello Guys,

I`m using the Zwave-Binding since 2 years and it´s great. Also i´m using Zigbee2mqtt (zigbee2mqtt.io) with a CC2531 Coordinator. Both are running great. 

SOme Days ago i bought a new CC2652RB Coordinator (more Power, more controllable Zigbee-Devices), see https://slae.sh/projects/cc2652/.

The old CC2531-Coordinator come´s up on a /dev/ttyACM*-Port. 
The same applies for my Aeotec Z-Stick. With appropriate UDEV-Rules I pointed the both Controllers to:

- CC2531 (Zigbee2mqtt-Stick) --> /dev/ttyUSB-z2m

- Aeotec- Zwave-Stick --> /dev/ttyUSB1

With this Configuration all both networks are working. Now i decided to change the CC2531 to the CC2652RB Stick. With it´s Onboard Bootloader (UART) this Stick now connects by default to the serial-Port /dev/ttyUSB0. 

Now the Problem comes up. if the Zwave-Binding is active, although the Zwave-Stick is mapped to /dev/ttyUSB1, this Binding seems to Block the /dev/ttyUSB0 - Port further. Even if I Map the CC2652RB to  another Port, for example /dev/ttyUSB33 via UDev-Rules and "Java-Opts" I can´t get Zigbee2mqtt running. Only if I stop the Zwave Binding it is usable. So I think the Zwave-Binding blocks in a certain way the Access on Port /dev/ttyUSB0. 

I Hope some can help me.

Best regards 

* OH2 Version: 2.5.7
* Binding Version: 2.5.7

cdjackson, 2020-08-04:
It might be worth discussing this on the community forum. The binding doesn’t directly access serial ports - they go through a middle layer - possibly this is causing an issue. If you discuss on the forum you will have a wider audience and get the benefit of difference experiences.

bifi2090, 2020-08-04:
Thx for the Info. I will take this issue to the forum. Beat regards

bifi2090, 2020-08-19:
Solution for this Issue:

https://community.openhab.org/t/apparently-the-zwave-binding-blocks-the-dev-ttyusb0-port-in-combination-with-a-cc2652rb-zigbee2mqtt-dongle/103245/63?u=bifi2090

Closing now

Best regards

closed 2020-08-19 by bifi2090