openhab/org.openhab.binding.zwave #1372: Philio PAN04 not updating values of specific endpoints automatically

DataWorm, 2020-08-08:
Device: Philip (013c) PAN04 (0001:0012), Firmware Version 1.4
OH2 Version: 2.5.7

Some info about the device:
The PAN04 relay switch includes two endpoints/instances/relays that I can switch to turn my lamps on/off. There are actually 4 endpoints. Endpoint zero is basically a combination of both endpoints (turns both relays on or off and is set to true if at least one switch is turned on and provides energy measurements for both endpoints together...). Not sure about the third endpoint but as far as I know it also combines the two physical relays somehow. The first and second endpoint are each related to one of both relays and those are the endpoints I actually want to use for my items.

My Problem:
I don't receive any updates for first or second endpoint but after a lot of tests it turns out that I get feedback for endpoint zero. So when I turn on one relay manually then the corresponding item for endpoint #1 never gets updated (neither switch state nor energy metrics). However the combined information on endpoint #0 receives an update. But I would also like to know which lamp is switched on or off. In Z-Way it worked just fine so it doesn't seem to be a problem with the device itself. I HABmin I can go to the thing and use the button at the end of the description view to update the items. After clicking that item the values of all items (including endpoint #1 and #2) get updated. So retrieving those values does obviously work and just need to be updated automatically.

I also checked the association groups. I have entries for "1: Relay 1 + 2", "2: Relay 1", "3: Relay 2". I tried to add/remove the Controller from those fields but somehow the only thing I can actually change is whether Controller is added in "Relay 1 + 2". When I try to remove it for "Relay 1" or "Relay 2" it tells me "in work" but when I reload the page they are added again. Also it does not matter if Controller is added to the "Relay 1 + 2" association group, the behavior is the same. I can't get any automatic updates for endpoint #1/#2. 

Logs doesn't contain anything helpful. Just the value updates for endpoint #0 but it does not show any attempt to get updates for the other endpoints.

cdjackson, 2020-08-08:
I think that these devices have a problem with associations in that they don't encapsulate using the multi-channel encap. Therefore reports always come from the root endpoint.

I'd suggest to do a search on the forum and you will probably find some comments about this. This has been an issue for quite a while - maybe you have a new version - I'm not sure if they have ever fixed this.

https://community.openhab.org/t/oh2-z-wave-refactoring-and-testing-and-security/21653/2648

https://community.openhab.org/t/tkb-tz06-in-wall-relay-switches-not-updating-state-changes/58915/8

https://community.openhab.org/t/configuring-pan06/94046

DataWorm, 2020-08-08:
Thanks for the resources. I saw some similar posts but since the device provides associations for "Relay 1" in theory and since everything worked fine in zway I thought it must be another issue. But maybe zway just made a workaround for that or so. Maybe they query all endpoints once one of them get updated?! Apparently several of those devices from that manufacturer have such problems. Maybe it is worth to add such a workaround into this binding?

mbronk, 2020-10-21:
AFAIK _(don't recall the source, but I am pretty positive I saw it)_... 
- This _seems_ to be a known bug in the Philio PAN04 firmware _(some distant reference: https://github.com/domoticz/domoticz/issues/630)_
- ZWay&others might indeed have applied a **workaround** _(I saw it in code somewhere, but can't recall if this was zway of sth else. In any case, the w/a was to use polling, AFAIR)_.

BTW. The same w/a can be done in OH, using the following scheme:
1. Listen on `:meter_watts` channel (note the nature of this bug is that it will receive alternate updates... but from either Relay1 or Relay2)
2. When an update is received on `:meter_watts`, send a `RefreshType.REFRESH` to the other channels (specifically: `meter_watts1`, `meter_watts2`). This will update both Relay states (btw. same works for `switch_binaryX` channels)

@cdjackson  - It would be nice, if the w/a could be implemented directly inside the binding. Do you think such '_device erratas_' can be implemented (similarly to what other automation solutions seem to be doing)?

----
FYI - I have the following rule running specifically for that purpose (it is event-triggered, so much more responsive than a regural polling):

``` java
import org.eclipse.xtext.xbase.lib.Functions

val Functions$Function2<NumberItem, Object, Boolean> shouldRefreshPan04 = [
    item, previousState |
        var double offset = 1.5 //this is to debounce the results and avoid refresh loop
        if(previousState === null || previousState == NULL || item.state === null) {
            return true
        }
        var previousValue = previousState as Number
        if (previousValue === null) {
            return true
        }
        var double prev = previousValue.doubleValue()
        var double curr = (item.state as Number).doubleValue()
        if( prev < curr - offset || prev > curr + offset) { 
            logDebug("PAN04", "Previous state: {}W, current {}W", prev, curr)
            return true
        }
        false
]

rule "PAN04 Refresh on wattage change (Node7)"
when
    Item Node7TriggerW changed
then
    if(shouldRefreshPan04.apply(Node7TriggerW, previousState)) {
        sendCommand(GNode7, "REFRESH")
    }
end
//Where: Node7TriggerW is an Item linked to ':meter_watts' channel and GNode7 is a Group for all the items defined for this PAN04 device