openhab/org.openhab.binding.zwave #1415: Z-Wave Binding...Issues with Aeotec Z-Stick Gen 5 [Z-Wave]

bd8392, 2020-10-01:
Hi community =),

I've maid some tests with the Aeotec Z-Stick Gen5 and discovered some issues. Here are the main problems.

## Current Bugs

1. Issue with the usb-port on raspberrypi (not resolveable as this is  problem from the aeotec z-stick -> https://www.raspberrypi.org/forums/viewtopic.php?t=245031
2. Devices get automatically added on search to the z-wave controller (You don't need to add as thing, they get automatically added as soon as discovered to the controller, which is not very convenient, especially on the following problem ->
3. Not able to delete z-wave devices from controller (option set as failed and delete from controller don't have effect)
4. Learning mode does not work

To summarize: All advanced options don't seem to work on the aeotec z-stick gen5 with openhab 2.5.7 in docker (privileged) and pi. The controlling works perfect

## Steps to Reproduce (for Bugs)
1. Install openhab in docker on pi with hypriot os and traefik reverse proxy
2. stick in usb-hub and aeotec z-stick (hub to resolve connection issue, see above)
3. setup the z-stick as controller
4. put a z-wave device in inclusion mode
5. start discovering with the inbox on paper-ui
6. see the issues

Sadly I have no Idea how to resolve this issues, but I can help testing, as much as possible! If you need logs or something else, I'll be here =)

For now I get around all issues, by plugging the z-stick to a windows pc, and delete the failed nodes with the silab software...which is very complicated...

## Your Environment
Raspberrypi 4 B+
hypriot os 1.12.3
docker: 19.0.3
openhab-docker 2.5.7

kaikreuzer, 2020-10-01:
Sure, done!

cdjackson, 2020-10-01:
I would really suggest to discuss this issue on the forum. I don't think that there are necessarily binding issues here - for example, the fact that the RPi can't resolve the USB port is not something the binding can resolve I think?

If devices are automatically added to teh controller, then this is simply how ZWave works - the binding cannot control this as it is related to the way the protocol works - not the binding.

It can be difficult to remove failed devices - this is true. Again, I'd suggest to discuss this on the forum as there are a number of approaches to do this.

Learning mode does work - otherwise no-one would be able to add any devices.

Again - I'd strongly suggest to discuss the issues on the forum where you can get help from a large number of people - I don't think there are binding bugs?

bd8392, 2020-10-01:
> Learning mode does work - otherwise no-one would be able to add any devices.

Sorry I meant learning from another suc controller, so OpenHab can be the secondary controller...(it's under advanced options)

> I would really suggest to discuss this issue on the forum.

I can file a theme there if you want :-) 

> I would really suggest to discuss this issue on the forum. I don't think that there are necessarily binding issues here - for example, the fact that the RPi can't resolve the USB port is not something the binding can resolve I think?

It's really hard to say...the controlling and rules works more than perfect! I can provide any log you need to help resolve or diagnose :-). I haven't tested it on another machine yet, but the stick gets recognized probably by the pi.../tty/acm0 or something like shows up...

cdjackson, 2020-10-01:
> Sorry I meant learning from another suc controller

This should work, but it's not something that I would look to spend time on at the moment as it is really uncommon for this to be used and therefore isn't something I can justify spending time on.

> It's really hard to say...the controlling and rules works more than perfect! I can provide any log you need to help resolve or diagnose :-). I haven't tested it on another machine yet, but the stick gets recognized probably by the pi.../tty/acm0 or something like shows up...

Many people use the binding with an RPi, so I'm sure that it works. There are problems with Pi USB ports and a hub is recommended. Again, discussing this on the forum is likely a lot more productive than discussing here as I don't use a Pi at all.

> I can file a theme there if you want :-)

Yes please - there are a lot more people that will be able to help there.  Thanks.

closed 2020-10-01 by cdjackson

bd8392, 2020-10-01:
> is really uncommon

Currently it's the only way to "cluster" multiple OpenHab instances for failover..at least in z-wave I think...but yes I also consider it rather uncommon at the moment...just for the record; to add the stick manually to another z-wave controller does not work for security-devices; (tested with Zipabox https://community.openhab.org/t/gradual-migration-from-zipabox-primary-and-secondary-controllers-etc/77525)

> USB ports and a hub is recommended.

Yes you HAVE TO use a hub to make Z-Stick gen5 even working...also discrbed here and above https://www.raspberrypi.org/forums/viewtopic.php?t=245031

> Many people use the binding with an RPi, so I'm sure that it works

Yes all basic-functionallity works fine, the only problems are with advanced z-wave and inclusion-functions...(like forget a device.

> Yes please - there are a lot more people that will be able to help there. Thanks.

Done here :-): https://community.openhab.org/t/z-wave-binding-issues-with-aeotec-z-stick-gen-5-z-wave-1415/105850

cdjackson, 2020-10-01:
> Currently it's the only way to "cluster" multiple OpenHab instances for failover..at least in z-wave I think..

Many devices will NOT work well with multiple controllers. A lot of devices will only support a single controller being added to the lifeline association group, so adding multiple controllers will typically result in a reduced system.

bd8392, 2020-10-01:
> Many devices will NOT work well with multiple controllers.

Okay good to know cause I also made this experience here...-> reaction times become pretty low

By the way; compared to Zipabox the OpenHab Z-Wave Binding is not only 20 times faster when controlling devices, but also extremely reliable in measures of controlling by rules so far! I love the z-wave binding in OpenHab! Never seen faster reactiontimes. OpenHab is great in general -> best and most versatile smarthome-system Ive seen so far.