openhab/org.openhab.binding.zwave #1419: PaperUI HTTP 500 when edit zwave thing

hoffe86, 2020-10-04:
hey all,
when I try to edit a z-wave thing via the PaperUI a HTTP 500 occur with following error:

`{"error":{"message":"org.openhab.binding.zwave.handler.ZWaveSerialHandler cannot be cast to org.openhab.binding.zwave.handler.ZWaveControllerHandler","http-code":500,"exception":{"class":"java.lang.ClassCastException","message":"org.openhab.binding.zwave.handler.ZWaveSerialHandler cannot be cast to org.openhab.binding.zwave.handler.ZWaveControllerHandler","localized-message":"org.openhab.binding.zwave.handler.ZWaveSerialHandler cannot be cast to org.openhab.binding.zwave.handler.ZWaveControllerHandler"}}}`

Not sure if its a problem with my installation or bug.

openhab-5iver, 2020-10-05:
When this is reported in the forum, it is usually due to having two zwave bindings installed. Run...

```
list -s|grep zwave
```
... in the console to see if this is the case.

hoffe86, 2020-10-05:
@openhab-5iver yes it seems there are two bindings

openhab> list -s|grep zwave
203 x Active x  80 x 2.5.2.202002050516      x org.openhab.binding.zwave
302 x Active x  80 x 2.5.9                   x org.openhab.binding.zwave

how can I remove one of these bindings?

openhab-5iver, 2020-10-06:
```
bundle:uninstall 203
```
This will remove the older one, but make sure the jar file is also removed from $OPENHAB_HOME/addons/.

hoffe86, 2020-10-08:
okay this fixed the problem.

closed 2020-10-08 by hoffe86