openhab/org.openhab.binding.zwave #1443: UFairy G.R. Tech - GR-105 valve no longer recognized by the binding

SebastianCornea, 2020-11-18:
* OH2 Version: 2.5.10 Release build
* Device name: GR-105 Valve
* Device version: 0152:0003:0512:0.2
* Link to [database] https://opensmarthouse.org/zwavedatabase/438

The device is no longer recognized by openhab, it appears as unknow node "Z-Wave Node 028 (0152:0003:0512:0.2)". It seems that there is some issue with the device DB as on OpenSmartHouse the device definition contains errors. 

Is there a way to quick fix this on my setup?

cdjackson, 2020-11-18:
I'm not sure why this hasn't exported. There are two versions in the database (0-10, and 10 upward) but for some reason it seems that the lower version (as you have) is not exporting.,

SebastianCornea, 2020-11-18:
Can I fix it manually on my setup somehow?

SebastianCornea, 2020-11-18:
It seems that Scott Rushworth just fixed the DB entry.

cdjackson, 2020-11-18:
Ok, I guess you have another thread open somewhere - please avoid cross posting issues in multiple places.

closed 2020-11-18 by cdjackson