openhab/org.openhab.binding.zwave #1458: [next.openhab.org] Things Summary Page Bad Link

bwosborne2, 2020-12-11:
On next.openhab.org and here
https://github.com/openhab/org.openhab.binding.zwave/blob/master/doc/things.md

the database reference guide is broken (relative) and attempts topoint to cd-jackson

I have not submitted a PR since this may be generated as part of the database export.

Bad:
For information on adding devices to the database, refer to the 
[database information page](http://www.cd-jackson.com/index.php/zwave/zwave-device-database/zwave-device-database-guide).

Fixed:
For information on adding devices to the database, refer to the 
[database information page](//opensmarthouse.org/blog/zwave_device_database_guide).

cdjackson, 2020-12-11:
Thanks Bruce. I think this is generated as part of one of the export scripts - I’ll try and find some time to look at this in the next day or two.

bwosborne2, 2020-12-11:
I thought it might be & I thought I fixed the broken link when I went through the scrips.

There is an Issue in docs that mentioned getting the addons updated too. I think Z-Wave looks OK.

cdjackson, 2020-12-11:
I think you might have, but it’s possible that we just haven’t exported this page into the OH3 repository - it’s not generated every time we export at the moment I think.

cdjackson, 2020-12-11:
I don’t suppose you remember where this file is generated? At the moment I can’t find it anywhere :(  I think there must be a separate php file to generate the index, but I can’t see it anywhere.

bwosborne2, 2020-12-11:
Offhand I thought it was the same script that generated the individual Things pages. I need to check later.

cdjackson, 2020-12-11:
Thanks. I don’t think it can be the same script as that script only generates a single thing - it doesn’t loop through the complete database. It would be far too slow to show the exports in the web page if it had to loop through the whole database (it takes a few minutes to export).

bwosborne2, 2020-12-11:
Thinking further, I had a test script but you said that page was part of the export. My GitHub link was not a clue?

cdjackson, 2020-12-11:
I’m not sure what link you mean regarding the clue? The links don’t relate to the scripts, so I don’t think it helps.

bwosborne2, 2020-12-11:
I meant its location in the repo. Not sure when we moved but the August 9 date for both 2.5 & 3.0 means that page is out of date.

It may not have made it over to your new export process. Just a guess on my part.

cdjackson, 2020-12-11:
Probably it will need to be rewritten then :( I probably won’t have time to look at that in the next couple of weeks but will take a look over Christmas.

bwosborne2, 2020-12-11:
Do you export using the API? I may be able to whip something up.

cdjackson, 2020-12-11:
I’m not sure what api you mean. There needs to be a script to generate the MD index file. This script will use the API to get the data, but somehow something needs to generate the index. It’s not a super complex script though.

bwosborne2, 2020-12-11:
I just wondered whether it was worth me tackling it.  I think we can wait. It is minor compared to other OH documentation issues.