openhab/org.openhab.binding.zwave #1460: Z-Wave Binding from 2.5.1 to 2.5.10 cause periodic reset of Arduino board connected to same device as Z-Wave Controller

novot86, 2020-12-14:
Dear developers,
Im facing with issue for several weeks regarding Z-Wave Binding, question on openhab forum was raised several weeks ago unfortunatelly without success:
https://community.openhab.org/t/z-wave-binding-from-2-5-1-2-5-9-cause-periodic-reset-of-all-devices-connected-through-usb/107699 

**Description:**
Im using RPi 3 with Arduino connected via USB in my projects from 2.4.0. Im using several ZWave devices successfully for two years. From 2.5.1 and higher OpenHAB (and ZWave Binding 2.5.1 or higher) I cannot use connected Arduino. When enable USB Z-Wave Controller, immediately get Arduino reseting every ~ 15 sec via USB. 

I tried install from scratch, issue is reproducible from versions of openhab 2.5.1 - 2.5.10. Im using preinstalled openhabian. 

* OH2 Version: 2.5.1 - 2.5.10
* Binding Version 2.5.1 - 2.5.10

If the problem is associated with a device, please provide the following -:

* Device name: RPi 3 B+
* Controller: Z-Wave.Me USB Smart Home Stick
* Link to [database](http://opensmarthouse.org/zwavedatabase):
* If the device is not in the database, provide a link to the manual.
* If there is an XML being generted for the device (in the /userdata/zwave folder), please provide it.

OpenHAB Log: (Added Controller, added device) Log does not contain new lines but arduino still reconnectiong periodically.
[openhab.log](https://github.com/openhab/org.openhab.binding.zwave/files/5687119/openhab.log)

What helps:
Simply disable Z-Wave Controller and everything works as expected. I tried more Z-Wave Controller devices. Version 2.4.x without any issue.

Question:
Is there any periodic rescan feature? How to debug what is on the background to find which process reset Arduino serial port?

Thank you for help in advance.

Majkee

cdjackson, 2020-12-14:
My guess is that this is related to issues with nrjavaserial. We see a number of problems with this driver and on other projects I’m involved with are moving to other drivers to see if it resolves the problems. Unfortunately there’s not a lot that can be done in the binding I think.

novot86, 2020-12-14:
Thank you for comment. Is there any recommandation for workaround? Is it possible to use ZWave Binding 2.5.0 with OpenHAB 2.5.10? We can w8 with previous version until OH3 and believe it will be changed in future.

cdjackson, 2020-12-14:
I don’t think using an older ZWave binding will help - the serial library not part of the binding - it is part of the core.

I don't think that there is any move to change away from using nrjavaserial - it is also used in OH3 (although possibly a newer version). For custom systems, I would recommend using something else - the only problem is what... You could look at JSSC, but it is licensed under GPL at present.

novot86, 2020-12-14:
Version of nrjavaserial library is the same 3.15 from OH 2.5.0 - 2.5.11. Therefore I though something changes in binding. Tested zigbee usb controller, no problem. Unfortunately, it seems like loop re-scan of serial ports when Z-Wave Controller thing is enabled. I don't know why and it is impossible for me to debug what's happens in the background.

5iver, 2021-01-14:
I suspect you have some other binding set to install that no longer exists. You could post logs or cleanup your addons.cfg and addons.config files, but it may be cleaner to just setup a test environment with a fresh OH install and only the Z-Wave binding and the Arduino. This should be easy to do with a new SD card and openHABian.

cdjackson, 2021-01-14:
I suggest that we close this and discuss further on the forum if needed. I think there are a few such discussions on the forum as there were some bugs in the core that caused bindings to restart if something is misconfigured (I think this is what @5iver is talking about). Additionally there are issues with nrjavaserial when it does port scans it can block ports which may cause the binding to restart the controller handler.

Either way, discussing on the forum is likely to be better as these issues are likely not directly in the binding.

closed 2021-01-14 by cdjackson