openhab/org.openhab.binding.zwave #1476: XStream not initalized

ljsquare, 2020-12-24:
* OH3 Version: 3.0.0 stable docker
* Binding Version (if SNAPSHOT, provide compile time as found on the filename in the console): 3.0.0

At runtime the log shows the error: Security framework of XStream not explicitly initialized, using predefined black list on your own risk.

According to this post: https://community.openhab.org/t/xstream-not-initalized/110879 it would be solved in the stable version 3.0.0, unfortunately it doesn't. It’s probably the unsafe usage of XStream by the Z-Wave binding: https://github.com/openhab/org.openhab.binding.zwave/blob/1bf81ffd0f70cfd293b6f40140db50591a962ade/src/main/java/org/openhab/binding/zwave/internal/protocol/initialization/ZWaveNodeSerializer.java#L46

t09, 2020-12-30:
Don't know it this helps

[image]

gdolfen, 2021-01-05:
+1; same for me on OH3.0.0 stable docker

13rac1, 2021-01-06:
Same here on Docker `3.0.0` (ee578b82ae73) and `3.1.0-SNAPSHOT` (8ea01abf1139) on every startup:

```
Launching the openHAB runtime...
Security framework of XStream not explicitly initialized, using predefined black list on your own risk.

Security framework of XStream not explicitly initialized, using predefined black list on your own risk.
Security framework of XStream not explicitly initialized, using predefined black list on your own risk.
Security framework of XStream not explicitly initialized, using predefined black list on your own risk.
Security framework of XStream not explicitly initialized, using predefined black list on your own risk.
Security framework of XStream not explicitly initialized, using predefined black list on your own risk.
```

BUT the error is logged constantly after power-loss reset. The UI starts, but the Zigbee and Z-Wave controllers do not start. I have to run `sudo docker-compose restart` to make the controllers function.

PR #1500 (2021-01-07): Fix XStream security framework not initialized warnings
Fixes #1476

See also this [community thread](https://community.openhab.org/t/xstream-not-initalized/110879?u=wborn)
commit 0b5dbe96 (2021-01-07): Fix XStream security framework not initialized warnings (#1500)

Fixes #1476
closed 2021-01-07 by cdjackson