openhab/org.openhab.binding.zwave #1485: OH3 and Aeon Labs ZW132

temuccio, 2020-12-30:
* OH3 Version: 3.0.0 stable docker
* Binding Version: 3.0.0

* Device name:AEON Labs ZW132
* Device version: 2.2
* Link to [database](http://opensmarthouse.org/zwavedatabase): https://opensmarthouse.org/zwavedatabase/1017

Hi All, 
I have update my openhab from version 2.5.11 to 3.0.0.
I have make a new configuration.

From OpenHab, I can turn on/off the ZW132 devices.
From ZW132, when I turn on or off from S1 or S2 the openhab doesn't update (switch1 and switch2 channel configured)

If I try to move switch2 to switch, when I turn on or off from S1 or S2 the openhab  update status, but in this case if I turn ON Switch from openhab, the device turn ON S1 and S2.

Do you can help me to solve this issue?

[node7.zip](https://github.com/openhab/org.openhab.binding.zwave/files/5754007/node7.zip)

cdjackson, 2020-12-30:
Probably the lifeline association is not configured. In theory, this should happen automatically.

I would suggest to discuss this on the openhab support forum rather than here as this is more for bug reports than support.

temuccio, 2020-12-30:
Hi @cdjackson , thanks for response.
I have write here because on forum [community](https://community.openhab.org/t/oh3-and-zwave-update-from-device/111537/18) every attemts have not results and because on OH2 this device works fine.

I suppose there is a bug

cdjackson, 2020-12-30:
I think it's unlikely there's a bug - the issue is almost certainly that the associations are not configured. If the association is configured, and the device is not sending reports, then this is a device problem.

temuccio, 2020-12-30:
This is the screen of association.
[image]

I have 4 of this device with same problem (the fibaro works fine)
[image]

cdjackson, 2020-12-30:
As I said the issue is almost certainly that the associations are not configured. If the association is configured, and the device is not sending reports, then this is a device problem.

temuccio, 2020-12-30:
@cdjackson I suppose that the associations are configured (or I see that).
the ZW132 send report not well on OH3 (in I configure channel switch, this send report but channel switch 1 on), but all 4 device ZW132 works fine on OH 2.5.11 (I have OH2 and OH3 in docker container and I can switch one or other).
In this case I exclude that the device has a problem.

cdjackson, 2020-12-30:
> I suppose that the associations are configured 

As I said earlier, they should be configured, but if something went wrong, then maybe not? Have you tried resetting the associations or reinitialising the device? If the device is not reporting, then this is a device problem (sorry to repeat myself).

What do the logs show? Maybe the device is reporting and there is another problem? I guess you have checked this already though?