openhab/org.openhab.binding.zwave #1494: OH3: logic/zdb5100

Roagert, 2021-01-02:
OH Version: 3 release stable

* Device name: Logic Group ZDB5100 
* Device version: HW: 2 FW: 1.01
* Link to [database](http://opensmarthouse.org/zwavedatabase):

Product:
https://products.z-wavealliance.org/products/3399?selectedFrequencyId=1

Issue:
I recognized when using trying to save new values into some configuration parameters i found that it didnt accept the values because of specific limit for that parameter. Although the limit is incorrect see the examples.

From what i can gather then in the first example parameter 23 is incorrect.
second example then parameter 18 and 26 is incorrect.
18 is incorrect in the range, and 26 is incorrect in range and default value.

**First Example:**

_Parameter 23: OFF indication RGB-saturation for pushbutton 1_
Values in the range 0 to 65535 may be set.
The manufacturer defined default value is 0.

This parameter has the configuration ID config_23_4 and is of type INTEGER.

_Parameter 31: OFF indication RGB-saturation for pushbutton 2._
Values in the range 0 to 4294967295 may be set.
The manufacturer defined default value is 791621376.

This parameter has the configuration ID config_31_4 and is of type INTEGER.

**Second Example:**

_Parameter 18: Multi-Level Switch Set values for pushbutton 1_

Values in the range 0 to 33514240 may be set.
The manufacturer defined default value is 33488896.
This parameter has the configuration ID config_18_4 and is of type INTEGER.

_Parameter 26: Multi-Level Switch Set values for pushbutton 2._

Values in the range 0 to 65535 may be set.
The manufacturer defined default value is 0.
This parameter has the configuration ID config_26_4 and is of type INTEGER.

_Parameter 34: Multi-Level Switch Set values for pushbutton 3_

Values in the range 0 to 4294967295 may be set.
The manufacturer defined default value is 33488896.

This parameter has the configuration ID config_34_4 and is of type INTEGER.

The range for all 3 are different

There might be other variables, but this is something i've found which has caused me some headaches!

Thanks.

cdjackson, 2021-01-02:
Thanks for reporting. Please register on the database site, and you can make the required changes directly in the database which will then flow through into the binding.

Thanks.

Roagert, 2021-01-02:
> Thanks for reporting. Please register on the database site, and you can make the required changes directly in the database which will then flow through into the binding. Thanks.

I'm no expert at changing this database, but i do not see any direction on how to change the configuration parameters and the ranges ? 

Also how long time does it take to refresh through the binding ? 

Thanks.

cdjackson, 2021-01-02:
> i do not see any direction on how to change the configuration parameters

Did you read the guide?

[image]

> Also how long time does it take to refresh through the binding ?

Normally a few days. Really, it is the only way to do this properly - if you are suggesting not to use the database to speed things up, that would be problematic as we try and use the database as a single source. The database is used by a number of projects - not just openhab - I forget the names now (I think Homie is one).

Roagert, 2021-01-02:
> Did you read the guide?
>
> [image]
>
>
> Normally a few days. Really, it is the only way to do this properly - if you are suggesting not to use the database to speed things up, that would be problematic [...]

Thanks i found the guide and created a ticket. Sorry for the incompetence :p First time messing with it and its not that straight forward when its all new :) 

But.. All in all i would like to mention that this binding is the first that actually works pretty great compared to much else i have tested.

I am not trying to speed things up i was just curious of how it works and where the data was comming from and who else uses that database. Nothing more than that. I completely agree this is a great solution and im already used to revision control of projects at work so im used to the delay ;)

cdjackson, 2021-01-02:
> First time messing with it and its not that straight forward when its all new :)

That's true :)

I'll check the tickets shortly...

A good place to discuss any issues is on the forum as there are a few other people there who can help, but they are less active here...

> I am not trying to speed things up i was just curious of how it works

Ok, understood. I try and do an update every few days, but it can be up to a week...

Roagert, 2021-01-02:
> That's true :)
>
> I'll check the tickets shortly...
>
> A good place to discuss any issues is on the forum as there are a few other people there who can help, but they are less active here...
>
>
> Ok, [...]

Thank you very much ! :) 

Great work, ill close this when i have finally made the change and the change has been applied to my own session :)

Roagert, 2021-01-03:
Okay so i managed to update the database with the changes from above. 

Although i still seem to have some issues with parameter 23 and 26. Doesnt seem right that the value cannot be larger than 65535?

[image]

cdjackson, 2021-01-03:
I guess that this is a problem in the UI?

```
      <parameter name="config_23_4" type="integer" groupName="configuration"
                 min="0" max="4294967295">
```

```
      <parameter name="config_26_4" type="integer" groupName="configuration"
                 min="0" max="4294967295">
```

@ghys any thoughts? Maybe it's also in the core I guess?

ghys, 2021-01-03:
Protip: you can use the widget feature in the UI to quickly emulate how the config parameters will render.
Just write them in YAML and click Set Props in the bottom toolbar.
[image]
[image]

So I guess in this case the UI seems to be behaving properly. Would be interesting to know if in the REST API response the max value is capped to 65535.

Roagert, 2021-01-03:
> Protip: you can use the widget feature in the UI to quickly emulate how the config parameters will render.
> Just write them in YAML and click Set Props in the bottom toolbar.
> [image]
> [image]
>
> So I [...]

Interesting, i am no expert herre so i ambolt totalt aware of how the REST API behaves and how to figure that out.
But god to know it’s not the configuration parameter and not the UI.

Roagert, 2021-01-04:
I also states that the default value is 1 + 255 + 0 + 0 which is 01FF0000 in hex and in decimal it gives 33488896

Roagert, 2021-01-05:
@cdjackson @ghys , I am able to use other software like OZWAdmin to give this value to the device, so i know for sure its openhab that is the issue. But where in openhab the issue lies i do not know. Can either one of you guide me to some document or some help to solve this issue ?

I tried to attach a debug logger for the zwave, but there was nothing interesting there.
I tried to use chrome webbrowser debugger (F12) to see if that helped me, it did show some rest api but api is not getting triggered by changing the value and clicking save.

Could it be some cached filed that is forcing it to this value after updating parameters in the opensmarthome database ?

I need some guidance on how to continue from here since this device will be in the majority of my house and i would like to solve the issue.

Thanks.

cdjackson, 2021-01-05:
I think that this must be a UI issue - either that or the binding that you have is not set correctly, and is set to have a max of 65535. The error that you see I assume is generated in the UI - the binding will not generate this, so I can only assume that the either the configuration is wrong, and it is set to 65535 max (and we see that at least the database NOW doesn't have that) or there is a problem in the UI.

I just confirmed that in the OH3 binding, it is set to 65535, so I think the mystery is solved 👍 

Hopefully we can close this?

Roagert, 2021-01-05:
@cdjackson 
But if the binding has a max on that value, how do we get around that value ?

Because i know for a fact this is allowed ?

cdjackson, 2021-01-05:
> But if the binding has a max on that value, how do we get around that value ?

You can't.

cdjackson, 2021-01-05:
To add - your only option is to use a newer binding.

Roagert, 2021-01-05:
> To add - your only option is to use a newer binding.

And how do i find a newer binding ? 

This is sadly kinda forcing me away from openhab if im not able to change the value that is specified by the manufacturer of the device.

cdjackson, 2021-01-05:
> This is sadly kinda forcing me away from openhab if im not able to change the value that is specified by the manufacturer of the device.

I can only apologise - I'm sorry about this, but mistakes do happen. We try our best, but even so errors creep through and this is a community effort so it's not just me that's to blame here.

Roagert, 2021-01-05:
@cdjackson Sorry im not trying to be a jerk. Im just saying that if i cannot change this value how can i then continue to use a device?

Maybe you meant that i am not able to use the current version until the binding is updated code wise? 
And a developer needs to fix the "bug" before i can get my issue solved ?

Ofcourse i know you're doing your best. Thank you very much for your effort !

cdjackson, 2021-01-05:
> Maybe you meant that i am not able to use the current version until the binding is updated code wise?

The binding has already been updated - as I said above, you simply to use the latest binding - ie the SNAPSHOT version. We can't re-release the released version.

> And a developer needs to fix the "bug" before i can get my issue solved ?

I'm not sure what you mean. The database was updated - I don't know if you did this as we discussed a few days ago, or if someone else did, but either way, the "bug" was fixed already. I'm not sure if that's what you mean here when you talk about "getting your issue solved"? Or do you mean closing this issue?

Either way, this should be resolved - you just need to use the latest binding as I'm reasonably sure it is already fixed there as you can see -:

https://github.com/openhab/org.openhab.binding.zwave/blob/master/src/main/resources/OH-INF/thing/logic/zdb5100_0_0.xml#L381

ghys, 2021-01-05:
> But how do i check the REST API ?

Open the developer tools (F12) of the browser before you go to the thing page, go to the Network tab, open the thing page, find a line with the UID of your thing, to to Preview tab, open configDescriptions and find your config parameter.

Like this (this example is for another device but you get the idea):

[image]

If you go to the Code view and enter your config in YAML (`config_xx_y: 123456789` under `configuration:`) and save with Ctrl+S, I believe the range constraints won't be enforced.

Roagert, 2021-01-05:
> The binding has already been updated - as I said above, you simply to use the latest binding - ie the SNAPSHOT version. We can't re-release the released version.
>
>
> I'm not sure what you mean. The [...]

I see, i thought the binding would get an "update" or the database was not "hard-coded" in the binding. Sorry for my incompetence in this area I'm really trying to learn how the software is working.

I now understand that this binding needs to be re-compiled and placed in my addon folder before i get the benefit of the new values.

> Open the developer tools (F12) of the browser before you go to the thing page, go to the Network tab, open the thing page, find a line with the UID of your thing, to to Preview tab, open [...]

Thanks for showing this!

cdjackson, 2021-01-05:
> I see, i thought the binding would get an "update" or the database was not "hard-coded" in the binding.

No - it's hard coded in the binding. That's just the way OH works - the thing defintions need to be compiled in (unless the binding was to rewrite a large chunk of core code).

> I now understand that this binding needs to be re-compiled 

You don't need to recompile - if you simply use the latest nightly build, it is already provided for you.  I do these exports every couple of days, so the binding is kept reasonably well up to date.

cdjackson, 2021-01-05:
> If you go to the Code view and enter your config in YAML (config_xx_y: 123456789 under configuration:) and save with Ctrl+S, I believe the range constraints won't be enforced.

Interesting. I've not looked at this, but how is this updating the config description in the binding? Or is this just bypassing the range enforcement in the UI - not the range enforcement in the core itself?

Roagert, 2021-01-05:
@cdjackson i have now installed the latest SNAPSHOT binding and it seems to work as it should.

Thank you for the help and sorry for the inconvenience.
This can be closed.

ghys, 2021-01-05:
> Or is this just bypassing the range enforcement in the UI - not the range enforcement in the core itself?

It's only bypassing the validation in the UI. I'm not aware that there's a range enforcement in the core at all, is there any?

At first I was of the position that these parameter descriptions were indicative only (https://github.com/openhab/openhab-webui/issues/267#issuecomment-635207373) and the user shouldn't be denied from saving the configuration even if invalid, because I knew too well that it would backfire if the parameter descriptions were wrong. But then I let myself be convinced.

cdjackson, 2021-01-05:
> I'm not aware that there's a range enforcement in the core at all, is there any?

I thought there was. Let me check.

cdjackson, 2021-01-05:
So, there is a [ConfigDescriptionParameterValidatorFactory](https://github.com/openhab/openhab-core/blob/774b9d607b8277f809c2b2b1d94da1be9c23156d/bundles/org.openhab.core.config.core/src/main/java/org/openhab/core/config/core/internal/validation/ConfigDescriptionParameterValidatorFactory.java) that I believe provides all the validators. One of these is `MinMaxValidator` which checks the range - I have not checked further, but I assume that this is being used and therefore range checking is enforced by the core.

ghys, 2021-01-05:
Interesting - going further, the only relevant occurrence of ConfigDescriptionValidator is the core of https://github.com/openhab/openhab-core/search?q=ConfigDescriptionValidator is in ThingManagerImpl:

https://github.com/openhab/openhab-core/blob/774b9d607b8277f809c2b2b1d94da1be9c23156d/bundles/org.openhab.core.thing/src/main/java/org/openhab/core/thing/internal/ThingManagerImpl.java#L287

...but I don't see this method called at all anywhere.

cdjackson, 2021-01-05:
Strange. So it seems that it should be checked - but possibly isn't... I've not actually confirmed, but if you're right I guess there's a bug in the core.....

cdjackson, 2021-01-05:
It is called from the BaseThingHandler. This called the callback. validateConfigurationParameters which in turn calls the manager validateConfigurationParameters…

It’s convoluted, but I think that is correct (if I followed it all through ok).

ghys, 2021-01-05:
You're right.
However these validators return ConfigValidationMessages like https://github.com/openhab/openhab-core/blob/774b9d607b8277f809c2b2b1d94da1be9c23156d/bundles/org.openhab.core.config.core/src/main/java/org/openhab/core/config/core/internal/validation/MinMaxValidator.java#L34

But in https://github.com/openhab/openhab-core/blob/774b9d607b8277f809c2b2b1d94da1be9c23156d/bundles/org.openhab.core.thing/src/main/java/org/openhab/core/thing/internal/ThingManagerImpl.java#L290
it appears to ignore the validator's return value.

ghys, 2021-01-05:
never mind, it appears to throw an exception here: https://github.com/openhab/openhab-core/blob/774b9d607b8277f809c2b2b1d94da1be9c23156d/bundles/org.openhab.core.config.core/src/main/java/org/openhab/core/config/core/internal/validation/ConfigDescriptionValidatorImpl.java#L129

cdjackson, 2021-02-03:
Please provide the logs that support the issue. Are you sure this is a binding issue or is it a UI issue (maybe the logs will help?).

Roagert, 2021-02-05:
i get this info in the log

2021-02-05 19:55:02.637 [INFO ] [openhab.event.ConfigStatusInfoEvent ] - ConfigStatusInfo [configStatusMessages=[ConfigStatusMessage [parameterName=config_14_4, type=PENDING, messageKey=null, arguments=null, message=null, statusCode=null]]]

2021-02-05 19:55:05.415 [INFO ] [openhab.event.ConfigStatusInfoEvent ] - ConfigStatusInfo [configStatusMessages=[ConfigStatusMessage [parameterName=config_14_4, type=PENDING, messageKey=null, arguments=null, message=null, statusCode=null]]]

cdjackson, 2021-02-05:
This is not really related. You need to provide a debug log please.

Roagert, 2021-02-08:
@cdjackson I will leave it open for a bit longer. I have reset my OH3 system and started from scratch since there were alot of issues in the end. i hope to see the errors gone now.

closed 2021-04-29 by Roagert