openhab/org.openhab.binding.zwave #1498: TODO: Implement processing of Request Message = -- (0xf8)

t09, 2021-01-04:
Hi,

I get this message in log:

```
2021-01-04 11:22:57.260 [WARN ] [ve.internal.protocol.ZWaveController] - TODO: Implement processing of Request Message = -- (0xf8)
2021-01-04 11:22:57.433 [WARN ] [.serialmessage.ZWaveCommandProcessor] - SerialMessage class null is not implemented!
2021-01-04 11:22:57.434 [WARN ] [ve.internal.protocol.ZWaveController] - TODO: Implement processing of Request Message = -- (0xf8)
2021-01-04 11:22:58.032 [WARN ] [.serialmessage.ZWaveCommandProcessor] - SerialMessage class null is not implemented!
2021-01-04 11:22:58.034 [WARN ] [ve.internal.protocol.ZWaveController] - TODO: Implement processing of Request Message = -- (0xf8)
2021-01-04 11:22:58.395 [WARN ] [.serialmessage.ZWaveCommandProcessor] - SerialMessage class null is not implemented!
2021-01-04 11:22:58.396 [WARN ] [ve.internal.protocol.ZWaveController] - TODO: Implement processing of Request Message = -- (0xf8)
2021-01-04 11:22:59.031 [WARN ] [.serialmessage.ZWaveCommandProcessor] - SerialMessage class null is not implemented!
2021-01-04 11:22:59.033 [WARN ] [ve.internal.protocol.ZWaveController] - TODO: Implement processing of Request Message = -- (0xf8)
2021-01-04 11:23:00.046 [WARN ] [.serialmessage.ZWaveCommandProcessor] - SerialMessage class null is not implemented!
2021-01-04 11:23:00.047 [WARN ] [ve.internal.protocol.ZWaveController] - TODO: Implement processing of Request Message = -- (0xf8)
2021-01-04 11:23:00.160 [WARN ] [.serialmessage.ZWaveCommandProcessor] - SerialMessage class null is not implemented!
2021-01-04 11:23:00.162 [WARN ] [ve.internal.protocol.ZWaveController] - TODO: Implement processing of Request Message = -- (0xf8)
```

* Device name: RPi 3b+ with UZB 1 with licence and latest firmware
* OH version: latest stable OH3

Thanks in advance

cdjackson, 2021-01-04:
It's a bit hard to know what to do here? Please provide logs, or a description of your devices or something that allows us to see what is happening. Otherwise suggest you can probably ignore this as it's probably not critical (but I can't really tell without more information).

t09, 2021-01-04:
Hi,

here is debug log - is there any additional Z-Wave log openHAB creates?
[part_openhab.log](https://github.com/openhab/org.openhab.binding.zwave/files/5765975/part_openhab.log)
Stick can be found here: https://z-wave.me/products/uzb/
Firmware: https://z-wave.me/support/uzbrazberry-firmwares/ (I think its 5.38 - but I can only check on Z-Way I think)
Rest of the network are Fibaro products.

Can't remember to have this on OH2.5 - but I think it slows down my network. Is it possible?
I changed my logging to ERROR now to get rid of those messages but I'm afraid I could miss future errors.

Please let me know if the log is helpful.

regards

cdjackson, 2021-01-04:
Thanks.

I'm not sure what these packets are -:

```
payload=01 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
```

They don't look right to me, and I would just ignore them.

It might be something non-standard that your controller has had added to the firmware - I'm not really sure.

closed 2021-01-05 by t09