openhab/org.openhab.binding.zwave #1511: Parameters 11/12 cannot be completely set with a QUBINO DIN RAIL DIMMER ZMNHSD H1S3P3 in Openhab3

DiKaY1969, 2021-01-17:
I and a friend use some QUBINO DIN RAIL DIMMERs ZMNHSD H1S3P3 with Openhab 3.

https://qubino.com/products/din-dimmer/din-dimmer-tech/

The problem is present in OH3.0.0 and 3.1.x.
The problem did not exist in OH2.5.x.

Unfortunately, the two parameters 11 & 12 cannot be set via the Thing configuration.
However, the two parameters can be set and saved via the code tab.
0 is selected by default.

Parameter no.11 - Turn Load 1 (Q) OFF Automatically with Timer
Parameter no.12 - Turn Load 1 (Q) ON Automatically with Timer

Here is the description of the parameter of Quibino:
default value 0
0 - Auto OFF / ON Disabled
1 - 32536 = 1 - 32536 seconds - Auto OFF / ON timer enabled for a given amount of seconds

Is that an error in the database or in Obenhab 3?

If there is an error in the database, can you please correct it?

Many Thanks

cdjackson, 2021-01-17:
Since you have posted exactly the same thing on the support forum, I'll close this here - I'm sure this will be resolved through the forum and possibly database updates.

closed 2021-01-17 by cdjackson

DiKaY1969, 2021-02-20:
Many Thanks