openhab/org.openhab.binding.zwave #1514: Changes for zdb5100

PoulKlemmensen, 2021-01-21:
Hi.

I have been using this with OpenZWave for some time but want to try a better HA system out. But in your configuration, I seem to be missing a pushbutton 5, that actually turns the dimmer completely off.

Secondly, I am missing some possibilities on Param 1. How the dimmer is controlled by the buttons. Combinations of the buttons are also possible:
3 : Button 1+2
5: Button 1+3
and so on...

I have an openzwave configuration file, that I can submit if you want to.

Roagert, 2021-02-08:
@PoulKlemmensen Hi Paul,

Regards to the devices i believe we can make them our selves.
I have modified ZDB5100 alittle to accomodate the max and min values of the color in the diodes.

today i have also added some parameters to the ZIF5028 because there were missing some parameters.
You can go here: 
https://opensmarthouse.org/

Here you can modify the XML files for the devices you just need to request access, try and read the "blog" post about how to use the site.

cdjackson, 2021-03-06:
If there are changes required to the database, please make them at opensmarthouse.org.

closed 2021-03-06 by cdjackson