openhab/org.openhab.binding.zwave #1518: Qubino ZMNHTD Smart Meter relay control endpoints 1 & 2 not found

RomanZ737, 2021-01-23:
Hello.

Z-Wave devices:

    Qubino ZMNHTD Smart Meter developed by Goap with the thing type UID of qubino_zmnhtd_00_000
    In addition to Z-Wave device I’m using: BICOM432 External IR Relay and IKA232 External Reley

Issue description:

    Not able to control External IR and External relays from openhab (paper UI or habpannel) - endpoints 1 & 2
    Relays status is not available form the openhab
    At the same time I can control both relays with the wired hardware switch.

It looks like endpoints 1 & 2 are missing during Initialization process

My system:
Openhab 2.5.11 Release Build
Z-Wave Binding 2.5.11

Z-Wave DEBUG logs:

```
2021-01-21 14:30:24.993 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 15: Command received zwave:qubino_zmnhtd_00_000:controller:smartMeterKitchen:switch_binary2 --> ON [OnOffType]
2021-01-21 14:30:24.993 [DEBUG] [converter.ZWaveBinarySwitchConverter] - NODE 15: Command class class COMMAND_CLASS_SWITCH_BINARY for endpoint 2 not found
2021-01-21 14:30:24.993 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 15: No messages returned from converter
2021-01-21 14:30:25.896 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 15: Command received zwave:qubino_zmnhtd_00_000:controller:smartMeterKitchen:switch_binary1 --> ON [OnOffType]
2021-01-21 14:30:25.897 [DEBUG] [converter.ZWaveBinarySwitchConverter] - NODE 15: Command class class COMMAND_CLASS_SWITCH_BINARY for endpoint 1 not found
2021-01-21 14:30:25.897 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 15: No messages returned from converter
```
Initialization unfiltered logfile:
[openhab.log](https://github.com/openhab/org.openhab.binding.zwave/files/5861019/openhab.log)

Node XML file:
[node_15.txt](https://github.com/openhab/org.openhab.binding.zwave/files/5861021/node_15.txt)

Would be appreciate for any help or advice.

cdjackson, 2021-01-23:
It seems from the XML that only endpoint 0 is provided by this device. I know Qubino have a habit of hiding endpoints so please read the manual to see if this requires additional configuration before he device will repair any other endpoints.

RomanZ737, 2021-01-23:
Thanks for quick reply.
But does binding actually "asking" the device about endpoint or just get them from recompiled xml files of the zwave binding upon discovering device model?

cdjackson, 2021-01-23:
The XML file contains information the binding downloads from the device - it has no link to any precompiled XML files.

RomanZ737, 2021-01-23:
Got it. Thanks for explanation. There is not much documentation about this device but I will try to check one more time carefully.

RomanZ737, 2021-01-24:
Chris, do you know if that possible to generate/write/fix my own Node xml file instead of generated aromatically? Probably not the whole file but add missing endpoints.

cdjackson, 2021-01-24:
No - you can't mess with these files. It will also make no difference - if the device doesn't support it, then it doesn't support it. Modifying the XML will not change that.

I had a quick look in the manual - I see there are commands to enable the switch - is it enabled?

RomanZ737, 2021-01-24:
Yes I did. According to the manual I can enable/disable endpoints using parameter 100. I have changed the parameter in order to enable the endpoints and did reinitialization after that but it didn't help. I also tried to exclude and include the device without restoring default settings but no luck with that. So finally I sent a letter to the Qubino support cause at this stage I do not have any idea what else I can do with that.

cdjackson, 2021-01-24:
You almost certainly need to exclude and reinclude the device after you’ve done this. This is because a device cannot enable and disable endpoints - these are fixed after a device is included. Other Qubino manuals say this but I didn’t see it here.

RomanZ737, 2021-01-24:
> No - you can't mess with these files. It will also make no difference - if the device doesn't support it, then it doesn't support it. Modifying the XML will not change that.

I'm pretty sure this device can control External relay via endpoints. At least all manuals saying that.

RomanZ737, 2021-01-24:
By default the device is coming with endpoints disabled. So you suggesting to try exclude and include the device after the setting 100 has been changed?

cdjackson, 2021-01-24:
Sure - but it cannot add endpoints - that's not possible. Once a device has been added to a network, it can't change it's features. If you enable the endpoint, you will need to exclude it.

RomanZ737, 2021-01-24:
Ok. I will try to do that again. The point is not to reset the device to factory defaults settings during exclusion.

RomanZ737, 2021-01-25:
Hi Chris,
Currently I manage to enable endpoints 1 & 2 (thanks for your advice) so now I can control IR and External relays from the openHab. 

But now I have faced to another kind of strange behaviour (not sure if it's normal). When I change the state of the switch binary 2 (endpoint 2) - it changes the state of External relay and the status of the relay is clearly reflected in the widget. But when I change state of the switch binary 1 - it changes the states of both IR and External relays and the status of both relays is also clearly reflected in the status board. So the switch binary 1 is changing the state of both relays simultaneously. At the same time I can control the relay state independently using wired hard ware switch. Is there any clues how to control the relays independently using openhab interface? 

Some logs if it would make any sense:

Switch Binary 2 set to ON

```
2021-01-25 20:50:23.186 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 13: Command received zwave:device:controller:node13:switch_binary2 --> ON [OnOffType]
2021-01-25 20:50:23.186 [DEBUG] [dclass.ZWaveBinarySwitchCommandClass] - NODE 13: Creating new message for application command SWITCH_BINARY_SET
2021-01-25 20:50:23.186 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 13: Encapsulating message, instance / endpoint 2
2021-01-25 20:50:23.186 [DEBUG] [class.ZWaveMultiInstanceCommandClass] - NODE 13: Creating new message for command MULTI_CHANNEL_ENCAP endpoint 2
2021-01-25 20:50:23.186 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 13: SECURITY NOT required on COMMAND_CLASS_MULTI_CHANNEL
2021-01-25 20:50:23.186 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 13: Command Class COMMAND_CLASS_MULTI_CHANNEL is NOT required to be secured
2021-01-25 20:50:23.186 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 13: Adding to device queue
2021-01-25 20:50:23.186 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 13: Added 6307 to queue - size 1
2021-01-25 20:50:23.187 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Transaction SendNextMessage 0 out at start. Holdoff false.
2021-01-25 20:50:23.187 [DEBUG] [wave.internal.protocol.SerialMessage] - Assembled message buffer = 01 0E 00 13 0D 07 60 0D 01 02 25 01 FF 25 5D 25 
2021-01-25 20:50:23.187 [DEBUG] [ing.zwave.handler.ZWaveSerialHandler] - NODE 13: Sending REQUEST Message = 01 0E 00 13 0D 07 60 0D 01 02 25 01 FF 25 5D 25 
2021-01-25 20:50:23.187 [DEBUG] [ing.zwave.handler.ZWaveSerialHandler] - Message SENT
2021-01-25 20:50:23.187 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Transaction SendNextMessage started: TID 6307: [WAIT_RESPONSE] priority=Set, requiresResponse=true, callback: 93
2021-01-25 20:50:23.188 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 06
2021-01-25 20:50:23.188 [DEBUG] [nal.protocol.ZWaveTransactionManager] - processReceiveMessage input 0<>128 : Message: class=null[0], type=ACK[2], dest=255, callback=0, payload=
2021-01-25 20:50:23.189 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Received msg (0): Message: class=null[0], type=ACK[2], dest=255, callback=0, payload=
2021-01-25 20:50:23.189 [DEBUG] [nal.protocol.ZWaveTransactionManager] - lastTransaction TID 6307: [WAIT_RESPONSE] priority=Set, requiresResponse=true, callback: 93
2021-01-25 20:50:23.189 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Received msg: ACK
2021-01-25 20:50:23.189 [DEBUG] [nal.protocol.ZWaveTransactionManager] - ZWaveReceiveThread queue empty
2021-01-25 20:50:23.189 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Transaction SendNextMessage 1 out at start. Holdoff false.
2021-01-25 20:50:23.195 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 04 01 13 01 E8 
2021-01-25 20:50:23.195 [DEBUG] [nal.protocol.ZWaveTransactionManager] - processReceiveMessage input 0<>128 : Message: class=SendData[19], type=Response[1], dest=255, callback=0, payload=01 
2021-01-25 20:50:23.195 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Received msg (0): Message: class=SendData[19], type=Response[1], dest=255, callback=0, payload=01 
2021-01-25 20:50:23.195 [DEBUG] [nal.protocol.ZWaveTransactionManager] - lastTransaction TID 6307: [WAIT_RESPONSE] priority=Set, requiresResponse=true, callback: 93
2021-01-25 20:50:23.195 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Checking outstanding transactions: 1
2021-01-25 20:50:23.195 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Last transaction: TID 6307: [WAIT_RESPONSE] priority=Set, requiresResponse=true, callback: 93
2021-01-25 20:50:23.195 [DEBUG] [ve.internal.protocol.ZWaveController] - Incoming Message: Message: class=SendData[19], type=Response[1], dest=255, callback=0, payload=01 
2021-01-25 20:50:23.195 [DEBUG] [l.serialmessage.SendDataMessageClass] - NODE 13: sentData successfully placed on stack.
2021-01-25 20:50:23.195 [DEBUG] [nal.protocol.ZWaveTransactionManager] - TID 6307: Advanced to WAIT_REQUEST
2021-01-25 20:50:23.195 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 13: TID 6307: Transaction not completed
2021-01-25 20:50:23.195 [DEBUG] [nal.protocol.ZWaveTransactionManager] - ZWaveReceiveThread queue empty
2021-01-25 20:50:23.195 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Transaction SendNextMessage 1 out at start. Holdoff false.
2021-01-25 20:50:23.211 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 07 00 13 5D 00 00 02 B4 
[... cut ...]
2021-01-25 20:48:09.341 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 13: Checking command org.openhab.binding.zwave.internal.protocol.ZWaveCommandClassPayload@328958d7.
2021-01-25 20:48:09.341 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Transaction completed - outstandingTransactions 0
2021-01-25 20:48:09.341 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Transaction completed - outstandingTransactions 0
2021-01-25 20:48:09.341 [DEBUG] [nal.protocol.ZWaveTransactionManager] - ZWaveReceiveThread queue empty
2021-01-25 20:48:09.341 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Transaction SendNextMessage 0 out at start. Holdoff false.
```

cdjackson, 2021-01-25:
I’m glad you got it working.

I’m not sure if you can control the different things separately - that is completely down to the device. The binding can only send the commands.

RomanZ737, 2021-01-25:
As I said I have sent a request to Qubino help desk. So I got a reply from them today (when I already have fixed my main problem.)
______________
_It is possible that you didn't activate both of the endpoints.. you need to set the parameter 100 to 3, exclude the device (without reseting it) and add it back to the gateway. After the inclusion the gateway will set the MC (multi channel) lifeline which is correct, but the problem is that the gateways widgets are not set to receive the MC that will lead you to a point where the device will work and send the proper reports but the gateway will not show them as it is not program to do so.
Solution you could use the PC controller to set the lifeline back to SC but in this case you will be unable to see the status on both of the contactor but only the status of the latest one._
_____________________________
I'm not sure if i understood what did he meant by "PC controller". Cause only option with the lifeline in the Smart Meter options is to set Controller of other Node.... 
I will defiantly ask the help desk about that and about my second problem (with independent relay contlol) but before that probably you have any clues... hope I'm not taking too much of your time.

alexus1211, 2021-01-26:
Hi,

i remember having trouble controlling this device as well. After making sure that both endpoints have been activated, and after successful exclusion/inclusion, i had to modify the device database locally to make it work as intended. 
Please bare in mind that i use my build the zwave locally before deploying it on my machine.
attached you find the xml file I use:

[zmnhtd_0_0.xml.zip](https://github.com/openhab/org.openhab.binding.zwave/files/5873187/zmnhtd_0_0.xml.zip)
Hope this helps

RomanZ737, 2021-01-26:
Hi,
Thanks for help.
I will try to make manual changes base on you xml.

cdjackson, 2021-01-26:
If there is a problem with the database definition, wouldn’t it make sense to update the database?

RomanZ737, 2021-01-26:
Do you mean to put a manually edited zmnhtd_0_0.xml to the new data base. Sorry I didn't get what do you mean by update the database.

cdjackson, 2021-01-26:
No - absolutely not. I assume that the XML provided here fixes something that is wrong in the database? If so, why not just fix the database rather than have to recompile the binding with a custom file? Then it is available for everyone to use.

RomanZ737, 2021-01-26:
That sound good. But I'm not familiar with the procedure. How can I do that? Ask the community?

cdjackson, 2021-01-26:
I guess this was more targeted at the other guy who provided the XML file. My point is that if there are things that are wrong, then the best place to fix them is in the database directly. There is guidance on the database site, or you can ask the community. Without knowing what’s wrong though, it’s hard to fix it...

RomanZ737, 2021-01-26:
Got it. I will ask the community.

cdjackson, 2021-01-26:
Ok, but first don’t you need to know what is wrong? All that I’ve seen is “you have to modify the database” - do you know what needs to be modified?

RomanZ737, 2021-01-26:
According to @alexus1211 advice the problem with the endpoints description in one single file zmnhtd_0_0.xml of the zwave binding. But I'm test in yet if it going to fix my problem

alexus1211, 2021-01-26:
@cdjackson : the thing is, I am running version 6.7 of the smart meter. I have received recently an update to 6.9 and I am not sure if the xml needs further modification to make it work. That's why I was a bit reluctant merging my changes into the main database. Hopefully I can provide some feedback until tomorrow.

@RomanZ737 : just google it and there are multiple helpful links to show you how to build your own binding. Basically you need:
1- setup the development tools (Eclipse, Java, ...)
2- git clone the Chris's repo to your machine
3- modify the xml
4- build
5- Make sure that you remove / deselect the zwave binding  from PaPerUI, otherwise OH will pick up the officially released binding
6- copy the generated jar and paste it to openhab addons directory. 

Hope this helps

cdjackson, 2021-01-26:
Ok, I'm pretty lost to be honest.

> According to @alexus1211 advice the problem with the endpoints description in one single file zmnhtd_0_0.xml of the zwave binding.

Sure - but if this file is wrong, then we should update the database - not this file. If you update the file, then you have to recompile the binding. If someone else wants to fix it, we will discuss this again, they they will have to recompile the binding. If we fix it in the database, then EVERYONE gets the fix.

> I am running version 6.7 of the smart meter. I have received recently an update to 6.9 and I am not sure if the xml needs further modification to make it work. 

I still don't know what changes are required, but you have said that this new XML is required to make it work. It seems now that this isn't the case? Are you able to actually detail what the changes are?

alexus1211, 2021-01-26:
Let me explain: when I bought this device back in 2017, the device was very buggy and I was facing lot of issues with it, not only the ability to control the single endpoints, but also general reporting issues. Qubino offered to flash the unit for me with their latest firmware back in 2018. That's when I got 6.7. Since then, it is working pretty nice I would say and with the xml modification I was able to control the endpoints separately. I still face some other minor reporting issues from time to time, but in terms of controlling the device everything works like a charm. 
Tonight I am planning on flashing it up to 6.9. I am just afraid that Qubino might have changed the endpoints mapping. For example, EP1 is used to control both relays in FW 6.7. I've reported that to Qubino as a minor issue. Qubino's argument was that EP1 is used to control the relays as all. Some of their customers will activate either the optical or wired relay, but not both. In that case EP1 would be adequate for controlling any of those. EP2 can refer in this case to either optical or wired, depending on what has been activated on the device. They handle these channels dynamically. 
In case you activate both, then EP2 and EP3 would refer to Optical and Wired relays successively.
I've told Qubino that this does not make sense in my opinion. My expectation would be to see a maximum of 2 endpoints based on any configuration. If only one relay was activated, then I would expect EP1 to control it. If both, then EP1 and EP2 accordingly.
I cannot comment on this behaviour in 6.9 since I've haven't had the time yet to flash it.
Hopefully tomorrow I can report on that.

RomanZ737, 2021-01-26:
6.9 is a Firmware version right? In this case 6.9 is not the latest one.
[image]

alexus1211, 2021-01-26:
It is for the HW variant of my device. I guess mine is of older generation and I am not allowed to flash it with a FW above 6.9.
Yours seems to be the latest.
What you can do, is send me the generated xml by the zwave binding and I can compare it with mine.
Depending on your installation, you can find it in the userdata folder (in my case /var/lib/openhab2/zwave/). All files there follow the same nomenclature: network_xxxxxxxx__node_y.xml. Just grab the file corresponding to the nodeID of your Qubino smart meter

RomanZ737, 2021-01-26:
Here is the one

[network_f844cc5f__node_13.xml.txt](https://github.com/openhab/org.openhab.binding.zwave/files/5874249/network_f844cc5f__node_13.xml.txt)

I suspect there is no much difference cause the behavior is pretty much the same... over the years =)

alexus1211, 2021-01-26:
mapping looks similar based on the file you have provided. The only difference is that your module supports cc_security, while mine doesn't. The modified xml that I've provided should be ok

RomanZ737, 2021-01-26:
Thanks for you time. 
At the moment I'm trying to build the .jar file with binding using Eclipse but openhab give the this error:
Invalid manifest header Require-Capability: "osgi.ee;filter:="
Sorry never tried to do that kind of operations before.

alexus1211, 2021-01-26:
Most probably due to java version mismatch between the build machine and the java runtime of your openhab. Try this: in eclipse do the following: Window -> Preferences, then select Java -> Compiler from the left pane and set the Compiler compliance Level to 1.8. Then rebuild.
This should do the trick.

RomanZ737, 2021-01-26:
Well I have tried and your advice did the trick but bonding it self generating faults. 
Would you be so kind to send you rebuild zwave binding? And I will try it on my system.

alexus1211, 2021-01-26:
you can download it from here and check if it works for you.
https://www.dropbox.com/s/lnd6t9f6hfgjphf/org.openhab.binding.zwave-2.5.12-SNAPSHOT.jar?dl=0

RomanZ737, 2021-01-27:
Your binding did the trick. Thanks a lot. Since you have experience with this device, did you add Smart Meter via PaperUI or via .things file? If you using files how do edit a special properties like “Polling time” and others?

cdjackson, 2021-01-27:
If that fixes the issue, then please update the database so that others benefit - and in fact so that you also benefit as otherwise you won’t be able to update the binding.

RomanZ737, 2021-01-28:
@alexus1211 have you had “ERROR 500 - Internal Server Error” when tried to check Zwave properties with updated binding? I’m trying to understand if than my local problem or a binding issue. 
[image]

alexus1211, 2021-01-29:
Hi,
I don't usually use PaperUI to manage Zwave things and items. Habmin is definitely a better choice.

RomanZ737, 2021-01-29:
I'm not usually using PaperUI ether but in Habmin those options are not available. Any way usually I'm using .things files to edit the properties. Did you add you smart meter via .things file? If so how did you describe a polling time and other properties there?

alexus1211, 2021-01-29:
I am afraid I can’t follow here. .things files are usually to configured bindings not items. Why do you need to poll this device. There are configuration parameters you can set to tell the device how often you need a report from it, or by which amount of change a report should be sent. Polling is a bad practice. Check parameters 40 and 42 of your device

RomanZ737, 2021-01-29:
That what I'm talking about - I'm not able to see those parameters after I installed a new binding. Parameters are missing in Habmin. I still not sure if it's my local problem. Probably if you are able to see and edit those parameters in your Hambin than the problem is on my side and not with the new binding.

alexus1211, 2021-01-29:
I am not sure what is happening on your device, but try the following:
- in HABmin, select the smart meter node and delete it. It should disappear from the left pane
- open a bash to your openhab server. Login, then navigate to: cd /var/lib/openhab2/zwave/
- delete the xml file related to the smart meter: network_f844cc5f__node_13.xml
- go back to habmin, start a zwave inclusion process
- the node should appear in the inbox
- wait a couple of minutes, refresh habmin until you see the discovered item in the inbox with full description: Node 13: Qubino Smart Meter or something similar
- click on add
- now it should appear again in the left pane
- try to see if you can see now the configuration parameters

hope this helps

RomanZ737, 2021-01-30:
Thanks for your help! I found the way to control the relays.

closed 2021-02-23 by cdjackson