openhab/org.openhab.binding.zwave #1524: missing ZBA7140 as xml but present as md file in latest snapshot

goodstone74, 2021-01-31:
I have bought a " Logic Group MATRIX ZBA7140" switch and when adding it to openhab3 (and using latest snapshot of zwave) I found that it is not recognized by the zwave binding.

The switch is present in the database and it states that it is part of openhab3: 
[https://www.opensmarthouse.org/zwavedatabase/1330](url)

When looking into the source I see that it is present under “docs in md file format”
[https://github.com/openhab/org.openhab.binding.zwave/tree/main/doc/logic](url)

However it is not present as the XML file used by the zwave binding:
[https://github.com/openhab/org.openhab.binding.zwave/tree/main/src/main/resources/OH-INF/thing/logic](url)

How can I add this device to the XML directory ?
It looks like it is some parsing which I do not have access to, so that is why I am issuing this ticket, I tried to discussing this in the openhab community but did not get any hints there [https://community.openhab.org/t/zwave-missing-zba7140-as-xml-but-present-as-md-file/115871](url)

The openhab versions that I am using are:
* OH3 Version: 3.0.1
* Binding Version latest snapshot version "org.openhab.binding.zwave-3.1.0-20210118.034709-17.jar"

If the problem is associated with a device, please provide the following -:

* Device name:  ZBA7140 
* Device version: unknown
* Link to [database]  : https://www.opensmarthouse.org/zwavedatabase/1330
* If there is an XML being generted for the device (in the /userdata/zwave folder),: This is missing.

cdjackson, 2021-01-31:
Possibly this isn't included in OH3 as the device was only added a day or two before OH3 was released and it therefore probably missed the cut,

> How can I add this device to the XML directory ?

I'm not sure what XML directory you mean? If you mean the one in the userdata folder, then you don't. This folder is for local persistence by the binding only and you should not mess with this.

As it's included in the database, I'd suggest to discuss this on the support forum where other users can help.

goodstone74, 2021-01-31:
Thanks for the quick reply

I was just wondering why it is present in the source document folder as
https://github.com/openhab/org.openhab.binding.zwave/tree/main/doc/logic/matrixzba7140_0_0.md

But it is not present in the source as XML file:
/tree/main/src/main/resources/OH-INF/thing/logic/

It looked like it made the cut but only half way.
Can I manually generate this file or do I need to wait till this file is added to the source?

cdjackson, 2021-01-31:
> But it is not present in the source as XML file:
/tree/main/src/main/resources/OH-INF/thing/logic/

I'm not really sure what these two folders are to be honest. The XML files are compiled into the binding, so unless you have extracted these from the JAR, I'm not really sure what you are looking at.

Do you have the XML file in the userdata folder mentioned earlier? If not, then you need to wake up the device as it's a battery device it will not be discovered until you do this.

> Can I manually generate this file or do I need to wait till this file is added to the source?

If you're using 3.1 snapshot, then it should be included already. Of course you might have a different version, but again if you don't have the XML in the userdata folder, then this means the device is not discovered. Until this happens, the database is irrelevant.

goodstone74, 2021-02-01:
Hi

My process was that I added the zwave device (ZBA7140) via discovery (Add a new Thing: zwave) and found that it was added to the system but noget recognized as a type by the zwave binding.
(The device was now shown as node 15 as XML file in the userdata folder)

So I started debugging the situation, and quickly found that ZBA7140 is known by the https://www.opensmarthouse.org/ database and it also showed that the device is present in both zwave binding version 3.0 and latest OH snapshot.

However when looking at the source on github it became obvious that it exists as a device in the github documentation folder:
https://github.com/openhab/org.openhab.binding.zwave/tree/main/doc/logic

But when looking closer at the source I realized that it is **not present** as a thing in the resource XML folder:
https://github.com/openhab/org.openhab.binding.zwave/tree/main/src/main/resources/OH-INF/thing/logic

And this is why I added this issue.

Today I have found that you have updated the database for release 2.5 and there is the ZBA7140 as XML file:
https://github.com/openhab/org.openhab.binding.zwave/blob/2.5.x/src/main/resources/ESH-INF/thing/logic/matrixzba7140_0_0.xml

So I manually took this file and added this to my local zwave binding JAR file and it worked.

Next time you update the database for the main branch could you please check that matrixzba7140_0_0.xml is added as a thing under the resources folder then I will be very happy.

Ps. Sorry for causing the confusion, it was not intentional.

Thanks,
Kurt

cdjackson, 2021-02-01:
Ok, thanks. I guess the entry was being updated, but sounds like it's sorted now.

closed 2021-02-01 by cdjackson