openhab/org.openhab.binding.zwave #1525: OH3 not populating lifeline and creating xml files with multiple battery devices

JustinGeorgi, 2021-01-31:
Similar issues have now been reported by at least three different posters to the forum: [here](https://community.openhab.org/t/zwave-aeotec-zwa003-nanomote-strange-behavior/115548), [here](https://community.openhab.org/t/z-wave-issue-update-channels-item-with-aeotec-door-window-sensor-7/115927), and most recently [here](https://community.openhab.org/t/oh-3-aeon-labs-minimote-not-working-via-ubz-z-wave-stick/115999).

The OH3 binding seems to be consistently failing when adding new battery run nodes by initializing all default parameters except for the lifeline group.

### System details
System: OH3 (version: 3.0.0) docker install on linux
Java: 11.0.9, (Zulu11.43+21-CA)
Z-Wave Stick: Aeotec Gen5
Z-Wave binding: 3.0.0
### Issue specifics
Devices: I have personally encountered this with two different Aeotec nanomote quads and one Zooz outdoor sensor.  Others reports include Aeotec Door Window sensor 7 and Aeon minimote

Symptoms: When a battery run device is included in the network through OH3, device inclusion and initialization seem to complete (after necessary multiple wake-ups).  However, no further communication between the devices and the controller appears to occur after that.  Further investigation reveals the following details:

1. No xml file for the devices is produced (this is not a zwave folder permissions issue, inclusion of mains powered devices creates xml files successfully)
2. The lifeline group for the device is blank.  All attempts to change this value fail (show in the logs as null status messages), but other parameters can be changed (show in the logs as populated status messages).
3. After the devices have been added to OH and they show properly in the things list with the correct manufacturer and id type, the binding still reports `Device discovery could not resolve to a thingType!` for that same node when the controller is set to inclusion mode.
4. Removing the thing from OH and trying to recreate it will never result in a proper re-discovery (regardless of the amount of manual wake-ups) and identification of the device.  Only actual network exclusion and re-inclusion will permit the binding to properly poll the device data again.
5. Attempts to heal the device do not work.

### Logs
[Link to a debug log file](https://azgeorgis.net/owncloud/s/pV3D40qXIfrtVW9) which includes the complete attempt to install one of the aeotec nanomotes.  Device was previously stuck in this non-repsonsive state.  I excluded it from the network, factory reset it, and then logged the inclusion to completion.  And here you can see that lifeline remains unset:
[image]

cdjackson, 2021-01-31:
> by initializing all default parameters except for the lifeline group.

The binding doesn't initialise ANY default parameters. It never has. Associations should be initialised though.

If the log is meant to be of node 46, then there is not a lot of information included. There doesn't appear to be anything wrong, but there's not a lot of communications.

Sorry - I'm quite busy moving house at the moment so won't have a lot of time for at least the next 3 weeks

cdjackson, 2021-01-31:
It's also worth mentioning that there are no changes from OH2 to OH3, so I don't think this is likely to be an issue with OH3.

JustinGeorgi, 2021-01-31:
> If the log is meant to be of node 46, then there is not a lot of information included. There doesn't appear to be anything wrong, but there's not a lot of communications.

That's what I've been seeing too, so I'm glad you see that same thing.  After the device is added there's almost nada in the way of communication regardless of what I try.

> Sorry - I'm quite busy moving house at the moment so won't have a lot of time for at least the next 3 weeks

No worries, it sounds like quite a job and I don't envy you.  I've got fallback solutions in place so it can wait until you've got a chance to catch your breath.

JustinGeorgi, 2021-01-31:
Sorry, I should have been more clear.  Yes, that's what the log shows, but after the devices goes to sleep I push the buttons, I send aeotec's "quick wake-up", I try several regular wake-ups and none of that activity registers at all in the log.  After the device sleeps and the node adding is complete there's virtually no evidence of communication with that node ever again no matter what I do.

cdjackson, 2021-01-31:
Well, there's no wakeup in the log, so the binding will not do anything until it gets the wakeup message. The binding needs to receive the wakeup notification.

cdjackson, 2021-02-05:
I propose to close this as it doesn't seem to be an issue with the binding as the device is not sending the wakeup.

closed 2021-02-05 by cdjackson

JustinGeorgi, 2021-02-05:
I have no idea if it is the binding or not, but I can't shake the feelings that 1) this is not just a device issue as it is too widespread and consistent and 2) I haven't quite figured out how to fully and accurately describe the issue for you.

As I said before, this is not immediately critical so I'm happy to let it go until you're settled in your new digs.  I'll check back in sometime in March and let you know if I have any additional findings or details.  Best of luck on the move.

cdjackson, 2021-02-05:
If you have more information then please provide it, but from what I’ve seen there is simply no wake up command being sent from the device, and until that is received the binding simply is unable to do anything. 

It doesn’t seem widespread to me - the binding will definitely process the wake up message if it’s received.