openhab/org.openhab.binding.zwave #1528: [zwave] Unable to remove failed node

ccutrer, 2021-02-05:
## Expected Behavior
Clicking "Remove Device From Controller" on the Thing's setting page should work. Then I should be able to delete the Thing, and not have it show up again.

## Current Behavior
When I click it and continue through the confirmation, the log says `NODE 80: Remove failed node failed as Controller Busy!` three times in a row. If I then delete the Thing, and go to inbox and re-scan z-wave, the Thing shows back up, because it was never removed from the actual network.

## Your Environment
* Version used: OpenHAB 3.0.0, Z-Wave binding 3.0.0
* Java 11
* Ubuntu 20.04 focal, amd64

kaikreuzer, 2021-02-05:
done.

cdjackson, 2021-02-05:
If the controller is busy, then there's not a lot that the binding is able to do to change this. This normally happens when there's a lot of traffic on the network - even if you can't see it. When I've seen this in the past, it is caused by an errant device flooding the network. These controllers are very slow microcontrollers and cannot cope with that.

andrewfg, 2021-02-05:
Sorry @cdjackson but I am not sure that I understand you. We are saying that other tools are able to send a specific set of three command to the controller that will remove a ghost device from the controller successfully. Whereas the OH binding seems not to send the same three commands and therefore fails to remove such a ghost device.

I don't see what the network traffic has to do with it. As the network is the same regardless of whether the stick is connected to another tool, or whether it is connected to OH. Or??

cdjackson, 2021-02-05:
Apologies - I'm only going on what is written above which is not the same thing as you are talking about.  

> I don't see what the network traffic has to do with it.

Well, the statement above shows that the controller is busy. Normally this happens when the controller is bust, and that is normally when the network has an issue.

> As the network is the same regardless of whether the stick is connected to another tool, or whether it is connected to OH. Or??

I'm fully aware that there is an issue with the commands that are sent to remove the device - but that is not what was mentioned in this issue which talked about the controller being busy. When the controller can't remove it due to the other issue, the error is normally that the device is not failed.

andrewfg, 2021-02-05:
@cdjackson sorry for any confusion.

I think there are four scenarios for device removal; and they all probably need the binding to send different combinations of commands to the Z-Stick and to the OH core..

1. Existing device in the network; existing Thing in OH
2. Existing device in the network; non instantiated Thing in Inbox
3. Ghost device in the network; existing Thing in OH
4. Ghost device in the network; non instantiated Thing in Inbox

I believe that @ccutrer and I are referring to scenario 4. above. And for us the only way to solve it was with the help of an external tool.

cdjackson, 2021-02-05:
Sure - but the error mentioned, that I responded to, is not related to the removal of the device. That's what I was responding to, but we now seem to be on a totally different, more general, topic.

Regarding the issue about removal, probably I won't really change this in the current binding as it's not super easy to implement within the binding structure.

andrewfg, 2021-02-05:
I will have a look at the code tomorrow, who knows I might be able to suggest something. I guess the alternative would be to reverse engineer the removal functionality of the ZenSys tool into a java app. Just an idea..

cdjackson, 2021-02-05:
As I said, I have this coded in another application, so it’s not a case of reverse engineering anything.

andrewfg, 2021-02-05:
Ah. Ok. Where is the other application?

cdjackson, 2021-02-05:
It was not an OS project.

andrewfg, 2021-02-06:
>It was not an OS project.

Understood. But in that case we can't really mark this issue as closed. Or??

cdjackson, 2021-02-06:
I understood that this issue was about a problem with the controller being busy - that is what is reported. This is not a binding problem.

I understand that it is now changed shape, and you’re now asking for something different which I propose to close as Will Not Fix. As already mentioned above, this is not something that is super easy to implement with the way the binding works - it might work - as it might work right now. However there is nothing to stop commands being sent in the wrong order within the binding, that will still mean that it doesn’t work - the same as now - it may work, or it may not.

andrewfg, 2021-02-06:
Hmm.

ccutrer, 2021-02-06:
The original issue is that I’m unable to remove a ghost node _because_ the controller is busy. I have a very large network (150+ nodes) - the network is always going to be “busy”. The issue is that some level of business is preventing the removal of failed nodes. Would upping the retry count above 3 be sufficient? My network tends to run perfectly when all nodes are responding, but lags when a node has failed (I.e. no longer exists either because I’ve decided to not use the device, battery is dead, or it’s just temporarily unpowered etc.). If the failure of a node is the cause of the busyness, isn’t that all the more reason that removing the failed node from the network should be robust in the face of a degraded network?? I really don’t see why busyness has anything to do with marking a node failed anyway. I’m telling you it’s not there - I don’t care for you to try and talk to it to see if I’m lying. As soon as you can send _any_ command to the controller, mark it as failed on the controller. If I made a mistake, my bad, and I’ll have to reset the device too to re-include it in the network.

cdjackson, 2021-02-06:
> Hmm.

I'm not sure what that means, and I'm sure you think I'm just being difficult @andrewfg but I spent quite a lot of time on this a year or two ago with the help of some others. Somewhere there is a very long thread about it. The binding was written well before any of the standards were released, and there was therefore not a full understanding of how things worked. I should also say that ZWave is about 20 years old, and to be blunt, the protocol is pretty crap in places.

So, I apologise if you think I'm being difficult here, but I have spent a lot of time over the years looking at exactly this issue.

ccutrer, 2021-02-06:
If it’s not going to be post to get it to work, would it at least be possible to surface the timeout back to the ui? It’s very frustrating to have the set failed command seem to be accepted, but you can’t trust it now. The device may or may not show up again on the next scan.

cdjackson, 2021-02-06:
> the network is always going to be “busy”.

But really it shouldn't be. IIRC this "busy" error occurs when the controller runs out of buffers and is unable to send data on the network. This should pretty much never happen and the only time I've seen it occur is when the network is being flooded by some device that is sending very quickly. Unfortunately there's no way to find out what is happening without a sniffer.

> Would upping the retry count above 3 be sufficient? 

No - clearly not. If you've tried 3 times, and each time has 3 retries, that's 9 tries - so there's clearly something wrong.

> I really don’t see why busyness has anything to do with marking a node failed anyway.

The error you reported above looks like you are trying to remove the device from the network - this will try and communicate with the device and it will therefore clearly need to talk on the network. However, this error comes from the controller - it's not a binding error, so I can't tell you what the binding is doing. I'm just trying to explain what I think is happening. 

> I’m telling you it’s not there - I don’t care for you to try and talk to it to see if I’m lying.

Sorry - but I have never accused you of lying!

cdjackson, 2021-02-06:
> I really don’t see why busyness has anything to do with marking a node failed anyway.

Just one other point on this - I think even the marking failed will try and talk on the network as the controller tried to communicate with the device and it will only mark it failed if it doesn't get a response. So all these command DO require the network to be clear enough to send data.

ccutrer, 2021-02-06:
> Sorry - but I have never accused you of lying!

Lol sorry, I didn’t mean you personally, I meant “you” as in the binding, or the controller.

My understanding of the situation is improving, but still falling short. If the controller (and not the binding) is so busy, how come every other command gets through to the controller just fine? If the controller is so busy due to a rogue device, how come the Zensys tool has little to no problem marking the node as failed? (I _have_ seen it timeout doing this before, but very rarely).

cdjackson, 2021-02-06:
>Lol sorry, I didn’t mean you personally, I meant “you” as in the binding, or the controller.

Sorry - I misunderstood :)

>  If the controller (and not the binding) is so busy, how come every other command gets through to the controller just fine? 

That is a very good question - unfortunately one that I can't answer. There are different buffer queues for different functions in the controller, so it may be that there are more buffers kept for "normal" commands, and less for (let's say) "maintenance" commands. The controller firmware is not available, and the ZWave docs are really quite poor around the serial API so I am guessing a little about what the BUSY error means. I just know that in the past when we get this when the binding is trying to send a normal command, it is often traced to some errant device that is sending data every few milliseconds and it is cleared when this device is reset.

> If the controller is so busy due to a rogue device, how come the Zensys tool has little to no problem marking the node as failed?

Again - good question, but I can't answer. All I can say is that the controller is reporting it is busy. I don't know what else is going on at the time - all I see is the error that it is busy so I'm offering my thoughts based on this, and general experience.

ccutrer, 2021-02-06:
Is there a way to increase logging to find a rogue device like that? Or is it something only the controller itself would be able to see?

cdjackson, 2021-02-06:
You can increase logging on the binding side - the binding doc describes how to do this. If the binding is doing something wrong, or sending loads of data, then this will help. However if the issue is on the network side, as I suspect, then this will not help as you can't see what the controller is doing.

andrewfg, 2021-02-10:
> Hmm.

> I'm not sure what that means, and I'm sure you think I'm just being difficult 

Not at all. I meant no offence. So sorry if it seemed that way.

thekev, 2021-02-25:
[image]

I am having the same issue. Here's a debug with FAILed selected, it looks like the right command.

cdjackson, 2021-03-06:
@thekev you seem to have a different issue. Have you read through this issue fully - this issue is about the network being busy.

As stated above, it is also understood that there are problems removing devices, but that will not easily be fixed in the current binding.

closed 2021-03-06 by cdjackson

thekev, 2021-03-06:
The error is the same, but not the cause. The messaging could be improved to better show the controller’s response, but that’s a separate issue. I added my solution for the benefit of others who might stumble upon this.

cdjackson, 2021-03-06:
> The error is the same

Ok, I can't see the log, so I can't verify this, but it looks different to me.

> The messaging could be improved to better show the controller’s response

The log DOES show the controller response.