openhab/org.openhab.binding.zwave #1531: [Database]

bwosborne2, 2021-02-10:
Database updates seem to be working for 2,5,x but not main or 3.x branches.

The Zooz ZEN34 xml is out of date here compared to the database.

cdjackson, 2021-02-10:
Database updates are only done for the master and 2.5 branches - not the 3.x branch. I’ve not checked specifically, but it strange if they are working for one branch then I don’t really know why it would have stopped on another as it’s all running from fundamentally the same script.  I will take a look at this toward the end of next week once the shippers have been.

bwosborne2, 2021-02-10:
It appears they got politically correct and renamed `master` to `main`

cdjackson, 2021-02-10:
What do you mean? Do you mean that the repository has changed the name of the master branch? I’ve not seen any issue or notes on this in the repository so that’s rather annoying if that’s happened!!! Crazy.

I will look at this in a week or so - at the moment I’m up to my neck in shipping details with exactly 1 week before shippers arrive :(

bwosborne2, 2021-02-10:
[image]

bwosborne2, 2021-02-10:
Would it make sense for me to change the scripts and try?

cdjackson, 2021-02-10:
I don’t know who changed that, but it seems crazy to make such a fundamental change without any consultation with repository maintainers at least!

bwosborne2, 2021-02-10:
I just searched and, although new repos get that naming, I saw no indication of an automatic change. Just an indication of the crazy world today.

Roagert, 2021-02-10:
Yeah im missing a device that is listed in the opensmarthouse database. I could really use a database update. :) 
Sounds strange that they would make such a big change

bwosborne2, 2021-02-10:
I hope to take a look at it this evening.  I have a couple of ideas top at least get it more current.

Roagert, 2021-02-12:
Im actually still missing this device to be update strangely enough.
https://opensmarthouse.org/zwavedatabase/1000

bwosborne2, 2021-02-12:
> Im actually still missing this device to be update strangely enough.

Not too strange. That device was only approved on 2/9, after the export the PR above is based on.
There are still 3.x update issues and real life is adjusting priorities. Are you running version 3.x or 2.5.x ? 
This device MAY be in the latest 2.5.x snapshot.

Roagert, 2021-02-12:
> Not too strange. That device was only approved on 2/9, after the export the PR above is based on.
> There are still 3.x update issues and real life is adjusting priorities. Are you running version 3.x [...]

I see, yeah no problem. real life comes first :) 
I am running 3.x version. 
Because i couldnt see the device being "updated" with the additional parameters i have added i tried to load my own jar file to the addon folder.
But when i did that there were other issues that appeared. So im kinda missing a straight forward guide on how to make local changes in a binding so i can test out some features before populating it to the database. Do you know of such a guide somewhere ?

bwosborne2, 2021-02-12:
> Because i couldnt see the device being "updated" with the additional parameters i have added i tried to load my own jar file to the addon folder.

If you have the latest snapshot jar of the binding it should work for you. I may get a chance to look at the export process but I have not been involved with that part previously.

williamkennyAK, 2021-02-12:
Changing the structure to OH-INF does appear to work with the 3.0 binding, however, it seems sadly my device isn't 100% working yet.  It does not show status changes but that may be because I'm doing something wrong.  I'm VERY new to using zwave on openHAB and openHAB itself and only have used 3.0.  Can you use the 2.5 binding on 3.0?

bwosborne2, 2021-02-12:
Did you try deleting & re-adding the Thing after making the binding changes?

> Can you use the 2.5 binding on 3.0?

No you cannot.

If needed I can try & submit another PR to update but somebody else would need to approve & merge.

In my other PR Chris said it would likely be a week or so before he has time to look at the process. I took a quick look and would need to set up a special environment because I do not have the same server access.

bwosborne2, 2021-02-17:
Looks like @cdjackson  has the process running again.

closed 2021-02-17 by bwosborne2