openhab/org.openhab.binding.zwave #1545: Qubino 3-phase power meter: Missing support for reported value

BultenBonk, 2021-02-26:
OH: openHAB 3.0.1
z-wave binding version: 3.0.1

* Device name: Qubino 3-phase power meter
* Device version: ZMNHXD (firmware: 50.6)
* Link to [database](http://opensmarthouse.org/zwavedatabase): https://opensmarthouse.org/zwavedatabase/900

* If there is an XML being generted for the device (in the /userdata/zwave folder), please provide it. This folder is called "/srv/openhab-userdata/zwave" in OH3 at least.

I have this mail conversation with the manufacture as well:

"Every z-wave device consists of endpoints. The 3 phase smart meter has 5 endpoints, by default. The root endpoint and endpoint1 both map to the same functionality, that is why the values are the same. In a nutshell: the root endpoint and endpoint1 have the same values (total power consumption, the accumulated energy), whereas endpoint 2,3,4 map to the three phases 1,2,3. Currently, z-wave.me only shows the “consumed” (rate Type Import) values, whereas it doesn’t display the “produced” (rate Type Export) values. We’ve notified z-wave.me about this a few weeks ago. They said that it will be added to their queue of features, which will be added.​
For now there are only three gateway, platforms that can receive and show the import and the export (produced power) on the market for now that are: : ioBroker, zwavejs2mqtt, Toon gateway.

The measures are sent in the reports
COMMAND_CLASS_METER 0x32
METER_REPORT 0x02
Rate Type"
[image]

Thank you.

bolausson, 2022-07-14:
Are there any news on this - I am interested as well in the "produced" values coming from a solar panel.

akashkumardfg, 2023-09-04:
I am using a 'hack' that may work until we wait for the official binding to support this.

I get both readings from the meter - import and the export values. However, the description in the raw command is somehow lost. However, it is easy to check manually which value is which since you 'know' in the night the values are usually only import values, whereas when you have excess solar energy, it is likely import. Assuming the two values are rather far apart, I just have a simple rule to check whether the incoming value is within +-1 of the old import value. If it is indeed the case, then I assign it to import, else to the export value. Below is an example. I kept it even simpler since my export value was larger than my import.

rule "Update Total Energy"
when
	Item Smart_Electric_meter_kWh_1 changed
then
	if((Smart_Electric_meter_kWh_1.state as Number) < ((EnergyFromGrid.state as Number) + 1))
		EnergyFromGrid.postUpdate(Smart_Electric_meter_kWh_1.state as Number)
	else
		SolarEnergySent.postUpdate(Smart_Electric_meter_kWh_1.state as Number)
end