openhab/org.openhab.binding.zwave #1547: Philio PST02A Slim Multi-Sensor configuration not possible for some parameters

chris922, 2021-02-27:
The Philio PST02A has really strange configuration parameters. One parameter can configure several functionalities at the same time by setting the bits to 0/1 of one parameter.

For example a very similar device (PST02C) has the possibility to enter the plain number so all functionalities could be changed by calculating the number based on the bits that should be set.
For the PST02A this is only possible for some parameters, but not for the following ones:
* 2: Basic Set Level
* 3: PIR sensitivity
* 4: Light threshold
* 5: Operation mode
* 7: Customer function

I had a look at the XML associated to this device in this repository and maybe just a `<limitToOptions>false</limitToOptions>` is missing for these parameters. E.g. here: 
https://github.com/openhab/org.openhab.binding.zwave/blob/b9433110776973e315a001c9966dc53aa35fa308/src/main/resources/OH-INF/thing/philio/pst02a_0_0.xml#L128-L139

* OH2 Version: 3.0.0
* Binding Version: 3.0.0
* Device name: Philio PST02A Slim Multi-Sensor
* Device version: ?
* Link to [database](http://opensmarthouse.org/zwavedatabase): https://opensmarthouse.org/zwavedatabase/168
* If the device is not in the database, provide a link to the manual.
* If there is an XML being generted for the device (in the /userdata/zwave folder), please provide it: [network_da20c288__node_66.xml.txt](https://github.com/openhab/org.openhab.binding.zwave/files/6055079/network_da20c288__node_66.xml.txt)

chris922, 2021-03-02:
I tried to fix this by manually changing the XML, see #1550

cdjackson, 2021-03-02:
Please update the database at opensmarthouse.org. Updates directly into the binding will not be accepted as they will just be overwritten within a few days from the database export.

chris922, 2021-03-02:
Aah thanks for the fast response!

I thought there will be a different place to do the updates seeing all your database imports. How can I update it on opensmarthouse.org? I just opened a ticket to get editor privileges so that I can update the entry there.

chris922, 2021-03-06:
Fixed by latest database update within #1552

closed 2021-03-06 by chris922