openhab/org.openhab.binding.zwave #1553: Enhancement Request - Command_Class_Schedule v1 - workaround via get/set payload string?

johnrowland, 2021-03-05:
Would it be possible to provide a "raw" implementation for Command_Class_Schedule v1?
I'm imagining that the Thing has a Channel (e.g. "raw_schedule") that can be used for transferring a raw Payload between a linked text/string Item and the physical device...

This partial implementation of the class in the binding would then allow interested users to get / set whatever they needed (probably via rules and other items) without the binding developers having to worry about how to integrate with the core UI.

Perhaps a bit messy, but some implementation has to be better than no implementation!

Has been mentioned on the forum before (e.g. https://community.openhab.org/t/danfoss-devolo-room-sensor-command-class-schedule-v1-unknown-command-5/81593)

My specific use-case is that I have a Secure SCS-317 thermostat that reports the current setpoint via this Schedule class (command 5) and without support for this class I am unable to use openHAB to monitor target setpoint changes that occur as a result of any physical interaction with the device (or due to the schedule). I am not happy with idea of leaving my heating system under open as opposed to closed control...!

OH version: 3.0.1
Binding: 3.0.1

cdjackson, 2021-03-06:
This is not the sort of thing I'm keen on and would propose to close this. It is open to error, which may cause problems with devices, and certainly cause issues with support. It will also take quite some effort to add and is something that is not really of a lot of interest to the wider community. I know people will challenge that statement, and I appreciate that some "power users" would like to be able to manipulate everything, but 99.9% of users do not want to do this.

johnrowland, 2021-03-07:
Following discussion on the forum, I agree with the proposal to close the issue.

closed 2021-03-07 by johnrowland