openhab/org.openhab.binding.zwave #1598: Default value of backlight configuration is wrong when Eurotronic Spirit reconnects after battery swap

jimon, 2021-04-29:
* openHAB: 3.0.1
* Binding Version: 3.0.1
* Device name: Eurotronic Spirit
* Device version: 0.16
* Link to [database](https://opensmarthouse.org/zwavedatabase/710)

Steps to reproduce:
- Connect Eurotronic Spirit thermostat to OH
- Set backlight to `disabled`, press save, confirm that it is disabled
- Remove batteries from thermostat
- Put batteries back into thermostat, press button on thermostat to proceed with "installation on the radiator"
- Wait until thermostat reconnects to OH, e.g. settings like setpoint are propagated
- Observe that OH reports backlight as `disabled`

Expected outcome: Backlight is disabled on the device.
Observed outcome: Backlight is enabled on the device (at least for first 1-2 minutes, haven't checked further), but OH reports backlight as disabled.

Current workaround: Set backlight to enabled, press save, wait a bit, set backlight to disabled, press save, observe backlight being disabled on the device.

I have suspicion that this is true for others configuration values as well, but it's harder to prove what their values are as the device doesn't report them. I'll try to capture the log if needed.

Should this also be fixed as part of healing?

cdjackson, 2021-04-29:
I'm not sure what the binding can do about this - it seems to be a device issue. Certainly this has nothing to do with healing.

Your workaround is the correct solution - if this is the default from the manufacturer then the binding will not change this.

closed 2021-04-29 by cdjackson

jimon, 2021-04-29:
@cdjackson sorry I don't have a full context of how exactly data between OH<->binding<->device propagates, but I still believe there is desynchronization happens: OH reports backlight disabled, but physical device has glowing backlight.
I'm not sure if it's reasonable to expect to device to keep track of configuration after the power loss, and changing batteries if an often enough procedure.

Is there maybe some "connected"/"woke up" Z-Wave frame that could be used to detect powerloss scenarios and trigger a full re-configuration of the device?

cdjackson, 2021-04-29:
No - this is not possible. If the device changes state for some reason, then the binding will not know and yes, there will be a synchronisation issue. There's nothing that can really be done to resolve this short of reading all the parameters periodically, and that would take a lot of time in large networks, and significantly impact battery life on any battery devices.