openhab/org.openhab.binding.zwave #1611: ABUS Z-Wave [Unknown Thing Type]

jbaudisch, 2021-05-27:
* OH2 Version: openHABian v1.6.5
* Binding Version (if SNAPSHOT, provide compile time as found on the filename in the console): 3.0.2
* Device name: ABUS Z-Wave button (Art. no. SHBE10000)
* Link to [database](http://opensmarthouse.org/zwavedatabase): https://www.abus.com/eng/content/view/full/62729

The device is shown as online but the thing type is unknown and no channels are provided and it says:
"The device is not in the database. [...] the database must be updated and you should raise an issue to get this addressed."

Hope you can add this :)

Thank you.

jbaudisch, 2022-01-23:
Anything new here? :)

cdjackson, 2022-01-23:
The database is a community effort, so please feel free to add devices. You can find the instructions here -:

https://opensmarthouse.org/blog/zwave_device_database_guide