openhab/org.openhab.binding.zwave #1627: OH3 - battery powered devices never complete interview

kpanchen, 2021-06-29:
OH3 Version: 3.1.0
Binding Version: 3.1.0

If the problem is associated with a device, please provide the following -:

* Device name: Aeotek TriSensor or NanoMote

Battery powered devices never completed interview / never created serialisation XML file. Basically after inclusion and initial interview both devices progresses in stages but after certain stage waking up devices just does nothing. Thing configuration looks fine but polling never started as initialization is not completed. After 24 hours NanoMote shows "Node initialising: STATIC_VALUES" but wake-up will not advance.
[image]
[image]

[openhab.log](https://github.com/openhab/org.openhab.binding.zwave/files/6737020/openhab.log)

cdjackson, 2021-06-29:
This can take some time. There is nothing especially wrong in the log - the interview progresses ok. You are using security so this will double or tripple the amount of traffic and slow things down significantly, but there doesn't appear to be anything wrong as far as I can tell.

kpanchen, 2021-06-30:
> This can take some time. There is nothing especially wrong in the log - the interview progresses ok. You are using security so this will double or tripple the amount of traffic and slow things down significantly, but there doesn't appear to be anything wrong as far as I can tell.

I don't really need security for sensor, I've tried without and result is the same. Remote is to control some vital functions like garage opener, need to be secure.

The thing is that on OH2 I had 5 battery devices included without any issues, NanoMote was the last one. With OH3 none of them finish inclusion. I agree inclusion proceeds fine till, I think, this point:

`2021-06-29 09:26:05.884 [DEBUG] [essage.ApplicationUpdateMessageClass] - NODE 2: Application update request. Node information received. Transaction null
2021-06-29 09:26:05.884 [DEBUG] [tocol.commandclass.ZWaveCommandClass] - NODE 2: Unsupported command class COMMAND_CLASS_SECURITY_2
2021-06-29 09:26:05.884 [DEBUG] [tocol.commandclass.ZWaveCommandClass] - NODE 2: Unsupported command class COMMAND_CLASS_TRANSPORT_SERVICE
2021-06-29 09:26:05.885 [DEBUG] [tocol.commandclass.ZWaveCommandClass] - NODE 2: Unsupported command class COMMAND_CLASS_SUPERVISION
2021-06-29 09:26:05.885 [DEBUG] [essage.ApplicationUpdateMessageClass] - NODE 2: Application update - no transaction.
2021-06-29 09:26:05.886 [DEBUG] [nal.protocol.ZWaveTransactionManager] - ZWaveReceiveThread queue empty
2021-06-29 09:26:05.886 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Transaction SendNextMessage 1 out at start. Holdoff false.
2021-06-29 09:26:07.441 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 05 00 13 1C 01 F4 
2021-06-29 09:26:07.442 [DEBUG] [nal.protocol.ZWaveTransactionManager] - processReceiveMessage input 0<>128 : Message: class=SendData[19], type=Request[0], dest=1, callback=28, payload=1C 01 
2021-06-29 09:26:07.443 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Received msg (0): Message: class=SendData[19], type=Request[0], dest=1, callback=28, payload=1C 01 
2021-06-29 09:26:07.443 [DEBUG] [nal.protocol.ZWaveTransactionManager] - lastTransaction TID 38: [WAIT_REQUEST] priority=High, requiresResponse=true, callback: 28
2021-06-29 09:26:07.443 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Checking outstanding transactions: 1
2021-06-29 09:26:07.444 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Last transaction: TID 38: [WAIT_REQUEST] priority=High, requiresResponse=true, callback: 28
2021-06-29 09:26:07.444 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Checking TID 38: (Callback 28)
2021-06-29 09:26:07.444 [DEBUG] [.serialmessage.ZWaveCommandProcessor] - Callback match!
2021-06-29 09:26:07.445 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Correlated to TID 38: callback 28
2021-06-29 09:26:07.445 [DEBUG] [ve.internal.protocol.ZWaveController] - Incoming Message: Message: class=SendData[19], type=Request[0], dest=1, callback=28, payload=1C 01 
2021-06-29 09:26:07.446 [DEBUG] [l.serialmessage.SendDataMessageClass] - NODE 2: SendData Request. CallBack ID = 28, Status = Transmission complete, no ACK received(1)
2021-06-29 09:26:07.446 [DEBUG] [e.internal.protocol.ZWaveTransaction] - TID 38: Transaction CANCELLED
2021-06-29 09:26:07.447 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 2: Got an event from Z-Wave network: ZWaveTransactionCompletedEvent
2021-06-29 09:26:07.447 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 2: Transaction failed waiting for REQUEST, assume sleeping device.
2021-06-29 09:26:07.448 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 2: CANCEL while sending message. Requeueing - 2 attempts left!
2021-06-29 09:26:07.448 [DEBUG] [e.internal.protocol.ZWaveTransaction] - TID 38: Transaction RESET with 2 retries remaining.
2021-06-29 09:26:07.448 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 2: Adding to device queue
2021-06-29 09:26:07.449 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 2: Added 38 to queue - size 2
2021-06-29 09:26:07.449 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Transaction SendNextMessage 0 out at start. Holdoff false.
2021-06-29 09:26:07.450 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 2: TID 38: Transaction not completed
2021-06-29 09:26:07.450 [DEBUG] [nal.protocol.ZWaveTransactionManager] - ZWaveReceiveThread queue empty
2021-06-29 09:26:07.450 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Transaction SendNextMessage 0 out at start. Holdoff false.
2021-06-29 09:26:14.011 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 2: Polling...
2021-06-29 09:26:14.012 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 2: Polling deferred until initialisation complete`

And it is pretty much complete by this time, not sure what else it suppose to do but no matter how many times device wake, it is doing similar thing. Another interesting observation that I can enable "always wake-up" mode but that wouldn't change anything, binding doesn't see that device alive and there's no way to push it to do re-discovery.

Anyway ideas? What to do next, at this point I'm willing to provide any help needed.

cdjackson, 2021-06-30:
> The thing is that on OH2 I had 5 battery devices included without any issue

There is pretty much no difference between OH2 and OH3 ZWave binding - so if it worked before, it should work now (depending on exactly what version of OH2 you had - if it was OH2.0 then there will of course be some difference as that was a few years old).

>  I can enable "always wake-up" mode but that wouldn't change anything,

Correct - this will not change anything.

> Anyway ideas?

Not really  - sorry. As above, there has been no significant change to the binding for quite a while, and I think it works fine for others so I'm not really sure what to suggest.

kpanchen, 2021-06-30:
> There is pretty much no difference between OH2 and OH3 ZWave binding - so if it worked before, it should work now (depending on exactly what version of OH2 you had - if it was OH2.0 then there will [...]

Latest, 2.5.4 if I remember correctly.

> 
> > I can enable "always wake-up" mode but that wouldn't change anything,
> 
> Correct - this will not change anything.

Maybe "push" mode can be added to the bundle? Not sure why "re-initialization" is disabled while it is not done at least once, the whole point of that is to fix bad one.

> 
> > Anyway ideas?
> 
> Not really - sorry. As above, there has been no significant change to the binding for quite a while, and I think it works fine for others so I'm not really sure what to suggest.

Still something is different, my install is a fresh brand new one, nothing is used from 2.5, controller hardware reset, inclusion, exclusion work fine. Just interview, not sure how it supposed to progress when NIF sent. I'm trying again with only one sensor just to get a cleaner log.

cdjackson, 2021-06-30:
> Latest, 2.5.4 if I remember correctly.

Then the binding is almost exactly the same as OH3.

> Maybe "push" mode can be added to the bundle?

I don't know what you mean by "push mode"? Is this something to do with ZWave or something else?

kpanchen, 2021-06-30:
> 
> > Maybe "push" mode can be added to the bundle?
> 
> I don't know what you mean by "push mode"? Is this something to do with ZWave or something else?

As I understand now the algorithm to wait for complete initiation and only after that to allow "re-initialization", however a lot of battery powered devices have "always awake" mode but no way to use it. If "re-initialization" allowed one can wake device and call for re-init similar to mains powered devices. 

Back to my issues, after another play I've got another piece of log:

`2021-06-30 00:55:45.956 [DEBUG] [alization.ZWaveNodeInitStageAdvancer] - NODE 2: Node Init response (0) org.openhab.binding.zwave.internal.protocol.ZWaveTransactionResponse@486d7b27
2021-06-30 00:55:45.956 [DEBUG] [alization.ZWaveNodeInitStageAdvancer] - NODE 2: Node Init transaction completed with response COMPLETE
2021-06-30 00:55:45.956 [DEBUG] [alization.ZWaveNodeInitStageAdvancer] - NODE 2: Node advancer - advancing to REQUEST_NIF
2021-06-30 00:55:45.956 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 2: Got an event from Z-Wave network: ZWaveInitializationStateEvent
2021-06-30 00:55:45.957 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 2: sendTransaction org.openhab.binding.zwave.internal.protocol.ZWaveSerialPayload@d6ffc4e
2021-06-30 00:55:45.957 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 09 01 41 53 9C 01 04 18 01 65 
2021-06-30 00:55:45.957 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 2: Bump transaction 326 priority from Controller to Immediate
2021-06-30 00:55:45.962 [DEBUG] [ing.zwave.handler.ZWaveSerialHandler] - Message SENT
2021-06-30 00:55:45.963 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Transaction SendNextMessage started: TID 325: [WAIT_RESPONSE] priority=Controller, requiresResponse=true, callback: 0
2021-06-30 00:55:45.963 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Received msg (0): Message: class=null[0], type=ACK[2], dest=255, callback=0, payload=
2021-06-30 00:55:45.963 [DEBUG] [nal.protocol.ZWaveTransactionManager] - lastTransaction TID 325: [WAIT_RESPONSE] priority=Controller, requiresResponse=true, callback: 0
2021-06-30 00:55:45.963 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Received msg: ACK
2021-06-30 00:55:45.963 [DEBUG] [nal.protocol.ZWaveTransactionManager] - ZWaveReceiveThread queue empty
2021-06-30 00:55:45.963 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Transaction SendNextMessage 1 out at start. Holdoff false.
2021-06-30 00:55:45.963 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 2: Adding to device queue
2021-06-30 00:55:45.963 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 2: Added 326 to queue - size 1
2021-06-30 00:55:45.963 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Transaction SendNextMessage 1 out at start. Holdoff false.
2021-06-30 00:55:45.964 [DEBUG] [nal.protocol.ZWaveTransactionManager] - processReceiveMessage input 0<>128 : Message: class=IdentifyNode[65], type=Response[1], dest=255, callback=0, payload=53 9C 01 04 18 01 
2021-06-30 00:55:45.964 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Received msg (0): Message: class=IdentifyNode[65], type=Response[1], dest=255, callback=0, payload=53 9C 01 04 18 01 
2021-06-30 00:55:45.964 [DEBUG] [nal.protocol.ZWaveTransactionManager] - lastTransaction TID 325: [WAIT_RESPONSE] priority=Controller, requiresResponse=true, callback: 0
2021-06-30 00:55:45.964 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Checking outstanding transactions: 1
2021-06-30 00:55:45.964 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Last transaction: TID 325: [WAIT_RESPONSE] priority=Controller, requiresResponse=true, callback: 0
2021-06-30 00:55:45.964 [DEBUG] [ve.internal.protocol.ZWaveController] - Incoming Message: Message: class=IdentifyNode[65], type=Response[1], dest=255, callback=0, payload=53 9C 01 04 18 01 ` 

And after couple of wake-ups/send NIF:
`2021-06-30 00:58:15.725 [DEBUG] [ve.internal.protocol.ZWaveController] - Incoming Message: Message: class=ApplicationUpdate[73], type=Request[0], dest=2, callback=132, payload=84 02 08 04 07 01 5E 98 9F 55 6C 
2021-06-30 00:58:15.725 [DEBUG] [essage.ApplicationUpdateMessageClass] - NODE 2: Application update request. Node information received. Transaction null
2021-06-30 00:58:15.725 [DEBUG] [tocol.commandclass.ZWaveCommandClass] - NODE 2: Creating new instance of command class COMMAND_CLASS_ZWAVEPLUS_INFO
2021-06-30 00:58:15.726 [DEBUG] [tocol.commandclass.ZWaveCommandClass] - NODE 2: Command class COMMAND_CLASS_ZWAVEPLUS_INFO, endpoint 0 created
2021-06-30 00:58:15.726 [DEBUG] [essage.ApplicationUpdateMessageClass] - NODE 2: Application update is adding command class COMMAND_CLASS_ZWAVEPLUS_INFO.
2021-06-30 00:58:15.726 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 2: Adding command class COMMAND_CLASS_ZWAVEPLUS_INFO to the list of supported command classes.
2021-06-30 00:58:15.727 [DEBUG] [tocol.commandclass.ZWaveCommandClass] - NODE 2: Creating new instance of command class COMMAND_CLASS_SECURITY
2021-06-30 00:58:15.727 [DEBUG] [tocol.commandclass.ZWaveCommandClass] - NODE 2: Command class COMMAND_CLASS_SECURITY, endpoint 0 created
2021-06-30 00:58:15.727 [DEBUG] [essage.ApplicationUpdateMessageClass] - NODE 2: Application update is adding command class COMMAND_CLASS_SECURITY.
[... cut ...]
2021-06-30 01:01:54.937 [DEBUG] [nal.protocol.ZWaveTransactionManager] - processReceiveMessage input 0<>128 : Message: class=ApplicationUpdate[73], type=Request[0], dest=2, callback=132, payload=84 02 08 04 07 01 5E 98 9F 55 6C 
2021-06-30 01:01:54.938 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Received msg (0): Message: class=ApplicationUpdate[73], type=Request[0], dest=2, callback=132, payload=84 02 08 04 07 01 5E 98 9F 55 6C `

Binding waiting for NIF but such NIF would come only if I manually send it which binding recognize as wake up event but not NIF or if I wait till device wake up ... I'm guessing here.

cdjackson, 2021-06-30:
> As I understand now the algorithm to wait for complete initiation and only after that to allow "re-initialization"

Again, I don't really understand what you refer to. What is "initialisation" here? There is a lot that happens here, but I don't know what you are exactly referring to and still don't really know what "push mode" is. The binding has to request all the data - there is no ability for the device to send the required information by itself if that's what you mean by "push mode".

> Binding waiting for NIF but such NIF would come only if I manually send it which binding recognize as wake up event but not NIF or if I wait till device wake up

You always need to manually wake up the device initially. Once the initialisation of the wakeup CC has been done, then the device will wake up by itself.

kpanchen, 2021-06-30:
> What is "initialisation" here? - Interview, which cannot be repeated unless first one is done. So if first one is not done there's no way to request to perform "the interview" again. Might be deleting thing or restarting binding will do that, not sure. Definitely no way to do it per CC. By "push" I meant not to wait till device wake up but try to ask for CC's, if device is awake it will simply provide data. (That's the way Z-way works). In OH case I can keep device awake and it will wait for commands from binding but binding doesn't know that and not doing anything.

> You always need to manually wake up the device initially. Once the initialisation of the wakeup CC has been done, then the device will wake up by itself.

So why binding is not recognising NIF when device manually awoken? It does see "wake event" but not NIF.

cdjackson, 2021-06-30:
The binding will treat a NIF exactly the same as a wake up notification. 

The binding DOES request data if the device is awake - so I guess it is doing what you call ‘push’ mode already. If things slow down though then it will go to sleep. That is how ZWave works.

cdjackson, 2021-06-30:
> So why binding is not recognising NIF when device manually awoken?

From your log, you can see that it DOES recognise the NIF and it treats this as a wakeup. Again - this is exactly the same as OH2 - nothing has changed.

```
2021-06-29 09:25:57.816 [DEBUG] [essage.ApplicationUpdateMessageClass] - NODE 2: Application update request. Node information received. Transaction null
...
2021-06-29 09:25:58.069 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 2: Is awake with 1 messages in the queue
```

kpanchen, 2021-06-30:
I might found what is the difference and why it doesn't work: the difference is that now I've decided to include everything in secure mode, guess it should work but who knows as I've decided to go back to regular inclusion. But with both types I noticed that on battery devices Wake Up interval and Wake Up node set to 0, where it should be 1800 and 1 respectively. This explains why device never send wake-up NIF to controller. Strangely enough NanoMote in non-secure mode set both correctly from very beginning. Mote and one sensor already produced XML and the other one is still in interview mode but now I'll just let it run and wake up by itself. 
Anyway, thank you for your support!! (Hope you enjoying New Zealand?) If you ever need something to test I'm always willing.

closed 2021-06-30 by kpanchen