openhab/org.openhab.binding.zwave #1630: Telldus Door sensor missing from database

empperi, 2021-07-08:
Bought four Telldus Door sensors and it seems like those are not currently in the database and would have to be added. It's a very simple battery operated door sensor with tamper switch.

[image]

These devices do not appear into inbox. Log snippet:

```
2021-07-08 12:49:29.079 [WARN ] [wave.discovery.ZWaveDiscoveryService] - NODE 7: Device discovery could not resolve to a thingType! Manufacturer data not known.
2021-07-08 12:49:29.087 [INFO ] [g.discovery.internal.PersistentInbox] - Added new thing 'zwave:device:f399601ded:node7' to inbox.
```

I'm assuming that since they are not in the database they won't show up even if the logs say so. I managed to add them via scanning but they appear as unknown devices. I activated debug logging and opened and closed one sensor couple of times. Log from this:

```
2021-07-08 13:20:43.531 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 10 00 04 00 07 0A 71 05 00 00 00 FF 06 16 00 00 7D
2021-07-08 13:20:43.536 [DEBUG] [nal.protocol.ZWaveTransactionManager] - processReceiveMessage input 0<>128 : Message: class=ApplicationCommandHandler[4], type=Request[0], dest=7, callback=0, payload=00 07 0A 71 05 00 00 00 FF 06 16 00 00
2021-07-08 13:20:43.538 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Received msg (0): Message: class=ApplicationCommandHandler[4], type=Request[0], dest=7, callback=0, payload=00 07 0A 71 05 00 00 00 FF 06 16 00 00
2021-07-08 13:20:43.540 [DEBUG] [nal.protocol.ZWaveTransactionManager] - lastTransaction null
2021-07-08 13:20:43.541 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 7: Application Command Request (ALIVE:REQUEST_NIF)
2021-07-08 13:20:43.542 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 7: Incoming command class COMMAND_CLASS_ALARM, endpoint 0
2021-07-08 13:20:43.542 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 7: SECURITY not supported
2021-07-08 13:20:43.543 [DEBUG] [tocol.commandclass.ZWaveCommandClass] - NODE 7: Received COMMAND_CLASS_ALARM V0 NOTIFICATION_REPORT
2021-07-08 13:20:43.545 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 7: NOTIFICATION report - 0 = 0, event=22, status=255, plen=0
2021-07-08 13:20:43.546 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 7: Alarm Type = ACCESS_CONTROL (0)
2021-07-08 13:20:43.547 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 7: Got an event from Z-Wave network: ZWaveAlarmValueEvent
2021-07-08 13:20:43.548 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 7: Got a value event from Z-Wave network, endpoint=0, command class=COMMAND_CLASS_ALARM, value=255
2021-07-08 13:20:43.549 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 7: Commands processed 1.
2021-07-08 13:20:43.550 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 7: Checking command org.openhab.binding.zwave.internal.protocol.ZWaveCommandClassPayload@5441fd70.
2021-07-08 13:20:43.551 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Transaction completed - outstandingTransactions 0
2021-07-08 13:20:43.552 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Transaction completed - outstandingTransactions 0
2021-07-08 13:20:43.553 [DEBUG] [nal.protocol.ZWaveTransactionManager] - ZWaveReceiveThread queue empty
2021-07-08 13:20:43.555 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Transaction SendNextMessage 0 out at start. Holdoff false.
2021-07-08 13:20:44.439 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 09 00 04 00 07 03 80 03 64 11
2021-07-08 13:20:44.442 [DEBUG] [nal.protocol.ZWaveTransactionManager] - processReceiveMessage input 0<>128 : Message: class=ApplicationCommandHandler[4], type=Request[0], dest=7, callback=0, payload=00 07 03 80 03 64
2021-07-08 13:20:44.445 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Received msg (0): Message: class=ApplicationCommandHandler[4], type=Request[0], dest=7, callback=0, payload=00 07 03 80 03 64
2021-07-08 13:20:44.448 [DEBUG] [nal.protocol.ZWaveTransactionManager] - lastTransaction null
2021-07-08 13:20:44.450 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 7: Application Command Request (ALIVE:REQUEST_NIF)
2021-07-08 13:20:44.451 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 7: Incoming command class COMMAND_CLASS_BATTERY, endpoint 0
2021-07-08 13:20:44.452 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 7: SECURITY not supported
2021-07-08 13:20:44.453 [DEBUG] [tocol.commandclass.ZWaveCommandClass] - NODE 7: Received COMMAND_CLASS_BATTERY V0 BATTERY_REPORT
2021-07-08 13:20:44.454 [DEBUG] [ommandclass.ZWaveBatteryCommandClass] - NODE 7: Battery report value = 100
2021-07-08 13:20:44.455 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 7: Got an event from Z-Wave network: ZWaveCommandClassValueEvent
2021-07-08 13:20:44.457 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 7: Got a value event from Z-Wave network, endpoint=0, command class=COMMAND_CLASS_BATTERY, value=100
2021-07-08 13:20:44.458 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 7: Commands processed 1.
2021-07-08 13:20:44.459 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 7: Checking command org.openhab.binding.zwave.internal.protocol.ZWaveCommandClassPayload@436a162a.
2021-07-08 13:20:44.460 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Transaction completed - outstandingTransactions 0
2021-07-08 13:20:44.461 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Transaction completed - outstandingTransactions 0
2021-07-08 13:20:44.462 [DEBUG] [nal.protocol.ZWaveTransactionManager] - ZWaveReceiveThread queue empty
2021-07-08 13:20:44.463 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Transaction SendNextMessage 0 out at start. Holdoff false.
2021-07-08 13:20:52.278 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 10 00 04 00 07 0A 71 05 00 00 00 FF 06 17 00 00 7C
2021-07-08 13:20:52.281 [DEBUG] [nal.protocol.ZWaveTransactionManager] - processReceiveMessage input 0<>128 : Message: class=ApplicationCommandHandler[4], type=Request[0], dest=7, callback=0, payload=00 07 0A 71 05 00 00 00 FF 06 17 00 00
2021-07-08 13:20:52.283 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Received msg (0): Message: class=ApplicationCommandHandler[4], type=Request[0], dest=7, callback=0, payload=00 07 0A 71 05 00 00 00 FF 06 17 00 00
2021-07-08 13:20:52.285 [DEBUG] [nal.protocol.ZWaveTransactionManager] - lastTransaction null
2021-07-08 13:20:52.286 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 7: Application Command Request (ALIVE:REQUEST_NIF)
2021-07-08 13:20:52.288 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 7: Incoming command class COMMAND_CLASS_ALARM, endpoint 0
2021-07-08 13:20:52.289 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 7: SECURITY not supported
2021-07-08 13:20:52.290 [DEBUG] [tocol.commandclass.ZWaveCommandClass] - NODE 7: Received COMMAND_CLASS_ALARM V0 NOTIFICATION_REPORT
2021-07-08 13:20:52.291 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 7: NOTIFICATION report - 0 = 0, event=23, status=255, plen=0
2021-07-08 13:20:52.293 [DEBUG] [.commandclass.ZWaveAlarmCommandClass] - NODE 7: Alarm Type = ACCESS_CONTROL (0)
2021-07-08 13:20:52.294 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 7: Got an event from Z-Wave network: ZWaveAlarmValueEvent
2021-07-08 13:20:52.295 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 7: Got a value event from Z-Wave network, endpoint=0, command class=COMMAND_CLASS_ALARM, value=255
2021-07-08 13:20:52.296 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 7: Commands processed 1.
2021-07-08 13:20:52.297 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 7: Checking command org.openhab.binding.zwave.internal.protocol.ZWaveCommandClassPayload@7aacc4f3.
2021-07-08 13:20:52.298 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Transaction completed - outstandingTransactions 0
2021-07-08 13:20:52.299 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Transaction completed - outstandingTransactions 0
2021-07-08 13:20:52.301 [DEBUG] [nal.protocol.ZWaveTransactionManager] - ZWaveReceiveThread queue empty
2021-07-08 13:20:52.302 [DEBUG] [nal.protocol.ZWaveTransactionManager] - Transaction SendNextMessage 0 out at start. Holdoff false.
2021-07-08 13:20:53.187 [DEBUG] [WaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 09 00 04 00 07 03 80 03 64 11
[... cut ...]
        </supportedCommandClasses>
      </endPoint>
    </entry>
  </endpoints>
  <nodeNeighbors>
    <int>1</int>
    <int>2</int>
    <int>5</int>
    <int>8</int>
    <int>9</int>
  </nodeNeighbors>
  <lastReceived>2020-02-16 18:04:28.29 UTC</lastReceived>
</node>
```

As a side note, I need these to work real soon and would love to add these to database by hand (I'm an experienced programmer) but would need some basic guidelines how to do this. If I would get them to work then I could create a PR for the update.

cdjackson, 2021-07-08:
This needs to be added to the database. Please see the below link -:

https://opensmarthouse.org/blog/zwave_device_database_guide

empperi, 2021-07-30:
Just realized the XML file I had pasted was **not** the correct one. It was a stale piece of data and as it is now, it looks like Openhab is not generating any XML files into userdata folder. Any help on this is appreciated.

cdjackson, 2021-07-30:
I have already answered on the forum - please avoid posting the same question in two places.