openhab/org.openhab.binding.zwave #1637: Z-Wave Serial Controller - Change zwave_neighbours (2,3,5)

jonas-f21, 2021-07-25:
Z-Wave Serial Controller
Actually no change of the "zwave_neighbours" is possible.

Change zwave_neighbours

Default seems to be channel 2, 3 and 5

cdjackson, 2021-07-25:
Please describe your issue as I don't really understand this at all.

jonas-f21, 2021-07-25:
> Please describe your issue as I don't really understand this at all.

I would like to change the ZWAVE neighbours ID as my Network is only connected to one possible "neighbour" --> Channel 5 (see pictures attached).

If this ID's can be changed, I could add channel 10 or 11 as neighbour which provides a more stable network (Network Wide Inclusion)

[image]
[image]

cdjackson, 2021-07-25:
> I would like to change the ZWAVE neighbours

This is not possible. The neighbours are simply the devices that other devices can hear. You can't change this other than to physically move devices.

jonas-f21, 2021-07-25:
> > I would like to change the ZWAVE neighbours
> 
> This is not possible. The neighbours are simply the devices that other devices can hear. You can't change this other than to physically move devices.

Physically Node ID 8, 10, 15, 16 and 17 are much closer to the Controller (ID 1), but looking at the network map, the controller always connects to node ID 5 which is one of the "neigbours" ID's.

So I dont understand why the controller (ID 1) does not mesh around the other ID's.

cdjackson, 2021-07-25:
I don't know - this is completely controlled by the controller - the binding has no influence over this. Sorry.

closed 2021-07-25 by cdjackson