openhab/org.openhab.binding.zwave #1641: Feature request: Consistent serialization of XML to avoid unnecessary changes

ssalonen, 2021-07-31:
Using OH 3.1.0.

Similar to https://github.com/openhab/openhab-core/issues/2436 (and related community discussion https://community.openhab.org/t/storing-configuration-userdata-in-git/124727/) which was discussing jsondb.

Currently the zwave bundle serializes ZWave node information in XML files. The way the serialization has been implemented, we are not guaranteed to get very stable ordering of entries. In my use case I see this clearly as I use git as a backup method for the `userdata` folder of openHAB -- as the XML ordering changes, the diffs are very hard to read, and real changes get lost in the noise. Specifically, entries of `<endpoints class="concurrent-hash-map">` have a quite unpredictable serialization order.

We could serialize the hashmap in alphabetical order to avoid the issue.

I can contribute PR to this one but wanted to open up this issue first to get feedback on the suggestion.

cdjackson, 2021-07-31:
Sorry, but this isn't really workable. There is a lot of infrastructure that relies on the XML files, and changing to a different format such as json would take an awful lot of work that I can't support.

If this is a problem for your backup, you could elect not to back up these files as they are only internal to the binding, and will be regenerated if they are lost.

ssalonen, 2021-07-31:
Thanks @cdjackson for the speedy response.

I am not suggesting to migrate to json but instead sort entries in XML (keeping it backwards compatible).

Would this work for you?

cdjackson, 2021-07-31:
Yes, the order doesn’t matter to me, so if you can modify this so that it serialises in a fixed order, so long as it doesn’t cause some other issue, then I’m fine with that.

PR #1642 (2021-08-20): Map and Set to have consistent XML serialization order

Resolves #1641
commit 67fec3f7 (2021-08-20): Map and Set to have consistent XML serialization order (#1642)

Map and Set to have consistent XML serialization order
Resolves #1641
closed 2021-08-20 by cdjackson