openhab/org.openhab.binding.zwave #1644: Openhab 3.1 binding Zwave Aeotec Multisensor 6 configuration

cambronf, 2021-08-07:
Changing the configuration is not possible due to incorrect handling of parameter 8: Timeout after wake up. Default it shows -1, but **any** value is rejected, no matter the value of it (it says value should be higher than 8). As parameter 8 can not be set, it blocks the change of any other parameter.
(I tried several codes in the range 8 to 255, but all are rejected. Both in the form page or code page)

    8: Timeout after wake up
    **Select a value which is not lower than 8.**
    Set the timeout of awake after the Wake Up CC is sent out…

Sensor info:
zwave_class_basic BASIC_TYPE_ROUTING_SLAVE
zwave_class_generic GENERIC_TYPE_SENSOR_MULTILEVEL
zwave_neighbours 4,5
modelId ZW100
zwave_version **1.15**
zwave_plus_devicetype NODE_TYPE_ZWAVEPLUS_NODE
versionMin 1.10
vendor AEON Labs
defaultAssociations 1
zwave_routing true
zwave_beaming true
zwave_secure false
zwave_class_specific SPECIFIC_TYPE_ROUTING_SENSOR_MULTILEVEL
zwave_devicetype 2
zwave_frequent false
zwave_listening true
manufacturerId 0086
manufacturerRef 0002:0064,0102:0064,0202:0064,1A02:0064
dbReference 778
zwave_deviceid 100
zwave_nodeid 3
zwave_lastheal 2021-08-04T07:20:00Z
zwave_plus_roletype UNKNOWN_FF
zwave_manufacturer 134

Openhab version: 3.1.0 stable

apella12, 2021-08-10:
Not saying this is not an issue, but 
[image]
did you try using the Developer Tool API to put directly?

Smudo1291, 2022-07-21:
Not sure about OH 3.1.0, but I had no issues changing parameters of Aeotec Multisensors 6 in OH 3.2 or 3.3. Param 8 is set to 15 in my config. I would upgrade to a later OH version and check it again.