openhab/org.openhab.binding.zwave #1648: Feature Request: Expose states of devices for metrics polling

utdrmac, 2021-08-08:
This is a feature request as recommended by https://github.com/openhab/openhab-addons/issues/11088

The desire is to be able to use prometheus to scrape the current status of all ZWave devices and channels. Many devices have such metrics, for example, a motion sensor with temperature, the open/close state of a door, or the kWh measurement of an electrical plug/sensor.

Having this information pollable by an external agent, like prometheus, would allow users to graph various aspects of their home.

effrafax, 2022-10-17:
Would like to see this too.

scotthraban, 2023-06-24:
Please comment on [my post here](https://community.openhab.org/t/proposal-micrometer-persistence-service/147306) if you think my proposal would give you what you are looking for.