openhab/org.openhab.binding.zwave #1652: Unsupported Z-Wave command class COMMAND_CLASS_SECURITY_2

igorky, 2021-08-13:
I am trying to add a new switch, but something went wrong. It shows as unknown device in Openhab.
It works if I send commands from Z-Wave PC Controller 5 though.

* OH2 Version: 3.1.0
* Binding Version (if SNAPSHOT, provide compile time as found on the filename in the console): 3.1.0

If the problem is associated with a device, please provide the following -:

* Device name: ZEN30
* Device version: 
* Link to [database](http://opensmarthouse.org/zwavedatabase): https://opensmarthouse.org/zwavedatabase/1182
* If the device is not in the database, provide a link to the manual.
* If there is an XML being generted for the device (in the /userdata/zwave folder), please provide it.

Provide a full debug level log of the problem. Preferably, you should try use the [log viewer](https://www.opensmarthouse.org/utilities/logviewer/zwave/) to see if you can identify the problem.

`
2021-08-13 12:37:45.014 [DEBUG] [age.SerialApiGetInitDataMessageClass] - NODE 18: Node found
2021-08-13 12:37:45.066 [DEBUG] [ve.internal.protocol.ZWaveController] - NODE 18: Init node thread start
2021-08-13 12:37:45.097 [DEBUG] [l.initialization.ZWaveNodeSerializer] - NODE 18: Serializing from file /var/lib/openhab/zwave/network_d93d0751__node_18.xml
2021-08-13 12:37:45.136 [WARN ] [ve.internal.protocol.ZWaveController] - NODE 18: Restore from config: Error. Data invalid, ignoring config.
2021-08-13 12:37:45.136 [DEBUG] [alization.ZWaveNodeInitStageAdvancer] - NODE 18: Starting initialisation from EMPTYNODE
2021-08-13 12:37:45.182 [DEBUG] [ve.internal.protocol.ZWaveController] - NODE 18: Init node thread finished
2021-08-13 12:37:45.182 [DEBUG] [alization.ZWaveNodeInitStageAdvancer] - NODE 18: Node advancer - advancing to IDENTIFY_NODE
2021-08-13 12:37:45.182 [DEBUG] [alization.ZWaveNodeInitStageAdvancer] - NODE 18: Node advancer: Initialisation starting
2021-08-13 12:37:45.810 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 18: Controller status changed to ONLINE.
2021-08-13 12:37:45.810 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 18: Controller is ONLINE. Starting device initialisation.
2021-08-13 12:37:45.810 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 18: Updating node properties.
2021-08-13 12:37:45.810 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 18: Updating node properties. MAN=2147483647
2021-08-13 12:37:45.811 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 18: Properties synchronised
2021-08-13 12:37:45.811 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 18: Initialising Thing Node...
2021-08-13 12:37:45.811 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 18: Polling initialised at 1800 seconds - start in 919800 milliseconds.
2021-08-13 12:37:45.811 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 18: Device initialisation complete.
2021-08-13 12:37:49.315 [DEBUG] [rialmessage.IdentifyNodeMessageClass] - NODE 18: ProtocolInfo
2021-08-13 12:37:49.315 [DEBUG] [rialmessage.IdentifyNodeMessageClass] - NODE 18: Listening = true
2021-08-13 12:37:49.315 [DEBUG] [rialmessage.IdentifyNodeMessageClass] - NODE 18: Routing   = true
2021-08-13 12:37:49.315 [DEBUG] [rialmessage.IdentifyNodeMessageClass] - NODE 18: Beaming   = true
2021-08-13 12:37:49.315 [DEBUG] [rialmessage.IdentifyNodeMessageClass] - NODE 18: Version   = 4
2021-08-13 12:37:49.315 [DEBUG] [rialmessage.IdentifyNodeMessageClass] - NODE 18: FLIRS     = false
2021-08-13 12:37:49.315 [DEBUG] [rialmessage.IdentifyNodeMessageClass] - NODE 18: Security  = false
2021-08-13 12:37:49.315 [DEBUG] [rialmessage.IdentifyNodeMessageClass] - NODE 18: Max Baud  = 40000
2021-08-13 12:37:49.315 [DEBUG] [rialmessage.IdentifyNodeMessageClass] - NODE 18: Basic    = BASIC_TYPE_ROUTING_SLAVE
2021-08-13 12:37:49.315 [DEBUG] [rialmessage.IdentifyNodeMessageClass] - NODE 18: Generic  = GENERIC_TYPE_SWITCH_MULTILEVEL
2021-08-13 12:37:49.315 [DEBUG] [rialmessage.IdentifyNodeMessageClass] - NODE 18: Specific = SPECIFIC_TYPE_POWER_SWITCH_MULTILEVEL
2021-08-13 12:37:49.315 [DEBUG] [tocol.commandclass.ZWaveCommandClass] - NODE 18: Creating new instance of command class COMMAND_CLASS_NO_OPERATION
2021-08-13 12:37:49.316 [DEBUG] [tocol.commandclass.ZWaveCommandClass] - NODE 18: Command class COMMAND_CLASS_NO_OPERATION, endpoint 0 created
2021-08-13 12:37:49.316 [DEBUG] [tocol.commandclass.ZWaveCommandClass] - NODE 18: Version = 1, version set. Enabling extra functionality.
2021-08-13 12:37:49.316 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 18: Adding command class COMMAND_CLASS_NO_OPERATION to the list of supported command                  classes.
2021-08-13 12:37:49.316 [DEBUG] [tocol.commandclass.ZWaveCommandClass] - NODE 18: Creating new instance of command class COMMAND_CLASS_BASIC
2021-08-13 12:37:49.316 [DEBUG] [tocol.commandclass.ZWaveCommandClass] - NODE 18: Command class COMMAND_CLASS_BASIC, endpoint 0 created
2021-08-13 12:37:49.316 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 18: Adding command class COMMAND_CLASS_BASIC to the list of supported command classe                 s.
2021-08-13 12:37:49.317 [DEBUG] [alization.ZWaveNodeInitStageAdvancer] - NODE 18: Node Init response (0) org.openhab.binding.zwave.internal.protocol.ZWaveTransact                 ionResponse@2bc141ad
2021-08-13 12:37:49.317 [DEBUG] [alization.ZWaveNodeInitStageAdvancer] - NODE 18: Node Init transaction completed with response COMPLETE
2021-08-13 12:37:49.317 [DEBUG] [alization.ZWaveNodeInitStageAdvancer] - NODE 18: Node advancer - advancing to PING
2021-08-13 12:37:49.317 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 18: Got an event from Z-Wave network: ZWaveInitializationStateEvent
2021-08-13 12:37:49.317 [DEBUG] [ndclass.ZWaveNoOperationCommandClass] - NODE 18: Creating new message for command NO_OPERATION_PING
2021-08-13 12:37:49.317 [DEBUG] [alization.ZWaveNodeInitStageAdvancer] - NODE 18: ZWaveCommandClassTransactionPayload - send to node
2021-08-13 12:37:49.317 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 18: SECURITY not supported
2021-08-13 12:37:49.317 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 18: Command Class COMMAND_CLASS_NO_OPERATION is NOT required to be secured
2021-08-13 12:37:49.317 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 18: sendTransaction org.openhab.binding.zwave.internal.protocol.transaction.ZWaveCom                 mandClassTransactionPayload@6ec3cab1
2021-08-13 12:37:49.317 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 18: Adding to device queue
2021-08-13 12:37:49.318 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 18: Added 26 to queue - size 5
2021-08-13 12:38:22.387 [DEBUG] [ing.zwave.handler.ZWaveSerialHandler] - NODE 18: Sending REQUEST Message = 01 08 00 13 12 01 00 25 0A D8
2021-08-13 12:38:22.394 [DEBUG] [l.serialmessage.SendDataMessageClass] - NODE 18: sentData successfully placed on stack.
2021-08-13 12:38:22.395 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 18: TID 26: Transaction not completed
2021-08-13 12:38:22.518 [DEBUG] [l.serialmessage.SendDataMessageClass] - NODE 18: SendData Request. CallBack ID = 10, Status = Transmission complete and ACK recei                 ved(0)
2021-08-13 12:38:22.519 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 18: Response processed after 132ms
2021-08-13 12:38:22.519 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 18: TID 26: Transaction completed
2021-08-13 12:38:22.519 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 18: notifyTransactionResponse TID:26 DONE
2021-08-13 12:38:22.520 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 18: Got an event from Z-Wave network: ZWaveTransactionCompletedEvent
[... 99 lines cut ...]
2021-08-13 12:38:47.507 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 18: Properties synchronised
2021-08-13 12:38:47.508 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 18: MANUFACTURER not set
2021-08-13 12:38:47.508 [DEBUG] [l.initialization.ZWaveNodeSerializer] - NODE 18: Serialise aborted as static stages not complete
2021-08-13 12:38:47.508 [DEBUG] [alization.ZWaveNodeInitStageAdvancer] - NODE 18: Node advancer - advancing to DYNAMIC_VALUES
2021-08-13 12:38:47.508 [DEBUG] [ding.zwave.handler.ZWaveThingHandler] - NODE 18: Got an event from Z-Wave network: ZWaveInitializationStateEvent
2021-08-13 12:38:47.508 [DEBUG] [alization.ZWaveNodeInitStageAdvancer] - NODE 18: Node advancer: DYNAMIC_VALUES - checking COMMAND_CLASS_ZWAVEPLUS_INFO
[... cut ...]
2021-08-13 12:39:16.602 [DEBUG] [tocol.commandclass.ZWaveCommandClass] - NODE 18: Unsupported command class COMMAND_CLASS_SECURITY_2
2021-08-13 12:39:16.603 [DEBUG] [ng.zwave.internal.protocol.ZWaveNode] - NODE 18: Unsupported Z-Wave command class COMMAND_CLASS_SECURITY_2
2021-08-13 12:39:16.603 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 18: Commands processed 1.
2021-08-13 12:39:16.603 [DEBUG] [nal.protocol.ZWaveTransactionManager] - NODE 18: Checking command org.openhab.binding.zwave.internal.protocol.ZWaveCommandClassPa                 yload@431028a3.

`

cdjackson, 2021-08-13:
Security 2 is not supported. The device will fall back to security 0 during inclusion. This isn't a bug. I would suggest that you discuss this on the support forum where you will find a lot more people who can help you.

One point though - I don't really know why security 2 is being used here. Maybe you included the device using a different software - eg the PC controller since you mention that above. This will clearly not work with openhab and you should include the device using OH.

I'll close this, but please discuss on the forum if you require more support.

closed 2021-08-13 by cdjackson

igorky, 2021-08-13:
Thanks @cdjackson, I removed it from a controller and added via openhab. It works fine after that.