openhab/org.openhab.binding.zwave #1662: Abus smoke detector SHRM1000 is included/online but channels have NULL

Peter-DD, 2021-09-23:
**Environment**
* openhab 3.1.0; 
* Raspberry PI 3; 
* openhabian; 
* Z-wave Stick ZMEEUZBB; 
* Latest updates installed;
* [Abus SHRM10000](https://www.openhab.org/addons/bindings/zwave/thing.html?manufacturer=abus&file=shrm10000_0_0.html)

**Problem description**
I want to use Abus SHRM10000 Smoke detectors. 
I can include them without problems. They were correctly detected as SHRM10000.
All channels (binary, smoke, battery) were created but they always have NULL values only. I observe this behavior on all of the three Abus smokedetectors.  In the network map I can't see the node-IDs of smokedetectors but the node shows the ONLINE state at the thing pages.
If I raise an alert with real smoke the detector works and it is alerting with sound, but no message or alarm will send to controller (controler node-ID is 1 - the association groups in the things were correctly set)

In the logs I don't see any wakeups within default period of 24 hrs.
I tried to exclude/reset/include the smoke detectors and hardly reset the controler - but all didn't help.

I queried the smoke detector thing and the smokealarm item from the REST api, here is the output:
**Thing**
`{
  "statusInfo": {
    "status": "ONLINE",
    "statusDetail": "NONE"
  },
  "editable": true,
  "label": "Schlafzimmer - SHRM10000 Smoke detector",
  "bridgeUID": "zwave:serial_zstick:bf6d0d67c4",
  "configuration": {
    "wakeup_node": 1,
    "wakeup_interval": 21600,
    "node_id": 9
  },
  "properties": {
    "zwave_class_basic": "BASIC_TYPE_ROUTING_SLAVE",
    "zwave_class_generic": "GENERIC_TYPE_SENSOR_NOTIFICATION",
    "zwave_frequent": "false",
    "zwave_neighbours": "",
    "modelId": "SHRM10000",
    "zwave_version": "0.0",
    "zwave_listening": "false",
    "manufacturerId": "0403",
    "manufacturerRef": "0002:0003,0400:0002",
    "dbReference": "1047",
    "zwave_deviceid": "3",
    "zwave_nodeid": "9",
    "vendor": "ABUS Security-Center GmbH & Co. KG",
    "defaultAssociations": "1",
    "zwave_routing": "true",
    "zwave_beaming": "true",
    "zwave_secure": "false",
    "zwave_class_specific": "SPECIFIC_TYPE_NOTIFICATION_SENSOR",
    "zwave_manufacturer": "1027",
    "zwave_devicetype": "2"
  },
  "UID": "zwave:device:bf6d0d67c4:node9",
  "thingTypeUID": "zwave:abus_shrm10000_00_000",
  "channels": [
    {
      "linkedItems": [],
      "uid": "zwave:device:bf6d0d67c4:node9:sensor_binary",
      "id": "sensor_binary",
      "channelTypeUID": "zwave:sensor_binary",
      "itemType": "Switch",
      "kind": "STATE",
      "label": "Binary Sensor",
      "description": "Indicates if a sensor has triggered",
      "defaultTags": [],
      "properties": {
        "binding:*:OnOffType": "COMMAND_CLASS_SENSOR_BINARY"
      },
      "configuration": {}
    },
    {
      "linkedItems": [
        "Schlafzimmer_Alarmsmoke"
      ],
      "uid": "zwave:device:bf6d0d67c4:node9:alarm_smoke",
      "id": "alarm_smoke",
      "channelTypeUID": "zwave:alarm_smoke",
      "itemType": "Switch",
      "kind": "STATE",
      "label": "Alarm (smoke)",
      "description": "Indicates if a smoke is triggered",
      "defaultTags": [],
      "properties": {
        "binding:*:OnOffType": "COMMAND_CLASS_ALARM,COMMAND_CLASS_BASIC;type=SMOKE"
      },
      "configuration": {}
    },
    {
      "linkedItems": [
        "Schlafzimmer_Batterieladung"
      ],
      "uid": "zwave:device:bf6d0d67c4:node9:battery-level",
      "id": "battery-level",
      "channelTypeUID": "system:battery-level",
      "itemType": "Number",
      "kind": "STATE",
      "label": "Batterieladung",
      "defaultTags": [],
      "properties": {
        "binding:*:PercentType": "COMMAND_CLASS_BATTERY"
      },
      "configuration": {}
    }
  ]
}`

**smokealarm-Item**
`{
  "link": "http://openhabian:8080/rest/items/BrandmelderFlurOben_Alarmsmoke",
  "state": "NULL",
  "stateDescription": {
    "readOnly": true,
    "options": [
      {
        "value": "OFF",
        "label": "OK"
      },
      {
        "value": "ON",
        "label": "Alarm"
      }
    ]
  },
  "commandDescription": {
    "commandOptions": [
      {
        "command": "OFF",
        "label": "OK"
      },
      {
        "command": "ON",
        "label": "Alarm"
      }
    ]
  },
  "editable": true,
  "type": "Switch",
  "name": "BrandmelderFlurOben_Alarmsmoke",
  "label": "Brandmelder Flur Oben",
  "category": "Smoke",
  "tags": [
    "Point"
  ],
  "groupNames": [
    "Brandmelder"
  ]
}`

I've no ideas anymore.

**Additional info**
Two weeks ago also start a thread in [OH community](https://community.openhab.org/t/z-wave-binding-abus-shrm10000-included-but-all-channels-null/126419) but until now I've got no solution

Peter-DD, 2021-10-02:
Today I've updated to openHAB 3.2.0.M2. No Change with this behaviour. Device will included, has ONLINE(but is not visible in Network-Map) State, but channel-Values are NULL.
Does anybody has an idea or can point me to way to solve or debug it?

I'm very frustrated

cdjackson, 2021-10-02:
If the channels are null then the device is not reporting. I’d suggest to discuss this in the forum where there is a much larger viewing than here as this is probably not a bug in the binding.

Peter-DD, 2021-10-02:
I don‘t get any answers in the forum. (but OK I will try it again) I see no other possibility to solve this than asking the developer. The Abus SHRM1000 is official supported in OpenHab. At least it would help me to know, if anybody get this smokedetectors working or how to debug this issue. If there is no issue in OpenHab or zwave binding I can throw the sensors away.

cdjackson, 2021-10-02:
>  I see no other possibility to solve this than asking the developer. 

Unfortunately I do not have this device so I'm unable to provide a lot of support - this is why the forum is a better place to ask questions relating to devices.

As I said above, if the binding doesn't receive anything, then the device is probably not reporting anything. This could be because it is not initialised, or not configured correctly.

Peter-DD, 2021-10-03:
From zwave binding doc:

> We then initialise some information in the device such as associations. Association configuration is read from the database although the Z-Wave Plus Lifeline association group will be configured automatically to report to openHAB. 

I observed that after inclusion I see no association groups were set. From your documentation the initial config should read from database. Is it possible that these infos (association groups) are missing there?

Thank you very much.

cdjackson, 2021-10-03:
No - the associations are not missing. Also, the lifeline will be set automatically for ZWave+ devices.

I guess this is a battery device - maybe it's just not completing initialisation yet and you need to wake it up a few times for this to complete?

Peter-DD, 2021-10-03:
Yes, its a battery device. Per manual the default wakup period is 24hrs for this device. There is no button to force a wakeup. I also tried to force a wakeup by replace the batterie. This wasn‘t successfull. If I‘m waiting 24hrs I never see a message from device within this timerange.

cdjackson, 2021-10-03:
That's a bit surprising to be honest. The default wakeup might not work - it will heavily depend on the network as clearly the wakeup will not know where to send the wakeup command to, so it may not work. It might try and send a broadcast, but these do not get routed in ZWave, so if the device is not close to the controller, it will not be able to send the wakeup to the controller.

Peter-DD, 2021-10-03:
Many thanks Chris,
I've just test it: I exclude the device, put the device very close to the controller and included it. It was included without problems, but all channels are NULL. 
I've changed the association group to controller ID. The changestatus keeps "pending". Since the device has a default wakeup of 24hrs and there is no other way to force a wakeup, I will have a look in 24 hrs, but I'm not hopeful that it works. The detector will keep stay closed to controller.

Peter-DD, 2021-10-03:
Hi Chris,
your advices were very valuable for me solving this issue! I assume the key point is how to force of a wakeup of this device. 
I guess the default period of SHRM1000 smoke detector of 24 hrs is very long and I guess openhab won't keep the changes in command queue for 24 hrs. But this is an assumption only I'm not an expert.

What I have done:
* Setting controller in low inclusion mode
* Include the SHRM1000 very closed to controller. Channels has NULL values, wake-up node is 0 (but the controller has id 1)
* Setting the wake up node to controller, the wake up interval to 60 sec (for testing purpose), and the association to controller with id 1
```
  wakeup_node: 1
  wakeup_interval: 60
```
* config status changes to "pending"
* ** Random guessing I try to push 3 times the network button, and the "pending" status disappears **

Now the device works as expected!

Cheers Peter

closed 2021-10-03 by Peter-DD