openhab/org.openhab.binding.zwave #1672: Aeotec TriSensor (ZWA005-C) - No Sensor Data / Communication Errors

innerand, 2021-10-17:
I have problems with an Aeotec TriSensor (ZWA005-C). 
I don't receive any sensor data (values are always 'NULL') and it shows communication errors sometimes. 
The Sensor is working fine with the same hardware setup and Home-Assistant. Therefore I think there is a problem with the
Openhab Zwave binding. 

I tested with OpenHab 3.1.0 and 3.2.0-M3. The Log entries are from 3.2.0-M3. 

Log Output when waking the device manually (short button press): 
```
13:11:59.665 [DEBUG] [ing.zwave.internal.protocol.ZWaveNode] - NODE 21: WakeupTimerTask 0 Messages waiting, state VERSION
13:11:59.666 [DEBUG] [ing.zwave.internal.protocol.ZWaveNode] - NODE 21: No more messages, go back to sleep
13:11:59.666 [DEBUG] [.commandclass.ZWaveWakeUpCommandClass] - NODE 21: Creating new message for application command WAKE_UP_NO_MORE_INFORMATION
13:11:59.667 [DEBUG] [ing.zwave.internal.protocol.ZWaveNode] - NODE 21: SECURITY required on COMMAND_CLASS_WAKE_UP
13:11:59.667 [DEBUG] [ing.zwave.internal.protocol.ZWaveNode] - NODE 21: Command Class COMMAND_CLASS_WAKE_UP is required to be secured
13:11:59.667 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - NODE 21: sendTransaction org.openhab.binding.zwave.internal.protocol.transaction.ZWaveCommandClassTransactionPayload@57b19f83
13:11:59.668 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - NODE 21: Bump transaction 627 priority from Immediate to Immediate
13:11:59.669 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - NODE 21: Adding to device queue
13:11:59.669 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - NODE 21: Added 627 to queue - size 4
13:11:59.669 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - Transaction SendNextMessage 0 out at start. Holdoff false.
13:11:59.670 [DEBUG] [ommandclass.ZWaveSecurityCommandClass] - NODE 21: isNonceAvailable = null
13:11:59.671 [DEBUG] [mandclass.impl.CommandClassSecurityV1] - Creating command message SECURITY_NONCE_GET version 1
13:11:59.671 [DEBUG] [zwave.internal.protocol.SerialMessage] - Assembled message buffer = 01 09 00 13 15 02 98 40 25 5B 54 
13:11:59.672 [DEBUG] [ding.zwave.handler.ZWaveSerialHandler] - NODE 21: Sending REQUEST Message = 01 09 00 13 15 02 98 40 25 5B 54 
13:11:59.674 [DEBUG] [ZWaveSerialHandler$ZWaveReceiveThread] - Receive Message = 06
13:11:59.675 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - processReceiveMessage input 0<>128 : Message: class=null[0], type=ACK[2], dest=255, callback=0, payload=
13:11:59.676 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - Received msg (0): Message: class=null[0], type=ACK[2], dest=255, callback=0, payload=
13:11:59.676 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - lastTransaction null
13:11:59.676 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - Received msg: ACK
13:11:59.676 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - ZWaveReceiveThread queue empty
13:11:59.681 [DEBUG] [ZWaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 04 01 13 01 E8 
13:11:59.684 [DEBUG] [ding.zwave.handler.ZWaveSerialHandler] - Message SENT
13:11:59.685 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - Transaction SendNextMessage started: TID 628: [WAIT_RESPONSE] priority=High, requiresResponse=true, callback: 91
13:11:59.685 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - Transaction SendNextMessage 1 out at start. Holdoff false.
13:11:59.685 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - processReceiveMessage input 0<>128 : Message: class=SendData[19], type=Response[1], dest=255, callback=0, payload=01 
13:11:59.686 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - Received msg (0): Message: class=SendData[19], type=Response[1], dest=255, callback=0, payload=01 
13:11:59.686 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - lastTransaction TID 628: [WAIT_RESPONSE] priority=High, requiresResponse=true, callback: 91
13:11:59.686 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - Checking outstanding transactions: 1
13:11:59.686 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - Last transaction: TID 628: [WAIT_RESPONSE] priority=High, requiresResponse=true, callback: 91
13:11:59.686 [DEBUG] [ave.internal.protocol.ZWaveController] - Incoming Message: Message: class=SendData[19], type=Response[1], dest=255, callback=0, payload=01 
13:11:59.687 [DEBUG] [ol.serialmessage.SendDataMessageClass] - NODE 21: sentData successfully placed on stack.
13:11:59.687 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - TID 628: Advanced to WAIT_REQUEST
13:11:59.687 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - NODE 21: TID 628: Transaction not completed
13:11:59.687 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - ZWaveReceiveThread queue empty
13:11:59.687 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - Transaction SendNextMessage 1 out at start. Holdoff false.
13:12:04.687 [DEBUG] [nsactionManager$ZWaveTransactionTimer] - NODE 21: TID 628: Timeout at state WAIT_REQUEST. 3 retries remaining.
13:12:04.688 [DEBUG] [nsactionManager$ZWaveTransactionTimer] - Aborting Transaction!
13:12:04.689 [DEBUG] [ve.internal.protocol.ZWaveTransaction] - TID 628: Transaction ABORTED
13:12:04.689 [DEBUG] [zwave.internal.protocol.SerialMessage] - Assembled message buffer = 01 03 00 16 EA 
13:12:04.690 [DEBUG] [ding.zwave.handler.ZWaveSerialHandler] - NODE 255: Sending REQUEST Message = 01 03 00 16 EA 
13:12:04.691 [DEBUG] [ding.zwave.handler.ZWaveSerialHandler] - Message SENT
13:12:04.692 [DEBUG] [ZWaveSerialHandler$ZWaveReceiveThread] - Receive Message = 06
13:12:04.692 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - Transaction SendNextMessage 1 out at start. Holdoff false.
13:12:04.693 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - processReceiveMessage input 0<>128 : Message: class=null[0], type=ACK[2], dest=255, callback=0, payload=
13:12:04.694 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - Received msg (0): Message: class=null[0], type=ACK[2], dest=255, callback=0, payload=
13:12:04.695 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - lastTransaction TID 628: [ABORTED] priority=High, requiresResponse=true, callback: 91
13:12:04.696 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - Received msg: ACK
13:12:04.697 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - ZWaveReceiveThread queue empty
13:12:04.697 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - Transaction SendNextMessage 1 out at start. Holdoff false.
13:12:08.177 [DEBUG] [ZWaveSerialHandler$ZWaveReceiveThread] - Receive Message = 01 07 00 13 5B 01 03 52 E0 
13:12:08.178 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - processReceiveMessage input 0<>128 : Message: class=SendData[19], type=Request[0], dest=1, callback=91, payload=5B 01 03 52 
13:12:08.179 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - Received msg (0): Message: class=SendData[19], type=Request[0], dest=1, callback=91, payload=5B 01 03 52 
13:12:08.180 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - lastTransaction TID 628: [ABORTED] priority=High, requiresResponse=true, callback: 91
13:12:08.181 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - Checking outstanding transactions: 1
13:12:08.181 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - Last transaction: TID 628: [ABORTED] priority=High, requiresResponse=true, callback: 91
13:12:08.181 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - Checking TID 628: (Callback 91)
13:12:08.182 [DEBUG] [l.serialmessage.ZWaveCommandProcessor] - Callback match!
13:12:08.182 [DEBUG] [rnal.protocol.ZWaveTransactionManager] - Correlated to TID 628: callback 91
13:12:08.183 [DEBUG] [ave.internal.protocol.ZWaveController] - Incoming Message: Message: class=SendData[19], type=Request[0], dest=1, callback=91, payload=5B 01 03 52 
13:12:08.183 [DEBUG] [ol.serialmessage.SendDataMessageClass] - NODE 21: SendData Request. CallBack ID = 91, Status = Transmission complete, no ACK received(1)
13:12:08.183 [DEBUG] [ve.internal.protocol.ZWaveTransaction] - TID 628: Transaction CANCELLED
[... 138 lines cut ...]
13:30:17.955 [DEBUG] [nding.zwave.handler.ZWaveThingHandler] - NODE 21: Configuration update ignored config_10_1 to 1 (BigDecimal)
13:30:17.955 [DEBUG] [nding.zwave.handler.ZWaveThingHandler] - NODE 21: Configuration update set group_1 to [controller] (ArrayList)
13:30:17.955 [DEBUG] [nding.zwave.handler.ZWaveThingHandler] - NODE 21: Association 1 consolidated to [controller]
13:30:17.956 [DEBUG] [nding.zwave.handler.ZWaveThingHandler] - NODE 21: Unknown association group 1
[... cut ...]
13:30:17.966 [DEBUG] [nding.zwave.handler.ZWaveThingHandler] - NODE 21: Configuration update ignored config_24_2 to 3600 (BigDecimal)
13:30:17.966 [DEBUG] [nding.zwave.handler.ZWaveThingHandler] - NODE 21: Configuration update ignored node_id to 21 (BigDecimal)
13:30:17.966 [DEBUG] [nding.zwave.handler.ZWaveThingHandler] - NODE 21: Configuration update ignored config_6_1 to 0 (BigDecimal)
13:30:17.968 [INFO ] [openhab.event.ConfigStatusInfoEvent  ] - ConfigStatusInfo [configStatusMessages=[]]
```

cdjackson, 2021-10-17:
It would probably be better if you discussed this on the support forum rather than here as I suspect there is no bug. You say that this works with Home Assistant, but it's unclear what you are doing - from this log there is no communication with the device. Given that security is being used, I could guess that maybe you have not configured security in OH and therefore OH binding cannot communicate with any secure devices. That's just a guess though as there is very little information.

Please discuss on the support forum if this doesn't fix the issue, and please provide full information about your system and what you have done to try and get this working.

closed 2021-10-17 by cdjackson

kpanchen, 2021-10-17:
You need to include it unsecurely and make sure that it finished interview, you might need to wake up it a few times.

kpanchen, 2021-10-17:
> No - you don't need to include it without security. If the device supports security, then this should be fine, but if it was securely included using Home Assistant, then you MUST copy over the security key to openHAB - otherwise OH cannot communicate with the device.

If you recall my migration story it didn't work with security with battery powered devices, few days of fighting and nothing, somehow secure with battery devices always timeout and never finish interview. Same issue TC has. I ended up using secure only with lock devices, not sensors.

Not sure about security key, why do you need to copy it if device is reset and I have a network with OH security key?

kpanchen, 2021-10-18:
> > If you recall my migration story
> 
> Apologies, but I don't recall.

No need for apologies, that was some months ago, I had also an issue opened and ended up with unsecure which works fine since than.

> 
> > it didn't work with security with battery powered devices
> 
> The binding will work fine with battery powered devices and security - this is fine and many people use this.

It might be an issue with particular Aeotec sensors, the problem was that interview never finished and even waking up devices won't push it to competition. Back then I tried with TriSensor and Moto control. TC can also try and confirm, I doubt that's an issue with binding, rather with the sensors themselves. 

> 
> > Not sure about security key, why do you need to copy it if device is reset and I have a network with OH security key?
> 
> As I understood, maybe incorrectly, the device was not reset when it was used in OH. I could be wrong. I understood that the user was using the same dongle in both HA and OH (is just swapping the dongles between systems), and in this case you MUST use the same security key or security will not work.

If that's the case then yes.

innerand, 2021-11-01:
Security is (of course) used. The key was added to OpenHAB (all other devices which also use security just worked out of the box). I also removed and re-added the device within OpenHAB. An Aetoec Z-Stick Gen 5 is used as controller. 

I still think this is a problem with the OpenHAB Z-Wave-Stack, since others (kpanchen) also have problems with battery powered devices.