openhab/org.openhab.binding.zwave #1692: Unknown device - GOCONTROL Z-Wave SMART GLASS BREAK DETECTOR - Model# 0GB00Z-5

abbasmfazal, 2021-12-04:
This issue list is to report bugs or enhancements for the openHAB Z-Wave binding. If you're not sure about something, please consider discussing it on the [forum](http://community.openhab.org) before raising an issue - more people follow the forum than monitor the issues list, so if you are having trouble with a device, or want to ask a question, you will have a much larger audience there.

In order to ensure we have the best information to deal with the issue, please fill in the following information (delete as necessary depending on the issue).

* OH2 Version: openHAB 3.1.0 Release Build
* Binding Version (if SNAPSHOT, provide compile time as found on the filename in the console): 3.1.0

If the problem is associated with a device, please provide the following -:

* Device name: GOCONTROL Z-Wave SMART GLASS BREAK DETECTOR - Model# 0GB00Z-5 
* Device version: 0
* Link to [database](http://opensmarthouse.org/zwavedatabase): https://opensmarthouse.org/zwavedatabase/565 {Model, device type/look is same, manufacturer seems to be different}
* If the device is not in the database, provide a link to the manual: https://cdn.shopify.com/s/files/1/0067/9814/7669/files/GB00Z-Install.pdf
* If there is an XML being generted for the device (in the /userdata/zwave folder), please provide it.: Not generated 

Provide a full debug level log of the problem. Preferably, you should try use the [log viewer](https://www.opensmarthouse.org/utilities/logviewer/zwave/) to see if you can identify the problem.

2021-12-01 21:17:50.059 [INFO ] [g.discovery.internal.PersistentInbox] - Added new thing 'zwave:device:9684715332:node58' to inbox.
2021-12-01 22:57:33.628 [WARN ] [wave.discovery.ZWaveDiscoveryService] - NODE 58: Device discovery could not resolve to a thingType! Manufacturer data not known.

If a device will not initialise, ensure the log is provided from the time OH starts.

Thank you.

cdjackson, 2021-12-04:
Hi.
You can add your device to the database - please see the link below.

https://opensmarthouse.org/blog/zwave_device_database_guide

closed 2021-12-04 by cdjackson

abbasmfazal, 2021-12-04:
Thanks Chris. I am trying to follow the instructions and have opened up a ticket to get access for adding a device to zwave database. Pls. let me know if I am not on the right track.

Regards,
Abbas

cdjackson, 2021-12-04:
Thanks Abbas. I've just updated your access. If you have any problems, please discuss on the forum if possible - otherwise a ticket is ok (but the forum allows more people to help).

Cheers
Chris